RROrg/rr #110: DS214Play support

bossindelebile, 2023-07-01:
sorry for the unusual question, is it possible to have i686 support for installing the DS214play model? or a way to compile myself something compatible with it? Thanks in advance.

wjz304, 2023-07-02:
You can refer to

https://github.com/wjz304/redpill-lkm

https://github.com/wjz304/arpl-addons

https://github.com/wjz304/arpl-modules

But if you don't have a programming foundation, give up,

bossindelebile, 2023-07-02:
no, I'm not a programmer, I'm the closest thing to a geek, mainly electronic. anyway thanks for the useful links, I'll give it a couple of tries anyway. actually I already have an Xpenology NAS based on Asrock j5005 with ARPL, DS3622xs+ and I'm fine with it. my very first NAS is based on a laptop motherboard with Z520 CPU and with Nanocore X86 and DSM 5.0 it still manages to have its say for small jobs, I can stream my movies with 4k plex despite the 100 lan. of course I only do this for fun. Thanks again for the support.

wjz304, 2023-07-02:
1. You need to compile lkm for the corresponding model,
reference: 
https://github.com/wjz304/redpill-lkm/blob/master/config/platforms.h
You can search for "DS918+" to replace it

2. You need to compile plugins and drivers for the corresponding platform
reference:
https://github.com/wjz304/arpl-addons/blob/main/.github/workflows/main.yml
https://github.com/wjz304/arpl-modules/blob/main/.github/workflows/main.yml
You can search for "DS918+" to replace it

3. Write the corresponding model's config.
Arpl reference:
https://github.com/wjz304/arpl-i18n/blob/main/files/board/arpl/overlayfs/opt/arpl/model-configs/DS920%2B.yml
Jun‘s reference:
https://github.com/wjz304/redpill-load/blob/main/config/configs.json

PS:
The first step is very simple during compilation, but it is difficult to troubleshoot kernel crashes,
The subsequent process is relatively simple.

wjz304, 2023-07-02:
Referring to models with the same kernel version will make it easier for them, and DS918+ may not be the best reference

closed 2023-07-02 by wjz304