RROrg/rr #120: 离线安装

ikapc, 2023-07-13:
大佬，能否提供离线上传pat编译引导方式呢

wjz304, 2023-07-13:
pat的 url和md5 你自己改啊

wjz304, 2023-07-13:
url 可以写你自己的服务地址

wjz304, 2023-07-13:
哦，有 mirror 判断会替换掉自己的域

iZiwer, 2023-07-14:
定制了官网下线的6.2.3版本pat地址和md5后，能正常通过下载和hash校验，但是会报错：
`Patching zImage..
UNRECOVERY ERROR: Found . init.text offset @ C8F000
Found .rodata address @ 81600000
Found .rodata offset @ 800000
Patching boot params.
Analyzing f() candidate @ C978DE, PUSH @ C97BBC
[?] Found possible f() @ C978DE
  [+] Found LOCK-OR#0 sequence @ C97962 => F0 80 0D 7C 8D FD FF 01 [RIP+C706DE]
  [+] Found LOCK-OR#1 sequence @ C979B5 =〉 F0 80 0D 29 8D FD FF 02 [RIP+C706DE]
  [+] Found LOCK-OR#2 sequence @ C97A6A =〉F0 80 0D 74 8C FD FF 04 [RIP+C706DE]
  [+] Found LOCK-OR#3 sequence @ C97CF2 =〉 F0 80 0D BB 89 FD FF 08 [RIP+C706AD]
  [-] LOCK-OR PTR offset mismatch - 3/4 matched
Analyzing f() candidate @ D48EA4, PUSH @ D4F223
Failed to find matching sequences

Analyzing f() candidate @ D4FAE2, PUSH @ D4FD95`

wjz304, 2023-07-14:
不支持623

iZiwer, 2023-07-14:
> 不支持623

能否让它支持一下  毕竟6.2.3挺经典的，还能QC

wjz304, 2023-07-14:
没精力

closed 2023-07-14 by wjz304