RROrg/rr #121: ARPL 在ESXI 6.7下无法识别直通的xHCI设备，无法使用usb网卡

ss1969, 2023-07-13:
Install ENV:
 * motherboard: 技嘉B75N
 * CPU: 1265L v2
 * NIC: 

Arpl version: 23.7.2

DSM model: 3622xs
DSM version: 7.2

Issue:
 在ESXI 6.7下无法识别直通的xHCI设备。直通XHCI设备给ARPL虚拟机后，
进入arpl的shell，lsusb显示找到2个Linux Fundation USB root hub，但是无法找到上面插的设备
这导致无法识别usb nic realtek 8156并驱动为网卡设备。

logs:
Call menu.sh to return to menu
root@arpl:/opt/arpl# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@arpl:/opt/arpl# 
...

wjz304, 2023-07-14:
lsmod 看下

ss1969, 2023-07-16:
> lsmod 看下

现在暂时看不到lsmod。不过可以补充一下相关信息：
1，首先怀疑是esxi本身的问题。因为我用ubuntu和windows10也都进行了测试。如果直通usb设备而不是xhci的话，os都认为网卡是挂在480M端口上；直通xhci清空待补充。
2，裸机跑arpl，同样的网卡识别正常，所以arpl应该是能处理r8152的

wjz304, 2023-08-15:
无进一步信息，本issue关闭，如有其他问题请重新创建

closed 2023-08-15 by wjz304