RROrg/rr #122: BUG Does not reboot when push "Boot the loader"

navhawk80, 2023-07-14:
Install ENV:
 * motherboard: Gigabyte C246N-WU2
 * CPU: E-2226G
 * NIC: Intel I219-V

Arpl version: 23.7.3

DSM model: DS918+
DSM version: DSM 7.2-64570 Update 1

Issue: "Build the loader" from menu - then push "Boot the loader" from menu and it doesn't reboot. If I hard reset the PC it loads DSM without any issue, next reboots works. In other way If I connect a keyboard and monitor to PC I see that the console is alive and it reboots only if I type command "reboot" and push "Enter". Even if I reboot to the loader by using "RebootToArpl" script to update the loader or just to testing purpose and trying to reboot from menu by pushing "Boot the loader" it doesn't reboot. This issue starts to heppen to me since 2.7.0 or some earlier version. If I use "exit" - "reboot" in menu all works.
On second PC with Asrock J5005-itx motherboard all works well without any issue.
Thanks.

logs:

...

wjz304, 2023-07-14:
Yes, some motherboards have compatibility issues with kexec,
But using kexec for kernel switching can reduce the probability of kernel crashes.

At present, it is not very clear what specific features will cause incompatibility.

You can execute this command in the shell
`sed -i ‘s/#poweroff/poweroff/; s/kexec -f/#kexec -f/’ /opt/arpl/boot.sh`
Then go back to the advanced menu and perform "Persistence of arpl modifications".

wjz304, 2023-07-14:
Alternatively, in the next version, let's make it a user's choice first

navhawk80, 2023-07-14:
Thank you. It works now. :))

closed 2023-07-14 by navhawk80