RROrg/rr #127: 怎么将usb硬盘映射为内置硬盘？

AIchar, 2023-07-19:
你好，我在PVE下将一个usb硬盘直通给群辉虚拟机(DSM7.1)，群辉把他识别成了外挂存储，有什么办法可以让他把这个usb硬盘识别成内置的硬盘吗

wjz304, 2023-07-19:
https://github.com/wjz304/arpl-i18n/issues/126

AIchar, 2023-07-19:
这是我的配置
[image]
[image]

[image]
[image]

在群辉内只能被识别为外接设备

AIchar, 2023-07-20:
motherboard: 零刻EQ12 pro
CPU: i3-N305
NIC: VirtIO(PVE7.4)
Arpl version: latest

DSM model: ds3622xs+
DSM version: 7.2.64570

James62370, 2023-07-28:
I did it using the guide below.

https://www.diytechguru.com/2020/11/28/proxmox-hdd-passthrough-to-vm/

AIchar, 2023-07-29:
This is RDM pass-through. I wanted to pass-through the controller to the virtual machine, but I have found another solution now. I bought a PCI to SATA card and it's working fine.

wjz304, 2023-09-06:
[internalportcfg,esataportcfg,usbportcfg计算.xlsx](https://github.com/wjz304/arpl-i18n/files/12537070/internalportcfg.esataportcfg.usbportcfg.xlsx)

closed 2023-09-06 by wjz304

doandat943, 2023-09-20:
> [internalportcfg,esataportcfg,usbportcfg计算.xlsx](https://github.com/wjz304/arpl-i18n/files/12537070/internalportcfg.esataportcfg.usbportcfg.xlsx)

i have try with 23.9.4 but the usb disk alway show as USB Device not internal disk
[image]

wjz304, 2023-09-20:
> > [internalportcfg,esataportcfg,usbportcfg计算.xlsx](https://github.com/wjz304/arpl-i18n/files/12537070/internalportcfg.esataportcfg.usbportcfg.xlsx)
> 
> i have try with 23.9.4 but the usb disk alway show as USB Device not internal disk [image]

你要sdd的类型改成1啊

doandat943, 2023-09-21:
> > > [internalportcfg,esataportcfg,usbportcfg计算.xlsx](https://github.com/wjz304/arpl-i18n/files/12537070/internalportcfg.esataportcfg.usbportcfg.xlsx)
> > 
> > 
> > i have try with 23.9.4 but the usb disk alway show as USB Device not internal disk [image]
> 
> 你要sdd的类型改成1啊

"Do you want to change the type of sdd to 1?"

I don't know how this affects the system, but I watched a few tutorials on YouTube like [this](https://www.youtube.com/watch?v=KvW6LRw1RZs).

i want use usb hard drives as internal hard drives on synology 7.2 Update 3