RROrg/rr #129: 设备启动后假死，怀疑内核崩溃？

Ourworldmediagroup, 2023-07-21:
Install ENV:
 * motherboard:七彩虹H61
 * CPU:i5-2500K
 * NIC:Realtek RTL8111E

Arpl version:23.7.4

DSM model:DS3617xs，DS3615xs，DS920+
DSM version:6.2,7.0,7.2

Issue:设备启动正常，关闭UEFI启动，硬盘模式AHCI，配备的硬盘全部删除分区，无论是否输入SN或MAC均在编译完成启动时假死，
屏幕显示如下：
                                                                                                                                                                                                                   
                                                                                           Welcome to ARPL-i18n v23.7.2                                                                                            
                                                                                                                                                                                                                   
                                                                                          BOOTING: [BIOS] [USB flashdisk]
型号:  DS3617xs
版本:  6.2(25556u7)
LKM:   prod
CPU:   Intel(R) Core(TM) i5-2500K CPU @ 3.30GHz
MEM:   7665 MB
*** 检测到 1 个网卡, 自定义了 1 个 MACs, 其他网卡将使用自身 MACs. ***
Cmdline:
noefi console=ttyS0,115200n8 earlyprintk earlycon=uart8250,io,0x3f8,115200n8 root=/dev/md0 loglevel=15 log_buf_len=32M syno_hw_version=DS3617xs netif_num=1 pid=0x1666 mac1=001132A1228D sn=18B5ODN107800 vid=0x0951 elevator=elevator HddHotplug=0 syno_hdd_detect=0 vender_format_version=2 syno_hdd_powerup_seq=0
检测到 1 个网卡.
检查连接.
eth0 connected.

实体机上光标不闪烁，网卡不活动，路由器后台以及群晖软件均无法找到，多次尝试编译多个型号均出现假死情况

logs:
磁盘以及引导盘内均无日志输出

wjz304, 2023-07-21:
https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

wjz304, 2023-08-05:
No further feedback, this issue is closed

closed 2023-08-05 by wjz304