RROrg/rr #131: HP Proliant Microserver Gen8 disk drive detection

hazawer, 2023-07-24:
'arpl-i18n' worked very well with my Hp ProLiant MicroServer Gen8 with Intel(R) Celeron(R) CPU G1610T @ 2.30GHz and 4Gb of RAM.
'arpl' is installed on a 32Gb SSD on port 5 using the BIOS enabled B120i Smart Disk controller
I have configured the server using the Synology model DiskStation® DS923+ and DSM 7.2 update 1
One issue is that the system doesn't detect drives in specific 'slots' i.e. if the physical drives are in slots 1,2,4 the system displays drives in slots 1,2,3
Another issue is dynamic discovery of drives after the system is started i.e. if the physical drives are in slots 1,2 and I install a drive in slot 3 the system doesn't recognize the drive until I reboot the system

wjz304, 2023-07-27:
Because 923 is the model of DT,

The disk is set through a dts file, and every time it starts, it will be remade through the disks plugin, so it does not support hot swapping,
The existing logic does not consider vacant positions. As, if you insert 123 or 124, the final sorting in DTS is 123

wjz304, 2023-07-27:
If you want it to be consistent with the actual situation, you need to create your own DTB file according to the PCI location of the motherboard, and then upload your own DTS file through the advanced menu to make it effective

I have written a Chinese explanation on how to create DTS files: https://github.com/wjz304/Redpill_CustomBuild/issues/47#issuecomment-1201485025

closed 2023-07-31 by wjz304