RROrg/rr #132: i18n引导在pve中正常安装，但无法连接，重启虚拟机后又要重新安装

ZY1014, 2023-07-24:
[image]

shu0802, 2023-07-27:
[image]
在电脑上https://finds.synology.com/搜索不到设备

wjz304, 2023-07-28:
https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

shu0802, 2023-07-30:
> https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

请大神适配一下蜗牛星际J1900主机，谢谢

shu0802, 2023-07-31:
> https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM
请大神适配一下蜗牛星际J1900主机，谢谢。现在用这个可以tinycore-redpill-uefi.v0.9.0.2.img.gz适配ds918+7.1.1-42962。

ZY1014, 2023-07-31:
问题解决了。我cpu型号是N100, 创建虚拟机的时候不能选择host，换一个其他的型号可以了

closed 2023-07-31 by ZY1014