RROrg/rr #135: Why did you delete the vmdk?

PeterSuh-Q3, 2023-07-26:
I wonder why vmdk is not provided since v23.7.4.

Now users have to convert vmdk from img file and use it?

wjz304, 2023-07-27:
总有人问两个vmdk的区别，被问烦了，而且esxi8和7格式也不一样，

wjz304, 2023-07-29:
Too many people are asking about the difference between two vmdks, which is a bit annoying,
And the formats of esxi8 and 7 are also different, So it's better to convert according to your own needs.

closed 2023-07-29 by wjz304

ThahLiengJang, 2023-07-29:
如何转换，我用StarWind V2V Converter转换失败了。有没有大神教一下？

ThahLiengJang, 2023-07-29:
https://github.com/wjz304/arpl-i18n/assets/44521804/54577a10-9835-4773-a9d9-9f620dbee8fb

比如这种情况？

wjz304, 2023-07-29:
> 2023-07-29.23-10-05.mp4 
> 比如这种情况？

因为你路径有中文

clonikal, 2023-07-29:
arpl-i18n-23.7.3.vmdk-flat.zip was working perfectly.
You can install that version and then update from the menu to the latest version.