RROrg/rr #137: BUG

longtongue91, 2023-07-27:
Install ENV:
 * motherboard: GIGABYTE GA-J3455N-D3H
 * CPU: Intel Celeron Processor J3455
 * NIC: Realtek RTL8168-8111

Arpl version: arpl-i18n-23.7.4

DSM model: DS920+
DSM version: 7.2(64570)

Issue:
Hello, after configuring the bootloader, there is no entry into the web interface. Indicates that there is a connection and gives the IP. The router shows that this ip exists. By ip is not pinged from another PC in the local network. Psole reboot writes a timeout when searching for a network. ip does not show. can't turn on. there is a solution?
P.S. I'm sorry for my English! I write via google
translator)

logs:

screenshots:
[image]

...

wjz304, 2023-07-28:
Get the vid pid of the network card from the arpl shell `lspci nn`

Also, you can refer to this
 https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

longtongue91, 2023-07-29:
> Получить vid pid сетевой карты из оболочки arpl`lspci nn`

[image]
If I understand correctly, this is [10ec:8168].

wjz304, 2023-07-29:
same NIC: https://github.com/wjz304/arpl-i18n/issues/142

wjz304, 2023-07-29:
https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

My English is also very poor, I don't know if you can understand this after translation

wjz304, 2023-07-29:
You can speak your mother tongue because I also need to translate English into Chinese, so I can directly translate your mother tongue

longtongue91, 2023-07-29:
> https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM
> 
> My English is also very poor, I don't know if you can understand this after translation

Я почитал эту информацию. К сожалению, для меня это слишком сложно( Но я один пункт смог выполнить. Я установил DS918+ и веб-интерфейс заработал. Но предпочтительнее DS920+, так как я купил серийный номер и маки для полноценной работа DSM.
P.S. Вы тоже можете писать сразу на китайском, если вам это удобно. Спасибо вам за помощь!

longtongue91, 2023-07-29:
@wjz304 можете подробнее объяснить этот пункт - "DT型号目前不兼容sas扩展, 如果有会在启动时kernel崩溃." ? Как проверить можно?

wjz304, 2023-07-29:
If the DT model (as DS920+) uses the SAS hard drive, a kernel crash will occur during startup

wjz304, 2023-07-29:
[image]

longtongue91, 2023-07-29:
> https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

Я не много разобрался в статье. Я попробовал пункты 1, 2, 3 и 4 по очереди, но нет никаких изменений. Страница браузера становится белой, а в название вкладки пишет SYnology Web Assistant, а дальше ничего.
[image]
Пункт 5 точно не  про меня. У меня стоит обычный sata диск. Попробую полностью установить и настроить DS918+.
Под каждую модель Synology своя сборка DSM? Со своим уникальным ядром и драйверами? Получается - это норма. Моя плата примерно с такими же параметрами, что и оригинальный DS918+. Наверно поэтому и заработало. На DS920+, к сожалению, другая платформа и мне не подходит. Буду тестировать DS918+.
Спасибо вам огромное за ваш проект и помощь!

longtongue91, 2023-07-30:
Здравствуйте, похоже я нашёл проблему. Хотя это скорее особенность DSM на это материнской плате. Происходит очень долгая загрузка. Я вчера скомпилировал загрузчик и оставил загружаться. Я занимался другими делами 10-15 минут, а когда проверил - веб-интерфейс заработал. Я запустил установку DSM и снова подождал. Удалось дождаться экрана настройки DSM. Я настроил и всё заработало. После это я через загрузчик вбил серийный номер и мак-адреса, добавил аддон cpu-info и обновил загрузчик. Минут через 10 система загрузилась и работает. Только вот новая проблема, но думаю это связано с самой DSM.
[image]
Возможно, в [инструкцию](https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM) следует добавить пункт 7: запаситесь терпением и подождите 15 минут)
Ещё раз спасибо!

wjz304, 2023-07-30:
This time should be related to the performance of the CPU

closed 2023-07-30 by longtongue91