RROrg/rr #138: Esata setting not working.

gbm777, 2023-07-27:
Install ENV:
 * motherboard:
 * CPU:N5105
PVE 8.0

Arpl version:newest

DSM model:SA6400
DSM version:7.2

Issue:

Esata setting not working.

logs:

root@arpl:/mnt/p1# dmesg | grep SATA
[   11.512964] ahci 0000:00:1f.2: AHCI 0001.0000 32 slots 6 ports 1.5 Gbps 0x3f impl SATA mode
[   11.519356] ata1: SATA max UDMA/133 abar m4096@0xc2244000 port 0xc2244100 irq 28
[   11.519368] ata2: SATA max UDMA/133 abar m4096@0xc2244000 port 0xc2244180 irq 28
[   11.519378] ata3: SATA max UDMA/133 abar m4096@0xc2244000 port 0xc2244200 irq 28
[   11.519388] ata4: SATA max UDMA/133 abar m4096@0xc2244000 port 0xc2244280 irq 28
[   11.519397] ata5: SATA max UDMA/133 abar m4096@0xc2244000 port 0xc2244300 irq 28
[   11.519407] ata6: SATA max UDMA/133 abar m4096@0xc2244000 port 0xc2244380 irq 28
[   11.521314] ahci 0000:06:07.0: AHCI 0001.0000 32 slots 6 ports 1.5 Gbps 0x3f impl SATA mode
[   11.527202] ata7: SATA max UDMA/133 abar m4096@0xc1602000 port 0xc1602100 irq 29
[   11.527215] ata8: SATA max UDMA/133 abar m4096@0xc1602000 port 0xc1602180 irq 29
[   11.527225] ata9: SATA max UDMA/133 abar m4096@0xc1602000 port 0xc1602200 irq 29
[   11.527236] ata10: SATA max UDMA/133 abar m4096@0xc1602000 port 0xc1602280 irq 29
[   11.527245] ata11: SATA max UDMA/133 abar m4096@0xc1602000 port 0xc1602300 irq 29
[   11.527256] ata12: SATA max UDMA/133 abar m4096@0xc1602000 port 0xc1602380 irq 29
[   11.837324] ata2: SATA link down (SStatus 0 SControl 300)
[   11.839707] ata1: SATA link down (SStatus 0 SControl 300)
[   11.853952] ata11: SATA link down (SStatus 0 SControl 300)
[   11.854803] ata6: SATA link down (SStatus 0 SControl 300)
[   11.855714] ata8: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
[   11.856499] ata10: SATA link down (SStatus 0 SControl 300)
[   11.862799] ata5: SATA link down (SStatus 0 SControl 300)
[   11.862804] ata7: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
[   11.865937] ata12: SATA link down (SStatus 0 SControl 300)
[   11.867774] ata4: SATA link down (SStatus 0 SControl 300)
[   11.867942] ata9: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
[   11.870184] ata3: SATA link down (SStatus 0 SControl 300)

`root@arpl:/mnt/p1# cat user-config.yml 
lkm: prod
directboot: "false"
notsetmacs: "false"
bootipwait: "1"
kernelway: power
model: "SA6400"
productver: "7.2"
buildnum: "64570"
smallnum: "1"
paturl: "https://cndl.synology.cn/download/DSM/release/7.2/64570-1/DSM_SA6400_64570.pat"
patsum: "83b591bfb240c51b2feff8f02eac8bde"
sn: "2270UMRG1F1ZA"
layout: qwerty
keymap: ""
zimage-hash: "fb2c85bf3d6512a59a61745e96a93f13cf96444cd917d26f6c835c6196334bcc"
ramdisk-hash: "b9abb59cf33d0426ebe50b5fc81d7cf8ab25c58f84f84232deb2a7a72be33e60"
cmdline: {}
synoinfo:
  support_disk_compatibility: no
  support_memory_compatibility: no
  support_led_brightness_adjustment: no
  support_syno_hybrid_raid: ""
  supportraidgroup: ""
  maxlanport: "8"
  netif_seq: 0 1 2 3 4 5 6 7
  esataportcfg: "0xfdf"
  maxdisks: "12"
  internalportcfg: "0x20"
addons:
  misc: ""
  acpid: ""
  reboottoarpl: ""
  codecpatch: ""
  console: ""
  cpuinfo: ""
  dbgutils: ""
  hdddb: ""
  lsiutil: ""
  powersched: ""
modules:
  8139cp: ""
  aic94xx: ""
  alx: ""
  aqc111: ""
  asix: ""
  atl1c: ""
  atl1e: ""
  atlantic: ""
  ax88179_178a: ""
  b44: ""
  be2net: ""
  blk-mq-virtio: ""
  bnx2: ""
  bnx2x: ""
  bnxt_en: ""
  button: ""
  cdc_ncm: ""
  cfbcopyarea: ""
  cfbfillrect: ""
  cfbimgblt: ""
  crc-itu-t: ""
  dme1737: ""
  e1000: ""
  e1000e: ""
  failover: ""
  fb: ""
  hwmon-vid: ""
  i40e: ""
  iavf: ""
  igb: ""
  igbvf: ""
  igc: ""
  intel_auxiliary: ""
  ixgbe: ""
  ixgbevf: ""
  libsas: ""
  mii: ""
  mlx4_core: ""
  mlx4_en: ""
  mlx5_core: ""
  mpt3sas: ""
  nct6775: ""
  net_failover: ""
  r8125: ""
  r8152: ""
  r8153_ecm: ""
  r8169: ""
  realtek: ""
  rtc-cmos: ""
  scsi_transport_sas: ""
  skge: ""
  sky2: ""
  tg3: ""
  thermal: ""
  usbnet: ""
  vesafb: ""
  virtio: ""
  virtio_input: ""
  virtio_mmio: ""
  virtio_net: ""
  virtio_pci: ""
  virtio_ring: ""
  virtio_scsi: ""
  vmw_pvscsi: ""
  vmxnet3: ""
original-mac1: 0ed06e3ad356
vid: "0x46f4"
pid: "0x0001"`

[image]
[image]
硬盘都是PVE虚拟的。

DSM 仍旧识别为SATA。
[image]

根据识别盘序，尝试调整esataportcfg: "0x7ff"  internalportcfg: "0x800"
或者esataportcfg: "0xffe"  internalportcfg: "0x1"

都宣告失败。

请解决一下，谢谢。

wjz304, 2023-07-28:
dt的型号 *portcfg应该也是无效。

gbm777, 2023-07-30:
像918这种型号能有效吗

wjz304, 2023-07-30:
> 像918这种型号能有效吗

嗯 型号列表里 有 -DT 后缀的 就是 device tree 的

wjz304, 2023-07-30:
> dt的型号 *portcfg应该也是无效。

不过这只是直觉，我并没有验证过他

closed 2023-08-15 by wjz304