RROrg/rr #139: DSM couldn't connect to Synology services

joachimpfeifer, 2023-07-28:
I've updated my HP ProLiant Gen8 to DSM 7.2. Everything works wonderfully except the internet access through the package center and the update window. Is this a know issue or are there some hints?
[image]
[image]

wjz304, 2023-07-28:
try restart, Currently, the reason is unknown

joachimpfeifer, 2023-07-28:
Unfortunately no success. Restarted several times and also changed serial number manually.

wjz304, 2023-07-28:
cat /etc/synoinfo.conf 
[image]
Check if these addresses are correct

joachimpfeifer, 2023-07-28:
First is right, second is different:
[image]
How could I update this?

wjz304, 2023-07-28:
Modify through the vi command, or copy to a PC, and then put it back after modification
```
rss_server="http://update7.synology.com/autoupdate/genRSS.php"
rss_server_ssl="https://update7.synology.com/autoupdate/genRSS.php"
rss_server_v2="https://update7.synology.com/autoupdate/v2/getList"
```

joachimpfeifer, 2023-07-28:
Have changed everything, unfortunately no improvement

joachimpfeifer, 2023-07-28:
None of them are listed:
[image]

wjz304, 2023-07-28:
Also, I would like to ask what version of boot you are using?
Upgrading from the original ARPL?

wjz304, 2023-07-28:
Because the latest arpl-i18n does not modify the link of 'rss_server'

joachimpfeifer, 2023-07-28:
I've used the original ARPL before and built a new loader with april-i18n yesterday. The DSM-installation has been adpoted. I'm using version 23.7.4.

joachimpfeifer, 2023-07-28:
I'm very sorry for the confusion. However the DSM wouldn't get a connection to Synology with my laptop but on my mobile phone everything works fine. I've to look on my laptop setting....

wjz304, 2023-07-28:
currently, I can't think of other reason,
or take a look at the firewall.

delta-whiplash, 2023-07-30:
hello I have the same issue

wjz304, 2023-07-30:
是的，最近很多人反馈这个问题，

delta-whiplash, 2023-07-30:
no solutions found ?
I found this link https://xpenology.com/forum/topic/53817-tinycore-redpill-loader-tcrp/?do=findComment&comment=289467

wjz304, 2023-07-30:
https://xpenology.com/forum/topic/65408-automated-redpill-loader-arpl/page/91/#comment-447688

delta-whiplash, 2023-07-30:
I see so am gonna wait to see If this error disappear thank you

wjz304, 2023-08-05:
It seems like it's okay

closed 2023-08-05 by wjz304