RROrg/rr #140: 7.2u2 nvme

wjz304, 2023-07-28:
Extracted from 64570u1
[libsynonvme.so.1.gz](https://github.com/wjz304/arpl-i18n/files/12193690/libsynonvme.so.1.gz)

PeterSuh-Q3, 2023-07-29:
Yesterday, in addition to the DT, other other case tests were conducted.

1. Case where libsynonvme.so.1 needs to be changed to xxd hexa
In the case of DS918+/DS1019+/DS1621xs+ models, you need to find the position to be replaced from the version extracted from 7.2-64570 U0 and use the resulting hexa value.
libsynonvme.so.1 recorded after the patch uses the one inside DSM as it is.
https://github.com/PeterSuh-Q3/tcrp-addons/tree/main/nvme-cache/releases

Otherwise, if you refer to U1 or any other version, you will fall into an infinite installation loop.

2. Another case DS3622xs+, etc. In other cases that are neither DT nor xxd Hexa change cases, libsynonvme.so.1 does not need to be replaced, just use the DSM internal ones as they are.

The script that adjusts according to these three patterns is implemented below, and it is designed to correspond to both the modules parameter and the late parameter so that it can be used immediately in ARPL.
If necessary, you can use it as an Addon of ARPL.

https://github.com/PeterSuh-Q3/tcrp-addons/tree/main/nvme-cache

ARC's disks is also a good addon, but it seems to do too many things at once in it.
Each one has not been verified, and it has been confirmed that some of the scripts cause errors.

wjz304, 2023-07-29:
@PeterSuh-Q3 Thank you !

wjz304, 2023-07-29:
@PeterSuh-Q3 
I don't seem to see any processing for SA6400. Is there a test?

jim3ma, 2023-07-30:
@PeterSuh-Q3 I think it is easy to solve update 2 issue: https://github.com/jim3ma/arpl-addons/commit/0ea7c369645057defb3e58eee7cebc794a5f07ba

wjz304, 2023-07-30:
> @PeterSuh-Q3 I think it is easy to solve update 2 nvme issue: [jim3ma/arpl-addons@0ea7c36](https://github.com/jim3ma/arpl-addons/commit/0ea7c369645057defb3e58eee7cebc794a5f07ba) I have tested it in SA6400 64570 Update 2.

加 power_limit 好使了？ 难道我加错了？

jim3ma, 2023-07-31:
> > @PeterSuh-Q3 I think it is easy to solve update 2 nvme issue: [jim3ma/arpl-addons@0ea7c36](https://github.com/jim3ma/arpl-addons/commit/0ea7c369645057defb3e58eee7cebc794a5f07ba) I have tested it in SA6400 64570 Update 2.
> 
> 加 power_limit 好使了？ 难道我加错了？

你姿势不对，躺着的时候加才行。

PeterSuh-Q3, 2023-07-31:
> 

前几天，在一位拥有正版DS920+的用户的帮助下，我也检测到了model.dts的差异。

https://svrforum.com/nas/878503?page=3#&gid=7&pid=1

但我不知道它有什么作用，所以我就传递了它。
power_lomit 是否仅对 SA6400 硬编码值 100？
除了SA6400之外，是否适用于其他型号？

PeterSuh-Q3, 2023-07-31:
> > > @PeterSuh-Q3 I think it is easy to solve update 2 nvme issue: [jim3ma/arpl-addons@0ea7c36](https://github.com/jim3ma/arpl-addons/commit/0ea7c369645057defb3e58eee7cebc794a5f07ba) I have tested it in SA6400 64570 Update 2.
> > 
> > 
> > 加 power_limit 好使了？ 难道我加错了？
> 
> 你姿势不对，躺着的时候加才行。
> 
> 可能和机型有关，有的机型是两个数字，SA6400 是一个数字

As you said we seem to need 2 for the other model (DS920+).

power_limit="14.85,7.425";

jim3ma, 2023-07-31:
> As you said we seem to need 2 for the other model (DS920+).
>
> power_limit="14.85,7.425";

Yes, it depends on machine type, like one for SA6400, two for DS920+.

All logic is hard code into libsynonvme.so

wjz304, 2023-07-31:
如果 power_limit 的数值影响结果，可以考虑从原文件中提取

PeterSuh-Q3, 2023-07-31:
For ds923+, the power_limit text exists in the libsynonvme.so.1 file.
Is the "power_limit.nvme_model_spec_get.c" file involved in this setting?

Let's take a look at the model.dtb file without the dt patch for ds923+.
It seems that the pure setting remains.

`root@VM_FUSION:/usr/lib# xxd -c 256 libsynonvme.so.1 | grep power_limit
0000b900: 5f62 795f 6c6f 632e 6300 2f73 7973 2f62 6c6f 636b 2f25 732a 6e2a 0025 733a 2564 2072 6561 6420 6572 726f 7220 3a25 7300 2f64 6576 2f25 7300 2573 3a25 6420 6f75 7420 6f66 206d 656d 6f72 7920 746f 2061 6c6c 6f63 2067 6c6f 6220 6c6f 6f6b 696e 6720 666f 723a 2573 006e 766d 655f 6372 6974 6961 6c5f 7761 726e 696e 675f 6765 742e 6300 6e76 6d65 5f70 6572 6365 6e74 6167 655f 7573 6564 5f67 6574 2e63 0070 6f77 6572 5f6c 696d 6974 006e 766d 655f 6d6f 6465 6c5f 7370 6563 5f67 6574 2e63 002c 0025 733a 2564 2042 6164 2070 6172 616d 7465 7200 7063 6965 7061 7468 002f 7275 6e2f 6d6f 6465 6c2e 6474 6200 2573 3a25 6420 6661 696c 2074 6f20 7265 6164  _by_loc.c./sys/block/%s*n*.%s:%d read error :%s./dev/%s.%s:%d out of memory to alloc glob looking for:%s.nvme_critial_warning_get.c.nvme_percentage_used_get.c.power_limit.nvme_model_spec_get.c.,.%s:%d Bad paramter.pciepath./run/model.dtb.%s:%d fail to read
0000c100: 696c 6564 2064 6f20 6368 6563 6b20 666f 7220 7365 7269 616c 2025 7320 6375 724c 6966 6520 2564 006e 766d 655f 6465 765f 6765 745f 706f 7765 725f 6c69 6d69 742e 6300 6e76 6d65 5f69 6e74 2e63 0025 733a 2564 2046 6169 6c65 6420 746f 2063 7265 6174 6520 2573 3a20 2573 0025 732f 2573 002f 6269 6e2f 6370 0025 733a 2564 2046 6169 6c65 6420 746f 2063 6f70 7920 2573 0025 733a 2564 2046 6169 6c65 6420 746f 2073 6574 206b 6579 2025 7300 0000 0000 2573 3a25 6420 4661 696c 6564 2074 6f20 6765 7420 6c6f 636b 2025 732c 2065 7272 6f72 3d25 7800 0000 2573 3a25 6420 4661 696c 6564 2074 6f20 756e 6c6f 636b 2c20 6572 726f 723d 2578 0000 0000 0000 0000  iled do check for serial %s curLife %d.nvme_dev_get_power_limit.c.nvme_int.c.%s:%d Failed to create %s: %s.%s/%s./bin/cp.%s:%d Failed to copy %s.%s:%d Failed to set key %s.....%s:%d Failed to get lock %s, error=%x...%s:%d Failed to unlock, error=%x........`

PeterSuh-Q3, 2023-07-31:
Come to think of it, I don't think I can access DSM without doing dt patch. Let's double check the /etc.default/model.dtb file at the junior level.
If this is not possible, you will need to seek help from a genuine DS923+ user again.

jim3ma, 2023-07-31:
> Come to think of it, I don't think I can access DSM without doing dt patch. Let's double check the /etc.default/model.dtb file at the junior level. If this is not possible, you will need to seek help from a genuine DS923+ user again.

Before update, move `reboot` and `poweroff` to other folder, and then update. The DSM will not reboot, so we can login and get  `/etc.default/model.dtb`.

wjz304, 2023-07-31:
更新成功重启之前挂载磁盘提取出来就行了，但是得想办法让脚本自己备份并抓取里面的参数

wjz304, 2023-07-31:
不过 固定的值如果有效就没必要备份它了，我先测试看看

jim3ma, 2023-07-31:
> 不过 固定的值如果有效就没必要备份它了，我先测试看看

强制开启 M.2 的，默认是没有 power_limit 的，SA6400 的 power_limit 是在 M2D20 和 E10M20-T1 扩展卡定义里面，是两个，看上去是和盘数量有关

wjz304, 2023-07-31:
[image]
power_limit=\"100\";
DS1821+

wjz304, 2023-07-31:
第三个 类型为 HDD，没有健康状态

jim3ma, 2023-07-31:
> 第三个 类型为 HDD，没有健康状态

所以这个是和 `机型+硬盘` 布局相关的，针对特定的机器，可能需要逆向去看一下

PeterSuh-Q3, 2023-07-31:
This is the result of DS923+.

         compatible = "Synology";
         model = "synology_r1000_923+";
         version = <0x01>;
         syno_spinup_group = <0x02 0x01 0x01>;
         syno_spinup_group_delay = <0x0b>;
         syno_hdd_powerup_seq = "true";
         syno_cmos_reg_secure_flash = <0xe0>;
         syno_cmos_reg_secure_boot = <0xe2>;
         power_limit = "14.85,9.9";

PeterSuh-Q3, 2023-07-31:
The above contents were confirmed by the contents of the original model.dtb file backed up in the Junior state.
from the model.dtb file before overwriting the model.dtb file.
It looks like you can extract the value of the original power_limit .
Wouldn't it be just a matter of putting this value back into model.dtb to be created?

wjz304, 2023-07-31:
> The above contents were confirmed by the contents of the original model.dtb file backed up in the Junior state. from the model.dtb file before overwriting the model.dtb file. It looks like you can [...]

At present, there is no idea that can be automatically completed through scripts unless all are manually obtained.
Alternatively, through patch, make libsynonvme.so not handle power_limit, According to jim3ma, it may still be related to other so

PeterSuh-Q3, 2023-07-31:
This is the result of DS1821+ 7.2-64570 U2.

        compatible = "Synology";
        model = "synology_v1000_1821+";
        version = <0x01>;
        syno_spinup_group = <0x08>;
        syno_spinup_group_delay = <0x00>;
        syno_hdd_powerup_seq = "true";
        syno_smbus_hdd_type = "microp";
        syno_smbus_hdd_adapter = <0x01>;
        syno_smbus_hdd_address = <0x47>;
        syno_cmos_reg_secure_flash = <0xe0>;
        syno_cmos_reg_secure_boot = <0xe2>;
        syno_uart_logout_gpio = <0x07 0x01>;
        power_limit = "14.85,9.075";

PeterSuh-Q3, 2023-07-31:
> At present, there is no idea that can be automatically completed through scripts unless all are manually obtained. Alternatively, through patch, make libsynonvme.so not handle power_limit, According [...]

Can you give me the result of using "14.85,9.075" instead of "100" for your DS1821+?

PeterSuh-Q3, 2023-07-31:
> 第三个 类型为 HDD，没有健康状态，这应该都是至多支持2个m2吧？

This means that the value is not fixed and can be variable depending on the number of disks or circumstances.
I can't find many cases because I don't have enough test environment right now, but I'll investigate the model.dtb file more after work.

wjz304, 2023-07-31:
Three days ago, I obtained the original value and that it was not OK, so I replaced the so of u1,
It seems that I made some operational mistakes before,

Wait a moment, I'll try again

> Can you give me the result of using "14.85,9.075" instead of "100" for your DS1821+?

wjz304, 2023-07-31:
14.85,9.075 Do not display SSD

wjz304, 2023-07-31:
> > 第三个 类型为 HDD，没有健康状态，这应该都是至多支持2个m2吧？
> 
> This means that the value is not fixed and can be variable depending on the number of disks or circumstances. I can't find many cases because I don't have enough test environment right now, but I'll investigate the model.dtb file more after work.

So there's no need to obtain other models,

PeterSuh-Q3, 2023-07-31:
> So there's no need to obtain other models,

thanks for the test It is correct to find a dynamically usable power_limit value to really implement the logic.

wjz304, 2023-07-31:
https://github.com/wjz304/arpl-addons/commit/1a604de31692ad4426678e4dd5bbc202d8b1be6e

PeterSuh-Q3, 2023-07-31:
> [wjz304/arpl-addons@1a604de](https://github.com/wjz304/arpl-addons/commit/1a604de31692ad4426678e4dd5bbc202d8b1be6e)

Does this script of yours work ok?

wjz304, 2023-07-31:
```
OK：
root@test:~# ls /sys/block/nvme* -ld
lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme0/nvme0n1
lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme1n1 -> ../devices/pci0000:00/0000:00:16.0/0000:0b:00.0/nvme/nvme1/nvme1n1
lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme2n1 -> ../devices/pci0000:00/0000:00:17.0/0000:13:00.0/nvme/nvme2/nvme2n1

NOK：
root@test:~# ls /sys/block/nvme* -ld
lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme0/nvme0n1
lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme1/nvme0n2
lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme2/nvme0n3

```

PeterSuh-Q3, 2023-07-31:
> ```
> OK：
> root@test:~# ls /sys/block/nvme* -ld
> lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme0/nvme0n1
> lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme1n1 -> ../devices/pci0000:00/0000:00:16.0/0000:0b:00.0/nvme/nvme1/nvme1n1
> lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme2n1 -> ../devices/pci0000:00/0000:00:17.0/0000:13:00.0/nvme/nvme2/nvme2n1
> 
> NOK：
> root@test:~# ls /sys/block/nvme* -ld
> lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme0/nvme0n1
> lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n2 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme0/nvme0n2
> lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n3 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme0/nvme0n3
> ```

Are you testing with VMWARE-FUSION?
The second test result is the same as when I made two or more virtual NVMe in VMWARE-FUSION.
Due to a bug in VMWARE-FUSION, the same address is assigned to the 2nd and 3rd NVMe.
This cannot be the case with normal bare metal.
So I always only test 1 NVMe in VMWARE-FUSION.

PeterSuh-Q3, 2023-08-01:
> [image] [image] [image] 这样就可以了

Chinese is supported by a translator.
Image capture is difficult to understand.
Is the setting screen of VMWARE-FUSION correct as I expected?

wjz304, 2023-08-01:
I haven't used vmware under Mac before, and I'm not sure if it's the same as vmware under Win,

Place nvme hard drive at 0:0, 1:0, 2:0 ... then It's OK
Place nvme hard drive at 0:0, 0:1, 0:2 ... then It's NOK

[image]
[image]
[image]

PeterSuh-Q3, 2023-08-01:
> I haven't used vmware under Mac before, and I'm not sure if it's the same as vmware under Win,
> 
> Place nvme hard drive at 0:0, 1:0, 2:0 ... then It's OK Place nvme hard drive at 0:0, 0:1, 0:2 ... then It's NOK
> 
> [image] [image] [image]

VMWARE Fusion for mac is missing the ability to freely map PCIe ports.
So, I installed the Windows version of the Player and set it up with the mapping method you told me, and the test was successful with two or more NVMes.
Refer to the following contents among the vmk file contents of the player
I ported it to FUSION and I was able to test the same.

`nvme0.present = "TRUE"
nvme0:0.fileName = "NVMe1.vmdk"
nvme0:0.present = "TRUE"
nvme0.pciSlotNumber = "160"
nvme0:0.redo = ""
nvme0.subnqnUUID = "52 2e f9 e1 67 53 15 91-ce 64 34 19 d8 82 5b b0"

nvme1.present = "TRUE"
nvme1:0.fileName = "NVMe2.vmdk"
nvme1:0.present = "TRUE"
nvme1.pciSlotNumber = "192"
nvme1:0.redo = ""
nvme1.subnqnUUID = "52 fd 26 96 ce bc 1a cc-93 d6 95 0b 1f 02 81 6f"
`

lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme0n1 -> ../devices/pci0000:00/0000:00:15.0/0000:03:00.0/nvme/nvme0/nvme0n1
lrwxrwxrwx 1 root root 0 Aug  1 00:18 /sys/block/nvme1n1 -> ../devices/pci0000:00/0000:00:16.0/0000:0b:00.0/nvme/nvme1/nvme1n1

wjz304, 2023-08-01:
> Yesterday, in addition to the DT, other other case tests were conducted.
>
> 1. Case where libsynonvme.so.1 needs to be changed to xxd hexa
>    In the case of DS918+/DS1019+/DS1621xs+ models, you need [...]

https://github.com/wjz304/arpl-addons/blob/main/nvmecache/all/usr/bin/nvmecache.sh
I made a rewrite, libsynonvme.so.1 has been modified, but it doesn't seem to be OK, I don't know if there were any omissions?

PeterSuh-Q3, 2023-08-02:
> https://github.com/wjz304/arpl-addons/blob/main/nvmecache/all/usr/bin/nvmecache.sh I made a rewrite, libsynonvme.so.1 has been modified, but it doesn't seem to be OK, I don't know if there were any [...]

The changed libsynonvme.so.1 file can be verified using xxd .

According to this script, the problem seems to be gone until 64570-U1,
According to my experience, it seems that there will be problems starting from U2.
So, I included the U0 version of the libsynonvme.so.7.2.xxd file in the addon in advance.

And after this script is installed, a reboot is required to start the service.
If you copy the libsynonvme.so.1 file to /tmpRoot at the late time, you do not need to install the service and restart.

I overhauled my addon yesterday, but you seem to be referring to an older version.
You have removed all the logic of your service form.

wjz304, 2023-08-02:
> https://github.com/wjz304/arpl-addons/blob/main/nvmecache/all/usr/bin/nvmecache.sh

https://github.com/wjz304/arpl-addons/blob/main/nvmecache/all/usr/bin/nvmecache.sh
I used PVE to simulate NVME testing yesterday and couldn't see nvme disk,
Just now, using win vmware testing seems to work OK.
It seems to still have a lot to do with the simulation method of nvme

PeterSuh-Q3, 2023-08-02:
> 

In the case of PVE, the passed-through NVMe would have been no problem.
I will also simulate virtual NVMe storage with my script in PVE for experience.

closed 2023-08-03 by wjz304