RROrg/rr #142: 型号SA6400无法驱动网卡r8169的BUG

zui-lang, 2023-07-28:
Install ENV:
 * motherboard: 梅捷N5095
 * CPU: N5095
 * NIC: r8169

Arpl version: 23.7.4

DSM model: 7.1/7.2
DSM version: SA6400

Issue:

在aprl安装界面可以成功驱动r8169的网卡，但是边缘完成启动群晖后无法驱动网卡，路由器后台也无法查找到，连接显示器ip也没有显示，一直显示`connect…`

更换型号为918，版本为7.2后一切正常

logs:

...

wjz304, 2023-07-28:
取下 网卡的 pidvip
lspci -nn

zui-lang, 2023-07-28:
请问应该在哪里执行`lspci -nn`这行命令，是在群晖的ssh还是在aprl的安装页面

zui-lang, 2023-07-28:
> 取下 网卡的 pidvid
> 
> lspci -nn

请问这个是什么意思

wjz304, 2023-07-28:
arpl 菜单 退出 子菜单 退出到shell

zui-lang, 2023-07-28:
```
root@arpl:/opt/arpl# lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:4e24]
00:02.0 VGA compatible controller [0300]: Intel Corporation JasperLake [UHD Graphics] [8086:4e55] (rev 01)
00:04.0 Signal processing controller [1180]: Intel Corporation Dynamic Tuning service [8086:4e03]
00:08.0 System peripheral [0880]: Intel Corporation Device [8086:4e11]
00:14.0 USB controller [0c03]: Intel Corporation Device [8086:4ded] (rev 01)
00:14.2 RAM memory [0500]: Intel Corporation Device [8086:4def] (rev 01)
00:15.0 Serial bus controller [0c80]: Intel Corporation Serial IO I2C Host Controller [8086:4de8] (rev 01)
00:15.2 Serial bus controller [0c80]: Intel Corporation Device [8086:4dea] (rev 01)
00:16.0 Communication controller [0780]: Intel Corporation Management Engine Interface [8086:4de0] (rev 01)
00:17.0 SATA controller [0106]: Intel Corporation Device [8086:4dd3] (rev 01)
00:19.0 Serial bus controller [0c80]: Intel Corporation Device [8086:4dc5] (rev 01)
00:19.1 Serial bus controller [0c80]: Intel Corporation Device [8086:4dc6] (rev 01)
00:1c.0 PCI bridge [0604]: Intel Corporation Device [8086:4db8] (rev 01)
00:1e.0 Communication controller [0780]: Intel Corporation Device [8086:4da8] (rev 01)
00:1e.3 Serial bus controller [0c80]: Intel Corporation Device [8086:4dab] (rev 01)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:4d87] (rev 01)
00:1f.3 Audio device [0403]: Intel Corporation Jasper Lake HD Audio [8086:4dc8] (rev 01)
00:1f.4 SMBus [0c05]: Intel Corporation Jasper Lake SMBus [8086:4da3] (rev 01)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Jasper Lake SPI Controller [8086:4da4] (rev 01)
01:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
```
[image]

wjz304, 2023-07-28:
10ec:8168

zui-lang, 2023-07-28:
> 10ec:8168

请问是什么意思，这个应该怎么解决

wjz304, 2023-07-28:
驱动有，

方便的时候 截个图吧

zui-lang, 2023-07-29:
eth r8169网卡正确驱动
[image]
型号选择SA6400，版本选择7.1后进行编译
[image]
然后启动
[image]
然后就会卡在这个界面，然后群晖助手什么的也找不到设备，就长按电源键手动重启机器
重启后机器自动进入DSM启动引导
[image]
这次重启后虽然驱动了网卡也有ip，但是这个地址打不开也ping不通

zui-lang, 2023-07-29:
> eth r8169网卡正确驱动 [image] 型号选择SA6400，版本选择7.1后进行编译 [image] 然后启动 [image] 然后就会卡在这个界面，然后群晖助手什么的也找不到设备，就长按电源键手动重启机器 重启后机器自动进入DSM启动引导 [image] 这次重启后虽然驱动了网卡也有ip，但是这个地址打不开也ping不通

这次虽然没有100%复现当时问题，应该是环境有所不同，我将路由器的静态IP分配关掉试试

zui-lang, 2023-07-29:
关闭路由器静态ip分配后，无论是是7.1还是7.2的sa6400都能正确驱动网卡，获取到ip，但ip均无法ping通，通过群晖助手等软件也找不到局域网内的群晖，目前和 #133 情况一致

wjz304, 2023-07-29:
https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

zui-lang, 2023-07-29:
`root@arpl:/opt/arpl#`这行的光标确实是不闪烁了，请问直接启动是那个arpl高级设置里面的那个吗
[image]

wjz304, 2023-07-29:
yes

zui-lang, 2023-07-29:
更换启动模式也没办法，只好放弃硬解了

zui-lang, 2023-08-09:
我问了一下tb官方客服，说网卡型号是rtl8111

zui-lang, 2023-08-09:
看了其他的issue，尝试使用23.5.10，编译时提示无法连接网络或本地磁盘无空间

wjz304, 2023-08-09:
qq吧

wjz304, 2023-08-09:
主页不是有群号

zui-lang, 2023-08-09:
已加群2

wjz304, 2023-08-09:
23.8.4

zui-lang, 2023-08-10:
中午回去试试

wjz304, 2023-08-10:
23.8.5

closed 2023-08-10 by wjz304