RROrg/rr #144: BUG - Bootloader not auto boot to synology

doandat943, 2023-07-29:
Install ENV:
 * motherboard: ASUS X541U
 * CPU: Intel Core i3 Skylake6100U
 * NIC: AX88179

Arpl version: [23.7.4](https://github.com/wjz304/arpl-i18n/releases/tag/23.7.4)

DSM model: DS918+
DSM version: 7.2 Update 1

Issue: Bootloader not auto boot to synology, my xpenology lost power and auto-start when power comeback, it stuck at boot menu like the picture and need press Enter to continue boot to synology. In previous version it working normal.

[image]

logs:

...

wjz304, 2023-07-29:
Some motherboards may experience this during abnormal power outages,

Check the BIOS to modify the options related to incoming calls and try it out

doandat943, 2023-07-29:
> 

thank you i will try check it

doandat943, 2023-07-29:
@wjz304 hi i have try reset and re setting mainboard, but it doesnt effect. I think so. The problem is still there. I need press enter to make bootloader boot to synology when grub menu appears.
[image]

doandat943, 2023-07-29:
I don't remember too much but it seems that in the old version when the GNU menu appeared, it would start counting down the time (maybe 5s), then it would automatically boot into DSM. But in the new version, it doesn't seem to count down anymore, forcing me to press enter when the computer boots up

doandat943, 2023-07-29:
It's true that recently my xpenology during abnormal power outages.

wjz304, 2023-07-29:
or check the options related to  "super io" , "serial port".

wjz304, 2023-07-29:
https://github.com/wjz304/arpl-i18n/issues/14

doandat943, 2023-08-02:
it works again in version [23.8.0](https://github.com/wjz304/arpl-i18n/releases/tag/23.8.0)

closed 2023-08-02 by doandat943