RROrg/rr #145: 918+ 7.2 cpu,内存性能指标信息不显示.

avatasia, 2023-07-29:
主机:  蜗牛星际B款.
系统: 918+, 7.2.1
问题:   资源监控里,性能全是0,  snmpd服务无法启动.

● snmpd.service - SNMP Daemon
   Loaded: loaded (/usr/lib/systemd/system/snmpd.service; static; vendor preset: disabled)
   Active: failed (Result: start-limit) since Sat 2023-07-29 23:57:18 CST; 4min 11s ago
  Process: 13953 ExecStart=/usr/syno/lib/systemd/scripts/snmpd.sh (code=dumped, signal=ABRT)
 Main PID: 13953 (code=dumped, signal=ABRT)
手动启动服务, 提示这个:
sudo /usr/syno/lib/systemd/scripts/snmpd.sh
*** buffer overflow detected ***: terminated
Aborted

重装了好几次, 问题依旧. 3637也试过啦.

wjz304, 2023-08-07:
Install ENV:

motherboard:
CPU:
NIC:
Arpl version:

avatasia, 2023-08-15:
已解决，貌似群晖主机名修改后， 会导致/etc/snmp/snmpd.conf丢掉sysName  <hostname> 这个信息，
补全后，systemctl start snmpd, 一切正常。

closed 2023-08-15 by avatasia

wjz304, 2023-09-16:
https://github.com/wjz304/arpl-i18n/issues/219