RROrg/rr #147: BUG-关于r8125b编译后重启找不到ip的测试

lingshengjiang898, 2023-07-31:
Install ENV:
 * motherboard:云星b365
 * CPU: i3-8300T
 * NIC: r8125b

Arpl version:

DSM model: ds920+
DSM version:7.1-7.2

Issue:编译后无法链接ip，切换LKM版本无效，切换直接启动无效，7.1版本下切换为原版v1.1beta2a可以，7.1和7.2版本切换最新版arpl-i18n无效

logs:

...
下面开启测试：
1、首先用原版测试（7.1版本 型号选择ds920+），经过测试完美
[image]
2、[arpl-i18n](https://github.com/wjz304/arpl-i18n) 23.7.3进行测试，当型号选择为ds920+、ds918+ 支持硬解型号时，编译重启之后能找到ip均无法连接上（7.1和7.2都是一样的，网卡识别正确）
[image]
lspci nn如下
[image]
[image]

神奇的事情就是，选择sa6400型号可以完美支持，重启能找到ip！！！！，但是不支持硬解。
一些说明：
主板用的是云星b365，自带2个2.5g r8125b网卡，自己加了一张双口万兆光网卡，型号OCE11102，群晖下均可免驱。
为了排除自己加的网卡影响，做过以下尝试
1、拔出OCE11102网卡，仅保留板载r8125b
2、插入pcie网卡，编译重启后网线连接OCE11102网卡
3、按照博主文章切换LKM版本，切换直接启动、切换不设置mac
以上测试均无效，必须型号修改为sa6400！但是想保留硬解，希望博主帮忙看看，为啥型号选920、918原版装7.1可以，这个版本7.1、7.2都不想行

lingshengjiang898, 2023-08-02:
目前停留在v1.1beta2a，想升级7.2可是无法选择硬解型号（920、918），希望大佬帮忙看看

wjz304, 2023-08-02:
确认下 mac 和网卡顺序吧

wjz304, 2023-08-02:
8125 遇到过非原mac 不能dhcp的问题。
https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

lingshengjiang898, 2023-08-02:
谢谢解答，我尝试过用只用板载的两个8125网卡，其中一个mac设置成TANK电玩 公开的半洗白码（sn+mac），另一个设置成自动获取mac，还是不行重启连不上，ping不通，find不到，怀疑不是dhcp问题应该没正确启动；
还尝试过模块那里禁用8169网卡驱动只保留8125也不行；
为了排除7.2问题，我安装7.1版本也不行
以下情况就很ok，直接装很完美
1、原版v1.1beta2a+7.1+任意型号（尝试过ds920+ds918+dva1622）均可以
2、只能选i18n+7.2/7.1+sa6400（尝试过ds920+ds918+dva1622均启动无法连接）
还尝试过先用原版v1.1beta2a在920下安装7.1，然后群晖内部升级成7.2，重新用最新的i18n进行引导，还是出现全新安装一样的问题，重启后连不上

bing1234, 2023-08-07:
和楼上完全一模一样的情况。纳斯B365主板，带有两个 RTL8125B 2.5G网口。
1、原版v1.1beta2a+7.1+任意型号（尝试过ds920+ds918+1621）均可以
2、i18n + 7.2+ 918/920/1621 都是可以安装系统，安装完成后无法发现设备。

lingshengjiang898, 2023-08-07:
大佬有没有原版小改支持7.2版本的引导，help～

marboroman, 2023-08-17:
I got the same issue with this motherboard. But I can build 920+ with any latest arpl-i18n and sometimes it receives dhcp and sometimes not. Can't find the way to make it work stable.

marboroman, 2023-08-17:
I saved logs after it failed to boot. Please [check](https://msnap.qpoe.com/logs.rar). It failed with the settings: "prod" and "disable direct boot".

marboroman, 2023-08-17:
> Attempt to uncheck button.ko in the module, compile and boot, and get logs again

did it: [link](https://msnap.qpoe.com/logs2.rar)

wjz304, 2023-08-18:
> > Attempt to uncheck button.ko in the module, compile and boot, and get logs again
> 
> did it: [logs2](https://msnap.qpoe.com/logs2.rar) [logs3](https://msnap.qpoe.com/logs3.rar)

The network card driver is normal, but there may be other incompatible drivers. 
Or try only enable r8125 in the module first.
Or try models without nuclear display, models other than Apollolake and Geminilake

wjz304, 2023-08-18:
```
acpi error:
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   39.121907] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   39.133873] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT0.PRT0._GTF] (Node ffff88025f0d8c08), AE_NOT_FOUND (20150930/psparse-540)
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   39.177229] ACPI Error: [DSSP] Namespace lookup failure, AE_NOT_FOUND (20150930/psargs-359)
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   39.189212] ACPI Error: Method parse/execution failed [\_SB.PCI0.SAT0.PRT0._GTF] (Node ffff88025f0d8c08), AE_NOT_FOUND (20150930/psparse-540)

i915 error:
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   40.587458] [drm:intel_dp_aux_ch [i915]] *ERROR* dp aux hw did not signal timeout (has irq: 1)!
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   40.619449] [drm:intel_dp_aux_ch [i915]] *ERROR* dp aux hw did not signal timeout (has irq: 1)!
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   40.641449] [drm:intel_dp_aux_ch [i915]] *ERROR* dp aux hw did not signal timeout (has irq: 1)!
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   40.654449] [drm:intel_dp_aux_ch [i915]] *ERROR* dp aux hw did not signal timeout (has irq: 1)!
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   40.654471] [drm:intel_dp_aux_ch [i915]] *ERROR* dp_aux_ch not done status 0xad4003ff
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   40.679451] [drm:intel_dp_aux_ch [i915]] *ERROR* dp aux hw did not signal timeout (has irq: 1)!
2023-08-17T04:01:20-07:00 SynologyNAS kernel: [   40.702455] [drm:intel_dp_aux_ch [i915]] *ERROR* dp aux hw did not signal timeout (has irq: 1)!

```

marboroman, 2023-08-18:
> models without nuclear display

what is "nuclear display"?  
do we see the list of used drivers when booting so I don't have to switch all of them one by one?

wjz304, 2023-08-18:
> > models without nuclear display
> 
> what is "nuclear display"? do we see the list of used drivers when booting so I don't have to switch all of them one by one?

igpu, Sorry, I didn't check if the words were appropriate because I used translation software.

Yes, but first of all, you can to be able to connect to it.
```
lsmod   # Loaded Drivers
```

marboroman, 2023-08-18:
I'd like to use hw transcoding so can't refuse igpu. will try to switch test\dev\prod and check that i915 error in logs.

wjz304, 2023-08-18:
> I'd like to use hw transcoding so can't refuse igpu. will try to switch test\dev\prod and check that i915 error in logs.

Just trying to locate the problem,
Determine if the problem is i915

marboroman, 2023-08-18:
> Determine if the problem is i915

ok, testing with ds923+ now

marboroman, 2023-08-19:
ds6400 is 100% stable: 10 successful reboots of 10. 
ds923+ is not stable - it's the same as ds920+: about 13 successful reboots of 20 and only if switched to "direct boot". 
923 and 6400 both have no intel igpu. So seems the problem is not about i915 or not just because of it.

marboroman, 2023-08-19:
[wjz304](https://github.com/wjz304), [here is](https://msnap.qpoe.com/logs6.rar) new  logs dump from ds923

wjz304, 2023-08-19:
> [wjz304](https://github.com/wjz304), [here is](https://msnap.qpoe.com/logs6.rar) new logs dump from ds923

Can 10.5.0.140 be ping ?

marboroman, 2023-08-19:
> Can 10.5.0.140 be ping ?

it always can be pinged for first 5 seconds right after booting then disappears and returns during next minute if boot successful or never returns if stuck.

wjz304, 2023-08-19:
[image]
Have you try this option?

marboroman, 2023-08-19:
[wjz304](https://github.com/wjz304), tried it long ago and didn't notice that it makes difference. Ok, will try it again. 

Now I tried to deselect all modules except r8125 for ds923+ and it reboots 100% successful so far.

marboroman, 2023-08-19:
Is it better to set static ip address in DSM to exclude router role?

wjz304, 2023-08-19:
> Is it better to set static ip address in DSM to exclude router role?
I think static IP should be set on the route.

marboroman, 2023-08-20:
[wjz304](https://github.com/wjz304), why does your loader have no "i915" in modules fo ds920+? 
[image]

marboroman, 2023-08-21:
so far reduced selected modules to this list: 

[image]

and it started to boot more stable for ds920+

I'm not sure what all these modules for but they seemed potentially system important. please let me know which else I could delete and if any important are missing.

wjz304, 2023-08-21:
> so far reduced selected modules to this list:
> 
> [image]
> 
> and it started to boot more stable for ds920+
> 
> I'm not sure what all these modules for but they seemed potentially system important. please let me know which else I could delete and if any important are missing.

That's fine, some of them can be removed, but it's not necessary.

wjz304, 2023-08-22:
https://github.com/wjz304/arpl-i18n/issues/157

lingshengjiang898, 2023-08-22:
请问下是不是星云b365主板是不是得按照上面截图来去掉不需要的组件才能正常找到ip？

wjz304, 2023-08-22:
> 请问下是不是星云b365主板是不是得按照上面截图来去掉不需要的组件才能正常找到ip？

差不多是吧

marboroman, 2023-08-22:
It stopped booting for me again. Nothing helps. I hate this motherboard and lan adapter.

lingshengjiang898, 2023-08-23:
> It stopped booting for me again. Nothing helps. I hate this motherboard and lan adapter.

可以尝试原版v1.1beta2a，可以在920、918正常工作，不过只能停留在 7.1，不支持7.2

marboroman, 2023-08-23:
@wjz304, It boots again for me. The reason was [R44SF for M.2 NVMe SSD Riser Card Extension Cable](https://aliexpress.com/item/1005005864827465.html) - I use it for one of the m.2 nvme slots and if m.2 storage is inserted in this adapted dsm stops booting. If I put off the storage from this adapter and insert it in second m.2 slot the DSM boots ok. If I leave m.2 nvme storage in this adapter it shows up in bios and different operating systems, I can format it and use. Does DSM or loader need some kind of driver or extension to make it work?

wjz304, 2023-08-23:
> @wjz304, It boots again for me. The reason was [R44SF for M.2 NVMe SSD Riser Card Extension Cable](https://aliexpress.com/item/1005005864827465.html) - I use it for one of the m.2 nvme slots and if [...]

As far as I know, some motherboards have shared channels between m2 and sata.

misamai, 2023-08-26:
用23.5.0版本就行了

lingshengjiang898, 2023-08-28:
> 

23.5.0版本好像没发选最新的dsm，下载会失败用不了

closed 2023-10-28 by wjz304

marboroman, 2023-10-28:
unfortunately it didn't solve the problem with booting... swithed to module 8125_v9 and it booted ok for the first time but another 2 reboots were bad then booted again. will stay on v9 for a while and observe...