RROrg/rr #148: BUG: incorrect CPU freq detection

TheChief5055, 2023-07-31:
Install ENV:
 * motherboard: ASROCK J3355B-ITX
 * CPU: Intel Celeron J3355
 * NIC: RTL8111GR

Arpl version: 23.7.5

DSM model: DS918+
DSM version: 7.2(64570u2)

Issue: incorrect CPU freq detection

ARPL prior DSM load:
Model: DS918+
Build: 7.2(64570u2)
LKM:   prod
CPU:   Intel(R) Celeron(R) CPU J3355 @ 2.00GHz
MEM:   7809 MB

logs:
# dmesg | grep GHz

[    0.049094] smpboot: CPU0: Intel(R) Celeron(R) CPU J3355 @ 2.00GHz (family: 0x6, model: 0x5c, stepping: 0x9)

It's in fact 2.5Ghz and was recognized as so in old ARPL.

wjz304, 2023-08-05:
lscpu

TheChief5055, 2023-08-05:
# lscpu
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  2
  On-line CPU(s) list:   0,1
Vendor ID:               GenuineIntel
  BIOS Vendor ID:        Intel
  Model name:            Intel(R) Celeron(R) CPU J3355 @ 2.00GHz
    BIOS Model name:     Intel(R) Celeron(R) CPU J3355 @ 2.00GHz Fill By OEM CPU
                          @ 2.0GHz
    BIOS CPU family:     15
    CPU family:          6
    Model:               92
    Thread(s) per core:  1
    Core(s) per socket:  2
    Socket(s):           1
    Stepping:            9
    Frequency boost:     enabled
    CPU(s) scaling MHz:  46%
    CPU max MHz:         2500.0000
    CPU min MHz:         800.0000
    BogoMIPS:            3993.73
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mc
                         a cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss
                         ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arc
                         h_perfmon pebs bts rep_good nopl xtopology nonstop_tsc
                         aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est
                         tm2 ssse3 sdbg cx16 xtpr pdcm sse4_1 sse4_2 x2apic movb
                         e popcnt tsc_deadline_timer aes xsave rdrand lahf_lm 3d
                         nowprefetch intel_pt ibrs ibpb stibp tpr_shadow vnmi fl
                         expriority ept vpid fsgsbase tsc_adjust smep erms mpx r
                         dseed smap clflushopt sha_ni xsaveopt xsavec xgetbv1 dt
                         herm ida arat pln pts arch_capabilities
Virtualization features:
  Virtualization:        VT-x
Caches (sum of all):
  L1d:                   48 KiB (2 instances)
  L1i:                   64 KiB (2 instances)
  L2:                    2 MiB (2 instances)
Vulnerabilities:
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Spec store bypass:     Vulnerable
  Spectre v1:            Vulnerable: __user pointer sanitization and usercopy ba
                         rriers only; no swapgs barriers
  Spectre v2:            Vulnerable, IBPB: disabled, STIBP: disabled
  Srbds:                 Not affected
  Tsx async abort:       Not affected

wjz304, 2023-08-05:
Is it still 23.7.5?
You can try 23.8.1 and the modules in 23.8.1 add back cpufreq_performance.ko Library.

TheChief5055, 2023-08-05:
It's 27.8.0. I'll try 23.8.1.

wjz304, 2023-08-05:
and update modules

TheChief5055, 2023-08-05:
On 23.8.1 cpu freq is detected correctly. Thanx!

closed 2023-08-05 by TheChief5055