RROrg/rr #149: BUG 今天执行了一下全部更显后Option "i915.enable_guc=-1" not recognized - ignoring

cjjdaq, 2023-07-31:
Install ENV:
 * motherboard:
 * CPU:i7-9500u
 * NIC:

Arpl version:

DSM model:
DSM version:

Issue:
之前可以正常识别，加载guc
logs:
[    0.000000] Command line: withefi synoboot_satadom=2 dom_szmax=1034 console=ttyS0,115200n8 earlyprintk earlycon=uart8250,io,0x3f8,115200n8 root=/dev/md0 loglevel=15 log_buf_len=32M syno_hw_version=DVA1622 i915.enable_guc=-1 netif_num=1 synoboot2 pid=0x0000 i915.enable_fbc=1 i915.force_probe=5916 mac1=0011327ef381 sn=2040UBRQMXSCL HddEnableDynamicPower=1 vid=0x0000 elevator=elevator intel_iommu=igfx_off i915.enable_dc=0 vender_format_version=2 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8
[    0.000000] Kernel command line: withefi synoboot_satadom=2 dom_szmax=1034 console=ttyS0,115200n8 earlyprintk earlycon=uart8250,io,0x3f8,115200n8 root=/dev/md0 loglevel=15 log_buf_len=32M syno_hw_version=DVA1622 i915.enable_guc=-1 netif_num=1 synoboot2 pid=0x0000 i915.enable_fbc=1 i915.force_probe=5916 mac1=0011327ef381 sn=2040UBRQMXSCL HddEnableDynamicPower=1 vid=0x0000 elevator=elevator intel_iommu=igfx_off i915.enable_dc=0 vender_format_version=2 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8
[    1.737429] <redpill/cmdline_delegate.c:417> Cmdline: withefi synoboot_satadom=2 dom_szmax=1034 console=ttyS0,115200n8 earlyprintk earlycon=uart8250,io,0x3f8,115200n8 root=/dev/md0 loglevel=15 log_buf_len=32M syno_hw_version=DVA1622 i915.enable_guc=-1 netif_num=1 synoboot2 pid=0x0000 i915.enable_fbc=1 i915.force_probe=5916 mac1=0011327ef381 sn=2040UBRQMXSCL HddEnableDynamicPower=1 vid=0x0000 elevator=elevator intel_iommu=igfx_off i915.enable_dc=0 vender_format_version=2 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8
[    1.738617] <redpill/cmdline_delegate.c:429> Param #10: |i915.enable_guc=-1|
[    1.738627] <redpill/cmdline_delegate.c:324> Option "i915.enable_guc=-1" not recognized - ignoring
[    1.738700] <redpill/cmdline_delegate.c:429> Param #14: |i915.enable_fbc=1|
[    1.738711] <redpill/cmdline_delegate.c:324> Option "i915.enable_fbc=1" not recognized - ignoring
[    1.738721] <redpill/cmdline_delegate.c:429> Param #15: |i915.force_probe=5916|
[    1.738732] <redpill/cmdline_delegate.c:324> Option "i915.force_probe=5916" not recognized - ignoring
[    1.738868] <redpill/cmdline_delegate.c:429> Param #22: |i915.enable_dc=0|
[    1.738878] <redpill/cmdline_delegate.c:324> Option "i915.enable_dc=0" not recognized - ignoring
[    1.765149] <redpill/sanitize_cmdline.c:102> Sanitized cmdline to: withefi console=ttyS0,115200n8 earlycon=uart8250,io,0x3f8,115200n8 root=/dev/md0 syno_hw_version=DVA1622 i915.enable_guc=-1 netif_num=1 synoboot2 i915.enable_fbc=1 i915.force_probe=5916 mac1=0011327ef381 sn=2040UBRQMXSCL HddEnableDynamicPower=1 intel_iommu=igfx_off i915.enable_dc=0 vender_format_version=2 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8
[    2.815538] i915 0000:0b:00.0: BAR 6: can't assign [??? 0x00000000 flags 0x20000000] (bogus alignment)
[    2.816030] [drm:gen9_set_dc_state [i915]] *ERROR* DC state mismatch (0x0 -> 0x2)
[    2.817474] [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[    2.824861] [drm] Initialized i915 1.6.0 20171222 for 0000:0b:00.0 on minor 0
[    2.943893] i915 0000:0b:00.0: fb0: inteldrmfb frame buffer device

...

wjz304, 2023-08-01:
modules 里 取消勾选 i915

cjjdaq, 2023-08-01:
今天将引导重新更换为23.7.4，并使用高级功能里的降级安装，重新安装后显卡依然失败，不晓得之前升级到新版引导导致系统修改了哪些东西，我该如何才能回到正常使用dva1622本身的i915.请指教。

closed 2023-08-13 by cjjdaq