RROrg/rr #152: 升级至 23.8.0 后 DSM 卡顿严重

xyr5668, 2023-08-02:
Install ENV:
 * motherboard:
 * CPU: 8300T
 * NIC:8125

Arpl version:23.7.6 升级至23.8.0

DSM model:920+
DSM version:7.2u2

Issue:
之前正常使用23.7.6 ，能看到 nvme（此时应该是使用 powerlimit 100 ），系统反应正常
今日看到有正式修复 nvme 问题，于是升级至 23.8.0，并手动启用新增的两个 nvme 补丁，开机后系统反应迟缓，登录界面输完密码都要等 1 分多钟才能进桌面，存储管理器 10 分钟都打不开，随即重新写引导再次安装 23.8.0，不启用两个 addons，然后恢复安装 dsm 后恢复正常（此时应该依旧是使用 powerlimit 100 进行的破解）
特地过来跟你反馈一下

logs:

...

wjz304, 2023-08-05:
但是我没有出现这种情况，不确定具体事什么原因，先观察着吧

xyr5668, 2023-08-05:
> 但是我没有出现这种情况，不确定具体事什么原因，先观察着吧

今天重刷 23.8.1，第一次配置好点击启动后，偶发这个：
/opt/arpl/boot.sh: 行 231: /dev/pts/1: Permission denied
在 7 月的版本中好像也见到过一次

wjz304, 2023-08-05:
> > 但是我没有出现这种情况，不确定具体事什么原因，先观察着吧
> 
> 今天重刷 23.8.1，第一次配置好点击启动后，偶发这个： /opt/arpl/boot.sh: 行 231: /dev/pts/1: Permission denied 在 7 月的版本中好像也见到过一次

偶现，一般是 打开了web 意外终止掉了，或者意外刷新造成的，没什么影响。

wjz304, 2023-08-21:
没有复现过该问题，本issue关闭，复现请重建

closed 2023-08-21 by wjz304