RROrg/rr #155: BUG

crisfd, 2023-08-07:
Install ENV:
 * motherboard: laptop asus / laptop lenovo / gigabyte b85m-d2v
 * CPU: i3gen4 / i5gen4
 * NIC: rtl8169

Arpl version:  23.5.5 / 23.6.8 / 23.8.1

DSM model: DS918 / DS1520 / DS1621 / DS1821 etc.
DSM version: 6.2.4-25556 update 7

Issue:
Hi, just wanted to let you know that in proxmox, all mentioned DS models works with 6.2.4-25556 update7. I've tried DS918/DS1520/DS1621/DS1821, all are working out of the box. But when I try on baremetal none of them works, I've got a reinstallation  loop ("we've detected that the hard drives of your current DSxxx have been moved from a previous DSxxx and installing a newer DSM is required before you continue").
It worked once, some time ago with DS1019, but I am unable to repeat that outcome again on the same hardware.
I know it is not a priority for you, but maybe for some people who have old hardware, this 6.2.4-25556 is enough, so if and when you have some spare time to look into it, I would be grateful.
Thank you.

PS: If someone else can test it on baremetal, it would be also appreciated.
logs:

...

wjz304, 2023-08-07:
Due to different kernel versions, there are some errors with DSM6 in existing plugins and drivers,
Testing and troubleshooting require a lot of time

crisfd, 2023-08-11:
FYI: I've run some more tests on different hardware for 6.2.4-25556 update 7

arpl-i18n 23.5.5/arpl-i18n 23.8.1
DS920/DS1621/DS1621xs/DS918  			works out-of-the-box
motherboard: Intel-C236/Skylake i7-6700  
network: intel i219-lm  e1000e 

arpl-i18n 23.8.1
DS920/DS1621/DS918/DS2419/DS1621xs 		does not work no matter what network card is
motherboard: B250/i7-7700 
network: rtl 8168/8111 r8169
network: intel i219-lm  e1000e 

arpl-i18n 23.8.1
DS918/DS1819/DS2419
motherboard: B85M/i3-4150	
network: r8169					does not work

arpl-i18n 23.8.1
DS1621xs
motherboard: B85M/i3-4150	
network: r8169					works out-of-the-box

arpl-i18n 23.8.1
DS918/DS1821/DS920 (this DS has bios reset bug for this motherboard)/
motherboard: B85M/i3-4150	
network: intel i219 e1000e 			works out-of-the-box

arpl-i18n 23.8.1
DS1819 (this DS has bios reset bug for this motherboard)/
motherboard: B85M/i3-4150	
network: intel i219 e1000e 			does not work

I will run more tests and post them here if it helps.
Thank you for your work. Afaik you are the only dev that kept 6.2.4 on the build list and I really really thank you for it.

wjz304, 2023-08-11:
Actually, I have two repo to create bootloader
https://github.com/wjz304/Redpill_CustomBuild

wjz304, 2023-08-11:
> FYI: I've run some more tests on different hardware for 6.2.4-25556 update 7
>
> arpl-i18n 23.5.5/arpl-i18n 23.8.1 DS920/DS1621/DS1621xs/DS918 works out-of-the-box motherboard: Intel-C236/Skylake [...]

Get the PID&VID of the network cards, Let me see if the driver supports it

crisfd, 2023-08-12:
For B85M motherboard
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 0c)
	Subsystem: Gigabyte Technology Co., Ltd Onboard Ethernet [1458:e000]
	Kernel driver in use: r8169
	Kernel modules: r8169
03:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
	Subsystem: Intel Corporation Gigabit CT Desktop Adapter [8086:a01f]
	Kernel driver in use: e1000e
	Kernel modules: e1000e

wjz304, 2023-08-15:
https://github.com/wjz304/arpl-i18n/issues/171

wjz304, 2023-08-15:
https://github.com/wjz304/arpl-i18n/issues/5

wjz304, 2024-04-19:
RR 24.4.5 后不再支持 6.2

closed 2024-04-19 by wjz304