RROrg/rr #156: 启动界面显示的版本号有误

gexiaopeng, 2023-08-08:
我的VM虚拟机 群晖dsm  7.2  64570 update 1升级到  7.2  64570 update 3(从官方下载的synology_broadwell_3617xs.pat)，启动以后界面显示 7.2（64570u2）,但从群晖助手搜索 与 dsm控制面板来看，都是 64570 update 3 
[image]
[image]
[image]

wjz304, 2023-08-08:
该界面是读取引导信息，7.2u2 到 7.2u3 并没有引导的变化.，版本号的显示就不会被修改，简单讲 7.2u3 只是系统文件变了一点，引导还是7.2u2的。

群晖助手等 是读取的系统信息，这个该显示啥显示啥。

gexiaopeng, 2023-08-10:
> 该界面是读取引导信息，7.2u2 到 7.2u3 并没有引导的变化.，版本号的显示就不会被修改，简单讲 7.2u3 只是系统文件变了一点，引导还是7.2u2的。
> 
> 群晖助手等 是读取的系统信息，这个该显示啥显示啥。

那启动界面显示的版本应该叫 引导版本，否则让人误解。

closed 2023-08-10 by wjz304