RROrg/rr #157: r8125 timeout

tommy030314, 2023-08-08:
Install ENV:
 * motherboard:msi z690 tomahawk
 * CPU:i5-12600T
 * NIC:r8125

Arpl version:23.8.2

DSM model:x
DSM version:x

Issue: Can't get ip when configuration
works on 23.6

logs:

...
[image]
[image]

tommy030314, 2023-08-08:
[image]
[image]

tommy030314, 2023-08-08:
> `dmesg | grep r8125` to check driver load info

[image]
[image]

tommy030314, 2023-08-08:
It shows timeout but I can actually connect on ip:7681, but after configuartion, I can't connect by ip:5000

tommy030314, 2023-08-08:
> Yes, the ARPL interface displays timeout, but if the IP is obtained, it does not affect usage. Because the initialization process cannot wait endlessly for IP acquisition.
> 
> The problem in DSM is the real problem, If you have customized the MACs, please cancel the customization first, And check if the original-mac* in /mnt/p1/user-config.yml matches the MACs and order of the real network card

the mac adress matches
[image]
after this, I can't connect on both port(5000,7681)

wjz304, 2023-08-08:
https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

Because my English is very poor, I didn't write it in English,
I don't know if you can understand Google Translate

wjz304, 2023-08-08:
Between 23.6 and 23.8, both the default values of LKM and the default values of the kernel way have changed, which will have an impact.

cr4zyc4t, 2023-08-18:
try "Direct boot", it worked for me, however sometimes my nas cannot reboot or boot from shutdown state

wjz304, 2023-08-22:
https://github.com/wjz304/arpl-i18n/issues/147

marboroman, 2023-08-22:
Try this:

1. Make clean loader install on usb
2. After first boot update modules, addons and lkm via update menu
3. Choose model and version
4. Go to modules menu and deselect all
5. Select manually only r8125 and i915
6. Build the loader then boot it. 

Booting should take about 3 minutes or less. Never longer then 4 minutes for me. Reset pc if not successful and try 2-3 times.

marboroman, 2023-08-22:
Actually nothing particular helps 100%.  It suddenly starts loading successfully and then it can load 20 times without problems. And on another day it can’t be loaded in any way. You sit for several hours and do not understand what's going on.

wjz304, 2023-08-22:
there is magic 
[image]

wjz304, 2023-08-22:
There should still be an incompatible driver, perhaps some processes suddenly used it

wjz304, 2023-09-21:
No further feedback, this issue has been closed. If there are any issues, please create a new issue

closed 2023-09-21 by wjz304