RROrg/rr #162: i915

AuxXxilium, 2023-08-11:
Do you know this Script for i915 patching? A user send this to me, because the current i915 patch is not working with 10th gen in his setup.

# -*- coding: utf-8 -*-

import os, stat, shutil, string

if __name__ == '__main__':

    ko = '/usr/lib/modules/i915.ko'
    subid = b'\x86\x80\x00\x00\x92\x3E\x00\x00'
    subend = b'\x30\x82\x01\xF7\x06\x09\x2A\x86'
    isChmod = False

    id = ''
    while True:
        id = input("[I] 请输入4位十六进制显卡ID(eg:7270):")
        if len(id) == 4 and all(c in string.hexdigits for c in id):
            break
        print("[E] 输入错误, 请重新输入.")
    
    if not os.path.exists(ko):
        print("[E] 未找到 {} 文件.".format(ko))
        exit(1)

    if not os.path.exists(ko+'.bak'):
        shutil.copyfile(ko, ko+'.bak')
        print('[I] 备份 {} 到 {} '.format(ko, ko+'.bak'))

    sBak = os.stat(ko).st_mode
    if sBak & stat.S_IWUSR != stat.S_IWUSR:
        os.chmod(ko, sBak + stat.S_IWUSR)
        isChmod = True

    data = None
    with open(ko+'.bak', 'rb') as f:
        data = f.read()

    idx = data.find(subid)
    if idx != -1:
        print('[I] 替换 {} 到 {} 的 {} 位置'.format(id, ko, idx))
        data = data.replace(subid, b''.fromhex('86800000'+id[2:4]+id[0:2]+'0000'))
    else:
        print("[E] {} 中未找到 {}, 替换失败.".format(ko, '86800000923E0000'))
        exit(2)

    idx = data.find(subend)
    if idx != -1:
        print('[I] 删除 {} 的 {} 位置的校验'.format(ko, idx))
        data = data[:idx]
    else:
        print('[W] {} 中未找到校验位, 跳过.'.format(idx))

    with open(ko, 'wb') as f:
        f.write(data)
    
    if isChmod:
        os.chmod(ko, sBak)

    print('[I] 修改完成, 请重启确认, (源文件备份到{}, 如有问题请还原).'.format(ko+'.bak'))

Do you think, it will possible to port this?

wjz304, 2023-08-11:
Actually, this Python script was written by me.

wjz304, 2023-08-11:
At the time of writing it, I had not yet done the i18n project, 
Now that we have written the i915le10th addon for the shell, we can extend its support for parameter passing.

But we should add his feedback ID to i915ids for automatic processing (i915ids is not complete, the difficulty is that no one provides feedback)

AuxXxilium, 2023-08-11:
:D so it is from you.

the problem that noone gives feedback is everytime the same.

AuxXxilium, 2023-08-19:
Update: His ID is already on list but not loading. Other tests with this ID and everything is working fine. He has to use you script, all other are working without this.

wjz304, 2023-08-19:
> Update: His ID is already on list but not loading. Other tests with this ID and everything is working fine. He has to use you script, all other are working without this.

Can he provide i915.ko files of two ways?

wjz304, 2023-08-19:
I checked the logic and seemed to know the cause of the problem,

Addons modified to files in tmpRoot, but because rd contains i915, i915 in rd is loaded, and files in tmpRoot will not be loaded.

If he manually performs insmod /lib/modules/i915.ko, it can confirm this issue.

AuxXxilium, 2023-08-22:
looks like it is working. thanks

do you know why i get this:
/addons/console.sh: line 10: /usr/sbin/modprobe: not found
/addons/console.sh: line 10: /usr/sbin/modprobe: not found
/addons/console.sh: line 10: /usr/sbin/modprobe: not found
/addons/console.sh: line 10: /usr/sbin/modprobe: not found
/addons/console.sh: line 13: /usr/sbin/modprobe: not found

? or can you help me? i looked thru the files and can't find the issue.

wjz304, 2023-08-22:
> looks like it is working. thanks
>
> do you know why i get this: /addons/console.sh: line 10: /usr/sbin/modprobe: not found /addons/console.sh: line 10: /usr/sbin/modprobe: not found [...]

After checking, I probably know the reason

/usr/bin/kmod 是在 eudev 安装时被拷贝到rd中.
因为我在 https://github.com/wjz304/arpl-i18n/commit/6d43dcdba4d5cef42cab9381e46b009acf0f37d1 中调整了 addons 的加载顺序，导致 console 在 eudev 之前运行，但是此时系统中不存在 /usr/bin/kmod

wjz304, 2023-08-22:
This problem is tricky because some addons involving ko should run before eudev.

AuxXxilium, 2023-08-22:
i understand, but the issue is that the console won't start :(

wjz304, 2023-08-22:
It appears that this issue also invalidates the loading of i915

AuxXxilium, 2023-08-22:
in a vm without i915?

wjz304, 2023-08-22:
Oh no, maybe I'm overthinking
[image]

wjz304, 2023-08-22:
Either go back first.

I need to think about other possibilities

wjz304, 2023-08-22:
[e7bc07e433e50cdd9e854736d2ed46dbde75059c](https://github.com/wjz304/arpl-i18n/commit/e7bc07e433e50cdd9e854736d2ed46dbde75059c)

AuxXxilium, 2023-08-22:
i will try it, thx

wjz304, 2023-08-22:
Or, add kmod to console 
Because every addon should be independent and complete, and should not be related on other addon

AuxXxilium, 2023-08-22:
changed order is working

closed 2023-08-25 by wjz304