RROrg/rr #165: BUG

Edwardpge, 2023-08-12:
Install ENV:  
 Gigabyte ga-H97-D3 motherboard:  
 AMD PHENOM X4 965  CPU:  
 NIC:  e1000e x4

Arpl version:  latest

DSM model:  16xx+
DSM version:  7.2 xxxx

Issue:  Stuck After Booting not comes up

logs:  Can't get logs beacuse system not reposnding.. while 7.1 xxxx is working fine and set
...

wjz304, 2023-08-12:
You can enable dbgutils addon in arpl, then build, boot and wait 10 minutes, finally check /mnt/p1/logs in arpl.

Edwardpge, 2023-08-12:
> You can enable dbgutils addon in arpl, then build, boot and wait 10 minutes, finally check /mnt/p1/logs in arpl.

Okay!

Edwardpge, 2023-08-12:
> You can enable dbgutils addon in arpl, then build, boot and wait 10 minutes, finally check /mnt/p1/logs in arpl.

Nothing

Edwardpge, 2023-08-12:
No logs

Edwardpge, 2023-08-12:
[image]

wjz304, 2023-08-12:
It's likely that the kernel crashed,

Try switching to lkm

Edwardpge, 2023-08-12:
> It's likely that the kernel crashed,
> 
> Try switching to lkm

To Test or Dev

wjz304, 2023-08-12:
Give it a try, different hardware is not fixed,

However, the CPU seems too old and may not be compatible with 7.2, or try waiting for a longer time

Edwardpge, 2023-08-12:
> Give it a try, different hardware is not fixed,
> 
> However, the CPU seems too old and may not be compatible with 7.2, or try waiting for a longer time

Okay Sure

Edwardpge, 2023-08-12:
Nothing Sir, 7.1 works smoothly and fine, while 7.2 stuck after booting nothing comes up..

Edwardpge, 2023-08-12:
Any other solution?

wjz304, 2023-08-13:
No logs or serial ports, can only guess

https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

Because my English is very poor, I didn't write it in English,
I don't know if you can understand Google Translate

Edwardpge, 2023-08-13:
> No logs or serial ports, can only guess
> 
> https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM
> 
> Because my English is very poor, I didn't write it in English,
> I don't know if you can understand Google Translate

I got that i'll try it.

closed 2023-08-23 by Edwardpge