RROrg/rr #168: arpl-i18n-23.8.3 choose NIC order

Tk95100, 2023-08-13:
Hello,

Status: My system is Optiplex 7050 bare metal, DSM 7.2-update3, model DS918+, arpl-i18n-23.8.3 loader. The built-in nic detected as e1000 (intel i219-lm), PCIe x1 dual rt8125 nic. ARPL autodetect in this order:
eth0 - rt8125, eth1 - rt8125, eth2 - e1000
DSM Lan 1, Lan 2 to rt8125, and Lan 3 to e1000
Which is fine. However, DSM setting only enable for Lan 1 (rt8125 port 1) and Lan 2 (rt8125 port 2) to be wake-on-lan (WOL). PCIe NIC have no WOL feature in BIOS or active link light when system is off.

Problem: When DSM is being shutdown, Lan 3 (e1000) link light will be turn off, and WOL will not receive package on this port. The only way to make the built-in NIC back to normal is to cycle the power, or at power loss, then link light will be active again, and WOL will work.

Question: Is there a way to reverse the NIC detection order, or manually force ARPL to put e1000 to be eth0?

I've tried to modify the netif order by ssh to DSM, "vi /etc.defaults/synoinfo.conf" command, edit to [netif_seq=”2 0 1 3 4 5 6 7”]. However, after restart the system, the configuration file reverse back to original orders.

Thank you!

wjz304, 2023-08-13:
[image]
Modify in the arpl -- synoinfo menu

Tk95100, 2023-08-13:
Thank you for the quick response.

I've tried:
netif_seq: 2 0 1 3 4 5 6 7
netif_seq: 1 0 2 3 4 5 6 7
However, the built-in nic (e1000e) still being detect last, as eth2. DSM still recognize it as Lan 3.

Is there another way to reverse or reserve the NIC detection priority to the built-in (e1000e) to be eth0?

Thank you.

wjz304, 2023-08-13:
I have been quite busy lately and don't have time to study. It may take some days.

Tk95100, 2023-08-14:
Sure no problem.  I will do some research, if I find a solution, will share it here.  I appreciate your works.

lazosweb, 2023-08-28:
@wjz304 I have a similar problem with igb and bnxt_en. If I power cycle the VM, bnxt_en will be eth0 and igb will be eth1 and that brakes my HA cluster. In order to fix it I have to stop and start the VM again and the order will change once more... This is on the passive server (arpl 23.8.9) of the HA cluster.

On my active server (arpl 23.7.1) which I have igb as eth0 and ixgbe as eth1 it does not happen.

There is no way to fix it?

wjz304, 2023-08-28:
因为mac地址是arpl传下去的，所以如果要调整顺序需要保持 arpl 和dsm严格的一致。
如果一些人有 usb网卡 这会导致 mac设置错乱，导致dsm无法获取到ip.

wjz304, 2023-08-28:
给你一个方法 自己改吧，
```
ip link set eth0 down
ip link set eth1 down
ip link set eth0 name eth99
ip link set eth1 name eth0
ip link set eth99 name eth1
ip link set eth0 up
ip link set eth1 up
/etc/rc.network start
```

wjz304, 2023-08-28:
可以写到计划任务里

lazosweb, 2023-08-28:
Ok, thank u.

wjz304, 2023-08-28:
/etc/rc.network restart

wjz304, 2023-10-12:
https://github.com/wjz304/arpl-i18n/releases/tag/23.10.2

Tk95100, 2023-10-13:
> https://github.com/wjz304/arpl-i18n/releases/tag/23.10.2

Thank You.  I will be testing this this release weekend.

wjz304, 2023-10-17:
No feedback, this issue close

closed 2023-10-17 by wjz304

Tk95100, 2023-10-23:
> https://github.com/wjz304/arpl-i18n/releases/tag/23.10.2

Hello Ing,

I was bit busy with life, just able to try out the "sortnetif" feature this weekend.  I rebuilt a brand new system to test it out.
System:
i5 6500, intel 270q chipset, 8gb ram, sata 500ssd storage, 16gb flash boot drive to usb 2.0 port
On board intel e1000e
Pcie rtl8125 chipset with dual rj45
DSM 7.2.1 (69057) - update 1
ARPL 23.10.3, addons 23.10.3, modules 23.10.0, LKMs 23.10.0

boot to arpl and dsm by default assign ip as order:
eth0/Lan1 detect as rtl8125 - nic1
eth1/Lan2 detect as rtl8125 - nic2
eth2/Lan3 detect as e1000e - built-in nic

I boot to arpl menu.  launch menu.sh.
select a - Addons menu -> a - Add an addon -> sortnetif
dialog open asking for optional params to addon (mac1,mac2,mac3)
ma:c1:e1:00:0e:xx, ma:c2:rt:81:25:x0, ma:c3:rt:81:25:x1
save, exit and reboot

After the config the sortnetif addon, ARPL still detect the order as:
eth0 detect as rtl8125 - nic1
eth1 detect as rtl8125 - nic2
eth2 detect as e1000e - built-in nic

launching DSM, script still shows:
eth0 detect as rtl8125 - nic1
eth1 detect as rtl8125 - nic2
eth2 detect as e1000e - built-in nic
But once the screen goes blank, DSM do not give IP to any NICs.  I have to hard reset at this point.

Re-launching ARPL and remove sortnetif addon, everything goes back to default.  DSM will launch normal with ip assign to:
eth0/Lan1 detect as rtl8125 - nic1
eth1/Lan2 detect as rtl8125 - nic2
eth2/Lan3 detect as e1000e - built-in nic

let me know if I doing something wrong. Thanks

wjz304, 2023-10-23:
看一下 user_config.yml 中 addons 部分

Tk95100, 2023-10-29:
> 看一下 user_config.yml 中 addons 部分 [image] 是这样的格式吗？

Today I've edited user_config.yml to add sortnetif:"e1000eMAC,rtl8125MAC1,rtl8125MAC2" like your instruction.  However, it does not change, the original order remains the same.  I've verified that the MAC ID is correct with "ip link" command.  both is arpl, pcie rtl8125 2x nic will be 1st to register as eth0 & eth1 then built in e1000e nic will be on eth2.  The same order is in DSM 7.2.1-update1 rtl8125 2x nic will be on LAN1 & LAN2, e1000e nic will be on LAN3.

wjz304, 2023-10-29:
只能取日志看一下了
https://github.com/wjz304/rr/issues/173