RROrg/rr #172: 蜗牛星际A款无法进入5000端口安装界面

Meteor2017, 2023-08-15:
Install ENV:  
 * motherboard:  蜗牛星际A款
 * CPU:  J1900
 * NIC:  I211

Arpl version:  23.8.5

DSM model:  DS918+
DSM version:  7.2

Issue:  原来是用的6.23，最近琢磨升级7.x，通过arpl-i18n23.8.5版本引导进行黑群晖7.2的安装（7.1也试过，同样的问题）。引导安装在蜗牛星际自带的16g sata固态中（非u盘引导）。在7681端口引导设置界面进行配置启动引导后，在路由器后台看到机器ip在线，且处于可ping通的状态，尝试过等待20多分钟，可却始终无法进入5000端口的群晖安装界面，群晖链接助手同样无法扫描到。在自己琢磨了一整天，外加群内两个热心老哥远程帮忙排查的情况下，问题仍未解决。特来此提出issue，请求帮助。

[image]
[image]
[image]
[image]
[image]

wjz304, 2023-08-15:
https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

Meteor2017, 2023-08-16:
> https://github.com/wjz304/arpl-i18n/wiki/%E5%85%B3%E4%BA%8E-%22%E5%90%AF%E5%8A%A8%22-%E5%90%8E%E6%97%A0%E6%B3%95%E6%89%BE%E5%88%B0DSM

尝试了里边的办法，依旧无法解决，ip一直在线，可ping通，但就是进不去5000端口安装界面。测试原版的arpl没有问题。但是咱们这个版本无论安装7.0还是7.1，都是一个情况

wjz304, 2023-08-16:
如果真尝试了，就不可能ping通

a8711531, 2023-08-20:
这个很大概率是kernel崩溃掉了，导致搜索不到ip请通过这个教程解决https://www.mi-d.cn/7303

zkname, 2023-09-16:
蜗牛C也不行。。。。

ilovelam, 2023-09-16:
惠普 gen10 也进不了

wjz304, 2023-09-16:
https://github.com/wjz304/arpl-i18n/issues/219

closed 2023-09-16 by wjz304

GuoliangCui, 2023-09-20:
> 蜗牛C也不行，启动立马网卡等就灭了，光标都卡没了。

同款同问题 求解决