RROrg/rr #176: Synology Surveillance Station网页版以及桌面端摄像头无画面，手机app有画面

cjameslynn, 2023-08-18:
Install ENV:  
 * motherboard:  技嘉H310M
 * CPU:  G5400
 * NIC:  r8169，r8125

Arpl version:  23.8.7

DSM model:  DVA3221
DSM version:  7.2

Issue:  
Synology Surveillance Station网页版以及桌面端摄像头无画面，手机app有画面

logs:  
强制重新安装DSM，问题依旧。
...

wjz304, 2023-08-21:
Need to drag the camera to the window?
The current bootloader doesn't do any control over this app.

OrpheeGT, 2023-08-28:
[image]

Surveillance station is totally unrelated to ARPL release/version.

It works perfectly fine, and it always worked for me, no matter the loader, and the hardware.

Try first with more conventional SS with like DS3622xs+ in your case.