RROrg/rr #178: Error Message Occurred : Unrecognized firmware version, please update the drive database

malgirlim, 2023-08-20:
Install ENV:  
 * motherboard:  ASUS H110M-K
 * CPU:  Intel Core i3-6100
 * NIC:  Realtek 8168
 * Host OS : Windows 11 Pro
 
Running in a Type2 virtualized environment. (VMware Workstation 17 Player, 17.0.0 build-20800274)
* CPU : 4Core
* MEM : 8GB
* HDD :  SATA Toshiba DT01ACA300(3TB) 2EA (Physical disk passthrough connection)

Arpl version:  23.8.7

DSM model:  DS918+
DSM version:  7.2-64570 Update 3

Issue:  
[image]
[image]
[image]

An error occurs that the disk firmware information cannot be retrieved like the screenshot above.
There seems to be no problem with the system operating, but there are periodic error messages in the notification window.
It's not accurate, but it's been happening since we recently updated ARPL to 23.8.7 version.
The previous version was 23.6.8.
When I try to update the database according to the instructions in the system, it shows that it has already been updated.
Is there any other information I should provide?
Thank you.

logs:  

...

wjz304, 2023-08-21:
Try attaching the hdddb addon

irishj, 2023-08-22:
I have a similar issue with three SSD's which I added to my system.
The "hdddb" addon has been selected, but I still see the message in storage manager, I don't know if that's expected or not.
I don't receive messages in the notification centre since adding the "hdddb" addon, so perhaps the message in storage manager is expected.

[image]

wjz304, 2023-09-15:
test the lastest addons

wjz304, 2023-09-21:
No further feedback, this issue has been closed. If there are any issues, please create a new issue

closed 2023-09-21 by wjz304