RROrg/rr #179: Ether 0 not connecting

BleepHazard, 2023-08-21:
Install ENV:  
 * Toshiba satellite c650 laptop
 * CPU:  Intel p6000
 * NIC:  Qualcomm Atheros AR8152

Arpl version:  23.8.7

Issue: 
Even if the laptop is connected to the net, the launcher is not capable to find the IP and so, to connect to the url to complete the instalation. 

I have read that my NIC has compatibility problems.
Is there something I can make to solve the problem or must be impossible to make a nas in this laptop?

I'm sorry about my english. 

Thank you all, for your job.

wjz304, 2023-08-21:
当前 dsm 没有wifi网卡的驱动, 也没有wifi 的控制程序，你需要有线网卡。

BleepHazard, 2023-08-21:
I think I don't have explained properly.
The laptop is already connected with cable to the net

wjz304, 2023-08-21:
Oh, sorry, I briefly searched for xxx network card, and mistakenly thought it was a wireless network card

wjz304, 2023-08-21:
Can you get the pid and vid of the NIC?
```shell
lspci -nn
```

BleepHazard, 2023-08-21:
PCI\VEN_1969&DEV_2060&SUBSYS_FF1E1179&REV_C1
PCI\VEN_1969&DEV_2060&SUBSYS_FF1E1179
PCI\VEN_1969&DEV_2060&CC_020000
PCI\VEN_1969&DEV_2060&CC_0200

Thank you for your time and effort

wjz304, 2023-08-22:
[image]
Currently, ARPL includes this driver,
Execute under arpl
```
Lsmod
Ifconfig
ls -ld /sys/class/net/*/device/driver
```

BleepHazard, 2023-08-22:
Thank You very much wjz304.
I have already crated my NAS.
I think the new version of arpl help me cause I was trying to create a 2422+ dsm.

closed 2023-08-23 by wjz304