RROrg/rr #180: crash arpl-i18n 23.8.8

wahabe, 2023-08-21:
Hello

arpl-i18n 23.8.8

crash when adding r8169

[image]
[image]

wjz304, 2023-08-21:
https://github.com/wjz304/arpl-i18n/actions/runs/5928852444

Download this attachment, unzip it, manually update the update.zip inside and try it out.
Or brush the img inside directly.

wahabe, 2023-08-22:
Same problem, waiting for the next release
Thank you
I have a special computer for experiments and a lot of parts. If you need to test it and send reports, tell me

wjz304, 2023-08-22:
https://github.com/wjz304/arpl-i18n/actions/runs/5935479003
Can you test it again?
Expected to complete compilation in half an hour

wahabe, 2023-08-22:
yes

wahabe, 2023-08-22:
Great job, works without problems

[image]
[image]

wjz304, 2023-08-22:
Can you help me test it again?

I just used the exclusion method to disable iov and wireless, but I'm not sure which one of them caused the problem, so I opened iov testing again.
https://github.com/wjz304/arpl-i18n/actions/runs/5935943428

wahabe, 2023-08-22:
No problems fast and smooth

[image]

wjz304, 2023-08-22:
OK, new version will be released later,
At that time, you can update it directly in the menu

wjz304, 2023-08-22:
(I have deleted your comments containing both Mac and SN)

wahabe, 2023-08-22:
Don't worry all the pieces are for testing
I am really happy to help you

wahabe, 2023-08-22:
I own a Synology E10G18-T2

QL41102 chip

Is it possible to work DS920+

wjz304, 2023-08-22:
Get the pid&vid of the NIC and take a look

wahabe, 2023-08-22:
QLogic FastLinQ QL41000 Series Controller (VBD Client) --------------------

 [General Information]
  Network Card:                           QLogic FastLinQ QL41000 Series Controller (VBD Client)
  Vendor Description:                     QLogic FastLinQ Ethernet Adapter (VBD Client)
  
 [Capabilities]
  Maximum Link Speed:                     10000 Mbps
  Transmit Buffer Size:                   2271000 Bytes
  Receive Buffer Size:                    6813000 Bytes
  Hardware ID:                            PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02
 [Driver Information]
  Driver Manufacturer:                    Cavium, Inc.
  Driver Description:                     QLogic FastLinQ QL41000 Series Controller
  Driver Provider:                        Microsoft
  Driver Version:                         8.33.20.103
  Driver Date:                            24-Apr-2018
  DeviceInstanceId                        PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02\4&1A972E93&0&0009
  Location Paths                          PCIROOT(0)#PCI(0101)#PCI(0000)

QLogic FastLinQ QL41000 Series Controller (VBD Client) --------------------

 [General Information]
  Network Card:                           QLogic FastLinQ QL41000 Series Controller (VBD Client)
  Vendor Description:                     QLogic FastLinQ Ethernet Adapter (VBD Client)
 
 [Capabilities]
  Maximum Link Speed:                     10000 Mbps
  Transmit Buffer Size:                   2271000 Bytes
  Receive Buffer Size:                    6813000 Bytes
  Hardware ID:                            PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02
 [Driver Information]
  Driver Manufacturer:                    Cavium, Inc.
  Driver Description:                     QLogic FastLinQ QL41000 Series Controller
  Driver Provider:                        Microsoft
  Driver Version:                         8.33.20.103
  Driver Date:                            24-Apr-2018
  DeviceInstanceId                        PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02\4&1A972E93&0&0109
  Location Paths                          PCIROOT(0)#PCI(0101)#PCI(0001)

wjz304, 2023-08-22:
[image]
At present, these architectures have the driver

wjz304, 2023-08-22:
I have updated to 23.8.9

wahabe, 2023-08-22:
geminilake Not in the group :(

wjz304, 2023-08-22:
> geminilake Not in the group :(

Well, this is another topic, create a new issue, I'll see if I can add it.

wahabe, 2023-08-22:
Don't hesitate if you need anything

wjz304, 2023-08-22:
The issue has been fixed, closed

closed 2023-08-22 by wjz304