RROrg/rr #182: QLogic FastLinQ QL41000 Series Controller

wjz304, 2023-08-22:
QLogic FastLinQ QL41000 Series Controller (VBD Client) --------------------

 [General Information]
  Network Card:                           QLogic FastLinQ QL41000 Series Controller (VBD Client)
  Vendor Description:                     QLogic FastLinQ Ethernet Adapter (VBD Client)
  
 [Capabilities]
  Maximum Link Speed:                     10000 Mbps
  Transmit Buffer Size:                   2271000 Bytes
  Receive Buffer Size:                    6813000 Bytes
  Hardware ID:                            PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02
 [Driver Information]
  Driver Manufacturer:                    Cavium, Inc.
  Driver Description:                     QLogic FastLinQ QL41000 Series Controller
  Driver Provider:                        Microsoft
  Driver Version:                         8.33.20.103
  Driver Date:                            24-Apr-2018
  DeviceInstanceId                        PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02\4&1A972E93&0&0009
  Location Paths                          PCIROOT(0)#PCI(0101)#PCI(0000)

QLogic FastLinQ QL41000 Series Controller (VBD Client) --------------------

 [General Information]
  Network Card:                           QLogic FastLinQ QL41000 Series Controller (VBD Client)
  Vendor Description:                     QLogic FastLinQ Ethernet Adapter (VBD Client)
 
 [Capabilities]
  Maximum Link Speed:                     10000 Mbps
  Transmit Buffer Size:                   2271000 Bytes
  Receive Buffer Size:                    6813000 Bytes
  Hardware ID:                            PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02
 [Driver Information]
  Driver Manufacturer:                    Cavium, Inc.
  Driver Description:                     QLogic FastLinQ QL41000 Series Controller
  Driver Provider:                        Microsoft
  Driver Version:                         8.33.20.103
  Driver Date:                            24-Apr-2018
  DeviceInstanceId                        PCI\VEN_1077&DEV_8070&SUBSYS_10027053&REV_02\4&1A972E93&0&0109
  Location Paths                          PCIROOT(0)#PCI(0101)#PCI(0001)

_Originally posted by @wahabe in https://github.com/wjz304/arpl-i18n/issues/180#issuecomment-1687820451_

wjz304, 2023-08-22:
[image]

wjz304, 2023-08-22:
@wahabe
Track here

wahabe, 2023-08-23:
Hello

i have installed DS920 on BareMetal 
i7-7700K Processor
6TB
8xWD red in RAID 5

2x TP-Link r8169

I'm waiting if possible to add QLogic FastLinQ QL41000

or buy TP-Link 10GB PCIe

wjz304, 2023-08-23:
It takes some time, and it is not necessarily possible to add

wahabe, 2023-08-23:
Thank you my friend

wjz304, 2023-08-23:
https://github.com/wjz304/arpl-modules/actions/runs/5951687792

Download the attachment, upload the local package in the update menu,
Select all modules in the modules menu, and rebuild and boot and try

wahabe, 2023-08-23:
I will try now

wahabe, 2023-08-23:
does not work

wahabe, 2023-08-23:
drivers for linux
https://www.marvell.com/support/downloads.html#
[image]

wjz304, 2023-08-23:
[qede.ko.gz](https://github.com/wjz304/arpl-i18n/files/12420227/qede.ko.gz)

wjz304, 2023-08-23:
下载解压， 在模块菜单中 上传外部模块
之后重新编译启动

wahabe, 2023-08-23:
ok

wahabe, 2023-08-23:
not work

[image]

wahabe, 2023-08-23:
Do you mean the interface?

[image]

wjz304, 2023-08-23:
> Do you mean the interface?
> 
> [image]

yes.

wjz304, 2023-08-23:
Because e1000e is working, can only one network card be seen in the DSM system?

wahabe, 2023-08-23:
arpl work ok 
dsm not work 
[image]
[image]

wahabe, 2023-08-23:
root@NIGHT:~# lsmod
Module                  Size  Used by
fuse                   87662  1
vfat                   10511  0
fat                    55795  1 vfat
vhost_scsi             28995  1
vhost                  30822  1 vhost_scsi
tcm_loop               15357  1
iscsi_target_mod      272569  1
target_core_user       27400  0
target_core_ep         50980  2
target_core_multi_file    29151  1
target_core_file       59354  1
target_core_iblock     17374  1
target_core_mod       893592  22 target_core_iblock,target_core_multi_file,vhost                                                                                                                                                             ,iscsi_target_mod,target_core_ep,target_core_file,target_core_user,vhost_scsi,tc                                                                                                                                                             m_loop
syno_extent_pool     1385313  0
rodsp_ep               91947  3 target_core_multi_file,syno_extent_pool,target_c                                                                                                                                                             ore_file
udf                    80886  0
isofs                  33143  0
synoacl_vfs            16969  1
btrfs                1259738  1
ecryptfs               85159  1 btrfs
zstd_decompress        60974  1 btrfs
zstd_compress         154833  1 btrfs
xxhash                  4611  2 zstd_compress,zstd_decompress
xor                    10808  1 btrfs
raid6_pq               97268  1 btrfs
8021q                  17933  0
acpi_cpufreq            6651  0
leds_lp3943             4918  0
aesni_intel           175290  0
glue_helper             4085  1 aesni_intel
lrw                     3744  1 aesni_intel
gf128mul                5752  1 lrw
ablk_helper             2076  1 aesni_intel
geminilake_synobios    82923  0
usblp                  11290  0
r8168                 540240  0
dm_crypt               18450  0
sg                     26471  0
dm_snapshot            29850  0
dm_bufio               14399  1 dm_snapshot
crc_itu_t               1595  1 udf
crc_ccitt               1595  0
psnap                   2004  0
p8022                   1307  0
llc                     3569  2 p8022,psnap
hfsplus                92606  0
md4                     3889  0
hmac                    3089  0
sit                    15976  0
tunnel4                 2389  1 sit
ipv6                  318022  61 sit
flashcache_syno       270416  1
flashcache             81203  0
syno_flashcache_control     1231  2 flashcache_syno,flashcache
dm_mod                 90426  12 flashcache_syno,dm_bufio,dm_crypt,flashcache,dm                                                                                                                                                             _snapshot
zram                   18699  6
arc4                    2168  0
crc32c_intel           13333  1
cryptd                  8677  2 aesni_intel,ablk_helper
ecb                     2241  0
aes_x86_64              7567  1 aesni_intel
authenc                 4209  0
des_generic            16285  0
ansi_cprng              4020  0
cts                     4520  0
md5                     2705  0
cbc                     2776  0
cpufreq_powersave       1190  0
[... 230 lines cut ...]
md4.ko                           x_tables.ko
md5.ko                           xt_addrtype.ko
mdio.ko                          xt_conntrack.ko
megaraid_mbox.ko                 xt_geoip.ko
megaraid_mm.ko                   xt_iprange.ko
megaraid_sas.ko                  xt_ipvs.ko
mii.ko                           xt_limit.ko
mlx4_core.ko                     xt_LOG.ko
mlx4_en.ko                       xt_mac.ko
mlx5_core.ko                     xt_mark.ko
mlxsw_core.ko                    xt_multiport.ko
mlxsw_pci.ko                     xt_nat.ko
mmc_block.ko                     xt_NFQUEUE.ko
mmc_core.ko                      xt_policy.ko
modules.alias                    xt_recent.ko
modules.alias.bin                xt_REDIRECT.ko
modules.builtin.alias.bin        xt_set.ko
modules.builtin.bin              xt_state.ko
modules.dep                      xt_TCPMSS.ko
modules.dep.bin                  xt_tcpudp.ko
modules.devname                  xxhash.ko
modules.softdep                  zram.ko
modules.symbols                  zstd_compress.ko
modules.symbols.bin              zstd_decompress.ko
root@NIGHT:~# ls -ld /sys/class/net/*/device/driver

wjz304, 2023-08-23:
ls -ld /sys/class/net/*/device/driver  
?

wjz304, 2023-08-23:
need enter ?

wahabe, 2023-08-23:
root@NIGHT:~# ls -ld /sys/class/net/*/device/driver
lrwxrwxrwx 1 root root 0 Aug 23 18:19 /sys/class/net/eth0/device/driver -> ../../../bus/pci/drivers/e1000e
root@NIGHT:~# ls -ld /sys/class/net/*/device/driver
lrwxrwxrwx 1 root root 0 Aug 23 18:19 /sys/class/net/eth0/device/driver -> ../../../bus/pci/drivers/e1000e

wjz304, 2023-08-23:
dmesg | grep qede

wahabe, 2023-08-23:
root@NIGHT:~# dmesg | grep qede
[    3.114655] qede_init: QLogic FastLinQ 4xxxx Ethernet Driver qede 8.4.0.0
[    3.126387] qede 0000:02:00.0: Direct firmware load for qed/qed_init_values_z                                                                                                                                                             ipped-8.4.2.0.bin failed with error -2
[    3.128657] qede 0000:02:00.0: Falling back to user helper
[   49.095562] qede: probe of 0000:02:00.0 failed with error -11
[   49.099417] qede 0000:02:00.1: Direct firmware load for qed/qed_init_values_z                                                                                                                                                             ipped-8.4.2.0.bin failed with error -2
[   49.100545] qede 0000:02:00.1: Falling back to user helper
[   49.165797] qede: probe of 0000:02:00.1 failed with error -11
[   49.218737]  aesni_intel glue_helper lrw gf128mul ablk_helper geminilake_syno                                                                                                                                                             bios(PO) usblp r8168(O) dm_crypt sg dm_snapshot dm_bufio crc_itu_t crc_ccitt psn                                                                                                                                                             ap p8022 llc hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O)                                                                                                                                                              syno_flashcache_control(O) dm_mod zram arc4 crc32c_intel cryptd ecb aes_x86_64 a                                                                                                                                                             uthenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance                                                                                                                                                              cpufreq_stats vxlan ip6_udp_tunnel udp_tunnel ip_tunnel loop sha256_generic syno                                                                                                                                                             rbd(O) synofsbd(O) hid_generic uas usb_storage usbhid hid kvm irqbypass i915 e10                                                                                                                                                             00e(OE) qede(OE) qed(OE) i2c_algo_bit iosf_mbi drm_kms_helper syscopyarea sysfil                                                                                                                                                             lrect sysimgblt fb_sys_fops drm drm_panel_orientation_quirks cfbfillrect cfbimgb                                                                                                                                                             lt cfbcopyarea etxhci_hcd fb fbdev(OE) thermal(OE) rtc_cmos(OE) video backlight                                                                                                                                                              button processor
root@NIGHT:~#

wjz304, 2023-08-23:
Okay, I need to study this error. It'll take some time. That's it for today.

wahabe, 2023-08-23:
Yes, thank you

wjz304, 2023-08-24:
https://github.com/wjz304/arpl-modules/actions/runs/5968986083

wjz304, 2023-08-25:
There should be no problem, I released

wjz304, 2023-08-25:
https://github.com/wjz304/arpl-i18n/releases/tag/23.8.10

closed 2023-08-25 by wjz304

wahabe, 2023-08-25:
does not work

wahabe, 2023-08-25:
[image]

wjz304, 2023-08-25:
https://github.com/wjz304/arpl-modules/actions/runs/5977301629

wahabe, 2023-08-25:
ok

wahabe, 2023-08-25:
Status
Failure

wahabe, 2023-08-25:
[image]

Finally, I am very happy. I will not forget this help

thank you my friend

wjz304, 2023-08-25:
Okay, I need to release the modules