RROrg/rr #183: 请问为什么mellanox 网卡添加模块后仍然无法识别到

ixgbevf, 2023-08-22:
Install ENV:  
 * motherboard:  krpa-u16
 * CPU:  7302p
 * NIC:  cx4421a

Arpl version:  23.8.9

DSM model:  ds3617xs
DSM version:  7.2

Issue:  添加模块后仍然无法识别到网卡

logs:  

...

wjz304, 2023-08-22:
有证据吗

ixgbevf, 2023-08-28:
这个问题我已经解决了。
就是字面意思，勾选mlx5 core模块编译后系统里认不出网卡。但是如果我使用之前巴西人用的那个模块（好像叫啥picoext的扩展）就可以认出网卡。应该是ko文件的问题。

18702540384, 2023-09-09:
请问picoext这个模块在哪里添加啊，找不到

ixgbevf, 2023-09-09:
> 请问picoext这个模块在哪里添加啊，找不到

https://github.com/pocopico/rp-ext

18702540384, 2023-09-10:
没有picoext这个扩展

18702540384, 2023-09-12:
用原版的可以驱动mcx4121a网卡

colinhd8, 2023-09-15:
有同样的问题。具体如下：
1、启动到arpl时，可以看到网卡：
[image]
dmesg也是正常的：
[image]

2、进dsm系统后，没有出现对应的接口。然后dmesg中有这个错误：
[image]
上面第3行的错误应该是问题的根源。

3、我试了这个：[https://github.com/pocopico/tinycore-redpill](https://github.com/pocopico/tinycore-redpill)，也是一模一样的问题。

4、但是这个却可正常识别：[https://github.com/PeterSuh-Q3/tinycore-redpill](https://github.com/PeterSuh-Q3/tinycore-redpill)

colinhd8, 2023-09-15:
> 用原版的可以驱动mcx4121a网卡

请教下是如何操作的？谢谢

colinhd8, 2023-09-15:
> 这个问题我已经解决了。 就是字面意思，勾选mlx5 core模块编译后系统里认不出网卡。但是如果我使用之前巴西人用的那个模块（好像叫啥picoext的扩展）就可以认出网卡。应该是ko文件的问题。

@ixgbevf 请问下你是在编译时加了新模块是吗？可否大概说下操作步骤？谢谢。

18702540384, 2023-09-15:
https://github.com/pocopico/4.4.180plus-apollolake
到这里下载网卡驱动，然后在arpl模组选项里面有加载外部模组选项，选中你下载的ko文件，然后重新编译引导就好了

colinhd8, 2023-09-15:
> https://github.com/pocopico/4.4.180plus-apollolake 到这里下载网卡驱动，然后在arpl模组选项里面有加载外部模组选项，选中你下载的ko文件，然后重新编译引导就好了

感谢，我先试试。

colinhd8, 2023-09-15:
> https://github.com/pocopico/4.4.180plus-apollolake 到这里下载网卡驱动，然后在arpl模组选项里面有加载外部模组选项，选中你下载的ko文件，然后重新编译引导就好了
@18702540384 这个搞定了，感谢。
我还得看下为啥connectx-2出不来。。。

18702540384, 2023-09-15:
是不是驱动用错了，我是mcx4121a,驱动是mlx5_core

colinhd8, 2023-09-15:
> 是不是驱动用错了，我是mcx4121a,驱动是mlx5_core

我有两张卡，一张是和你的一样，现在按你的方法解决了。还有一张卡是connectx-2，这个在arpl里面就出不来，我还得看看什么原因。当然，这个驱动应该是用mlx4_en, mlx4_core

wjz304, 2023-09-15:
https://github.com/wjz304/arpl-modules/releases/tag/23.9.2
plz test it

18702540384, 2023-09-16:
有人测试行不行

colinhd8, 2023-09-18:
> https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it
@wjz304 @18702540384 我上周五拉的最新代码编译的arpl，看里面的版本号就是23.9.2，但试下来还是和上面一样的现象。

另外，我有一个发现，就是上面起作用的貌似不是https://github.com/pocopico/4.4.180plus-apollolake这里的驱动。
而是在arpl中取消mlx_core这一模块。取消后，不管你有没有上传上面的驱动，在dsm中都是可以正常识别的。一旦勾选了自带的模块，则进到dsm后就看不到网卡。

当然，我用的是3622xs+，它貌似是自带驱动的。

另外，上传自已找的驱动后，如何验证它确实加进去了，比如说在user_config里面也看不出有个模块是自己上传的。

wjz304, 2023-09-18:
> @wjz304 @18702540384 我上周五拉的最新代码编译的arpl，看里面的版本号就是23.9.2，但试下来还是和上面一样的现象。 [...]

arpl 23.9.2 ≠ modules 23.9.2

colinhd8, 2023-09-18:
> > > https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it
> > 
> > 
> > @wjz304 @18702540384 我上周五拉的最新代码编译的arpl，看里面的版本号就是23.9.2，但试下来还是和上面一样的现象。
> > 另外，我有一个发现，就是上面起作用的貌似不是[https://github.com/pocopico/4.4.180plus-apollolake这里的驱动。](https://github.com/pocopico/4.4.180plus-apollolake%E8%BF%99%E9%87%8C%E7%9A%84%E9%A9%B1%E5%8A%A8%E3%80%82) 而是在arpl中取消mlx_core这一模块。取消后，不管你有没有上传上面的驱动，在dsm中都是可以正常识别的。一旦勾选了自带的模块，则进到dsm后就看不到网卡。
> > 当然，我用的是3622xs+，它貌似是自带驱动的。
> > 另外，上传自已找的驱动后，如何验证它确实加进去了，比如说在user_config里面也看不出有个模块是自己上传的。
> 
> 1. arpl 23.9.2 ≠ modules 23.9.2
> 2. 是的，有几个架构自带mellanox的驱动。
> 3. 自己上传的目前没有记录，
确实，我刚看了提交记录，是16号提交的，我今天晚些试试看。

eschcn, 2023-09-18:
> @wjz304 @18702540384 我上周五拉的最新代码编译的arpl，看里面的版本号就是23.9.2，但试下来还是和上面一样的现象。 [...]

感谢分享，为了这个事折腾了好几天，经群友提醒才知道这么一茬，一会试试去

colinhd8, 2023-09-18:
> https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it

@eschcn 你可以顺便也试下这个。

eschcn, 2023-09-18:
> 1. arpl 23.9.2 ≠ modules 23.9.2
> 2. 是的，有几个架构自带mellanox的驱动。
> 3. 自己上传的目前没有记录，

在

> 1. arpl 23.9.2 ≠ modules 23.9.2
> 2. 是的，有几个架构自带mellanox的驱动。
> 3. 自己上传的目前没有记录，

更新23.9.3的arpl和23.9.2的mod，然后重新编译了引导，进群晖还是没识别到

eschcn, 2023-09-18:
> > https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it
> 
> @wjz304 我刚刚试了。按以下步骤： 1、还是用的原来的我自编译的arpl 2、menu.sh里面在Modules menu里面选择Select loaded modules 3、看一眼show selected modules，确保mlx5_core在列表中 4、Build the loader 5、强制重装dsm 6、进dsm后正常识别网卡
> 
> 所以是不是在build loader过程中，自动用了最新的module？

我是用的i18n版本的23.9.3的arpl和23.9.2的mod，按照你2-5的步骤，进系统以后还是不能识别，不过在之前在模块选择界面取消mlx5_core这个模块则可以正常识别。
我用的是3221

colinhd8, 2023-09-19:
> > > https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it
> > 
> > 
> > @wjz304 我刚刚试了。按以下步骤： 1、还是用的原来的我自编译的arpl 2、menu.sh里面在Modules menu里面选择Select loaded modules 3、看一眼show selected modules，确保mlx5_core在列表中 4、Build the loader 5、强制重装dsm 6、进dsm后正常识别网卡
> > 所以是不是在build loader过程中，自动用了最新的module？
> 
> 我是用的i18n版本的23.9.3的arpl和23.9.2的mod，按照你2-5的步骤，进系统以后还是不能识别，不过这么干之前取消mlx5_core这个模块则可以正常识别。 我用的是3221

有点奇怪，我的好像不用取消也可以了。你是怎么用23.9.2的mod的？

eschcn, 2023-09-19:
> 有点奇怪，我的好像不用取消也可以了。你是怎么用23.9.2的mod的？

默认设置啊，我没加载外部MOD

eschcn, 2023-09-19:
> 有点奇怪，我的好像不用取消也可以了。你是怎么用23.9.2的mod的？

我又虚拟了一台920+,我之前用过的两种方法都试了下，都打不开DSM的:5000网页端

wjz304, 2023-09-21:
> > https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it
> 
> @wjz304 我刚刚试了。按以下步骤： 1、还是用的原来的我自编译的arpl 2、menu.sh里面在Modules menu里面选择Select loaded modules 3、看一眼show selected modules，确保mlx5_core在列表中 4、Build the loader 5、强制重装dsm 6、进dsm后正常识别网卡
> 
> 所以是不是在build loader过程中，自动用了最新的module？

是，modules中 勾选了就会用，不勾选就会用系统自带的（官方）

colinhd8, 2023-09-21:
> 是，modules中 勾选了就会用，不勾选就会用系统自带的（官方）

@wjz304 这么看来的话，改版后的module已经解决问题了。谢谢。

wjz304, 2023-09-21:
> @wjz304 这么看来的话，改版后的module已经解决问题了。谢谢。

那我关了，其他人有日志在新建吧。

closed 2023-09-21 by wjz304

eschcn, 2023-09-21:
> @wjz304 这么看来的话，改版后的module已经解决问题了。谢谢。

并没有吧，至少我用920+是不行的不管是否勾选都无法联网

eschcn, 2023-09-21:
> 那我关了，其他人有日志在新建吧。

我新建了920+还是不行的，用的最新的arpl和mod，不管是否勾选都一样，怎么样提供日志呢

eschcn, 2023-09-21:
> > > > > > > https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it
> > > > > > 
> > > > > > 
> > > > > > @wjz304 我刚刚试了。按以下步骤： 1、还是用的原来的我自编译的arpl 2、menu.sh里面在Modules menu里面选择Select loaded modules 3、看一眼show selected modules，确保mlx5_core在列表中 4、Build the loader 5、强制重装dsm 6、进dsm后正常识别网卡
> > > > > > 所以是不是在build loader过程中，自动用了最新的module？
> > > > > 
> > > > > 
> > > > > 是，modules中 勾选了就会用，不勾选就会用系统自带的（官方）
> > > > 
> > > > 
> > > > @wjz304 这么看来的话，改版后的module已经解决问题了。谢谢。
> > > 
> > > 
> > > 那我关了，其他人有日志在新建吧。
> > 
> > 
> > 我新建了920+还是不行的，用的最新的arpl和mod，不管是否勾选都一样，怎么样提供日志呢
> 
> [#183 (comment)](https://github.com/wjz304/arpl-i18n/issues/183#issuecomment-1720757338)

和#183一样的情况，在arpl页面正常识别
[image]

[image]

进了系统就报错
[image]

使用的是最新的arpl，没有加额外的MOD

wjz304, 2023-09-21:
> > > > > > > > https://github.com/wjz304/arpl-modules/releases/tag/23.9.2 plz test it
> > > > > > > 
> > > > > > > 
> > > > > > > @wjz304 我刚刚试了。按以下步骤： 1、还是用的原来的我自编译的arpl 2、menu.sh里面在Modules menu里面选择Select loaded modules 3、看一眼show selected modules，确保mlx5_core在列表中 4、Build the loader 5、强制重装dsm 6、进dsm后正常识别网卡
> > > > > > > 所以是不是在build loader过程中，自动用了最新的module？
> > > > > > 
> > > > > > 
> > > > > > 是，modules中 勾选了就会用，不勾选就会用系统自带的（官方）
> > > > > 
> > > > > 
> > > > > @wjz304 这么看来的话，改版后的module已经解决问题了。谢谢。
> > > > 
> > > > 
> > > > 那我关了，其他人有日志在新建吧。
> > > 
> > > 
> > > 我新建了920+还是不行的，用的最新的arpl和mod，不管是否勾选都一样，怎么样提供日志呢
> > 
> > 
> > [#183 (comment)](https://github.com/wjz304/arpl-i18n/issues/183#issuecomment-1720757338)
> 
> 和#183一样的情况，在arpl页面正常识别 [image]
> 
> [image]
> 
> 进了系统就报错 [image]
> 
> 使用的是最新的arpl默认设置，没有加额外的MOD
> 
> 是否要测试把最新的MOD[23.9.3](https://github.com/wjz304/arpl-modules/releases/tag/23.9.3)加进去？
> 
> 又虚了一台3622XS试了试，仍旧使用最新的arpl默认设置，没有加额外的MOD，在arpl页面正常识别 [image]
> 
> 进了系统报错 [image]
> 
> 重新进arpl配置页面将模组中的mlx5_core取消勾选，再进DSM系统正常识别网卡 [image]
> 
> [image] [image]
> 
> 另外DVA3221和3622XS也是相同的情况。

就是 DVA3221和3622XS 都在不勾选的时候就可以了呗？

eschcn, 2023-09-22:
> 就是 DVA3221和3622XS 都在不勾选的时候就可以了呗？

是的。
DS920+无论是否勾选都不行。

wjz304, 2023-09-22:
[image]
日志信息有限，目前不好解

eschcn, 2023-09-22:
> [image] 日志信息有限，目前不好解

需要我做什么？怎么做？我对群晖和linux不熟悉，不过也可以尽可能的配合测试，尽量提供信息

wjz304, 2023-09-22:
> > [image] 日志信息有限，目前不好解
> 
> 需要我做什么？怎么做？我对群晖和linux不熟悉，不过也可以尽可能的配合测试，尽量提供信息

看一下 modules 23.9.1 的驱动 报什么错？

eschcn, 2023-09-22:
> > > [image] 日志信息有限，目前不好解
> > 
> > 
> > 需要我做什么？怎么做？我对群晖和linux不熟悉，不过也可以尽可能的配合测试，尽量提供信息
> 
> 看一下 modules 23.9.1 的驱动 报什么错？

可以直接用23.9.4的arpl升级23.9.1的modules吗

wjz304, 2023-09-22:
可以
 

Ing
***@***.***

eschcn, 2023-09-22:
貌似是同样的报错
[image]

wjz304, 2023-09-22:
> 貌似是同样的报错 [image]
先用这些架构的型号吧，
[image]
我找找最新的驱动再说吧

eschcn, 2023-09-22:
> > 貌似是同样的报错 [image]
> > 先用这些架构的型号吧，
> > [image]
> > 我找找最新的驱动再说吧

呃，完全看不懂这些是啥，能细说吗

wjz304, 2023-09-22:
就是圈的几个型号先不要用
 

Ing
***@***.***

wjz304, 2023-09-22:
改了

eschcn, 2023-09-22:
> 改了

好的，谢谢了