RROrg/rr #184: Ask about storage panel add-on.

PeterSuh-Q3, 2023-08-23:
Looking at the contents of the script, the HDD is still the default 24 bay
It seems that the SSD is only limited to the default 8 bays.

Will it be modified in the future so that it can be changed by referring to the environment settings?

wjz304, 2023-08-23:
Support now,
hdd 24 and ssd 8 are the default values for not passing parameters.
You can pass the reference.
as:
curl -kL https://raw.githubusercontent.com/wjz304/arpl-addons/main/storagepanel/all/usr/bin/storagepanel.sh | bash -s RACK_20_Bay 2X10

curl -kL https://raw.githubusercontent.com/wjz304/arpl-addons/main/storagepanel/all/usr/bin/storagepanel.sh | bash -s RACK_60_Bay 4X15

closed 2023-08-23 by PeterSuh-Q3

PeterSuh-Q3, 2023-08-23:
This is my idea.

export HDD_Bay="TOWER_6_Bay"
export SSD_Bay="1X2"

https://github.com/PeterSuh-Q3/tcrp-addons/commit/3db536984cbd574b656e1a8471a9eebeac75cdfa

If you use it like this, even if it is installed as a service
It seems that you can change it to export at any time.
However, these two exported variables should always maintain their set values without disappearing.

PeterSuh-Q3, 2023-08-23:
> Support now, hdd 24 and ssd 8 are the default values for not passing parameters. You can pass the reference. as: curl -kL [...]

This is my idea.

export HDD_Bay="TOWER_6_Bay"
export SSD_Bay="1X2"

https://github.com/PeterSuh-Q3/tcrp-addons/commit/3db536984cbd574b656e1a8471a9eebeac75cdfa

If you use it like this, even if it is installed as a service
It seems that you can change it to export at any time.
However, these two exported variables should always maintain their set values without disappearing.

wjz304, 2023-08-23:
> export HDD_Bay="TOWER_6_Bay"
> export SSD_Bay="1X2"

As a tcrp addon, parameters need to be defined and passed in exts.sh
Or can be hardcoded directly in storagepanel.sh

PeterSuh-Q3, 2023-08-24:
> > export HDD_Bay="TOWER_6_Bay"
> > export SSD_Bay="1X2"
> 
> As a tcrp addon, parameters need to be defined and passed in exts.sh Or can be hardcoded directly in storagepanel.sh

[image]

There's typo opcional -> optional. ^^

wjz304, 2023-08-24:
Thank you very much, it seems that everyone's English is not good.
This sentence was written by fbelavenuto and has not been changed.

PeterSuh-Q3, 2023-08-24:
> Thank you very much, it seems that everyone's English is not good. This sentence was written by fbelavenuto and has not been changed.

Thanks to you, I was able to apply good features to my TCRP as well.

[image]

[image]

wjz304, 2023-08-24:
But according to user feedback, it seems that there are some issues with the SSD of 3622, and I am not a skilled JavaScript user. Have you studied it

wjz304, 2023-08-24:
Your English and Chinese are both very good

PeterSuh-Q3, 2023-08-24:
> But according to user feedback, it seems that there are some issues with the SSD of 3622, and I am not a skilled JavaScript user. Have you studied it

In the case of DS3622xs+, it seems that there was no indication of the M.2 SSD bay from the beginning.
That's how it turned out in the actual installation test.
We are looking for evidence of a bug that existed in the past but has recently disappeared in 7.2. However, it is difficult to find an overview image.
Let's try another Google image search.