RROrg/rr #186: storagepanel need additional option

category-9431, 2023-08-24:
Install ENV:  
 * motherboard:  ga-z370m-d3h
 * CPU:  qqls
 * NIC:  bnx2x

Arpl version:  23.8.9

DSM model:  ds918
DSM version:  7.2

Issue:  
need storagepanel option
TOWER_6_Bay_J
TOWER_6_Bay_S
or 
TOWER_X_Bay rowXcolumn (horizontal sata)
TOWER_X_Bay rowXcolumn (vertical sata)

TOWER_6_Bay 6X1 doesn't work!

logs:  

...

wjz304, 2023-08-24:
Parameter 1 is the style of HDD, which is already in the system and does not exist as TOWER_ 6_ Bay_ J and TOWER_ 6_ Bay_ S. 

Parameter 2 is the style of SSD, independent of parameter 1,

wjz304, 2023-08-24:
TOWER 6_Bay_J can do it youself, but the more code there is, the higher the risk of bugs

closed 2023-08-25 by wjz304