RROrg/rr #189: nvmevolume.sh asks about updating from 1.07 to 1.08.

PeterSuh-Q3, 2023-08-27:
Script execution in service is interrupted due to question answering about update in shell script.
If you want to use it without modifying nvmevolume.sh, you need to modify the service registration method a little.
The following is the result of testing when updating to 1.08 and when answering n to say no.
Even if you simply change the install,sh script as shown below, it seems possible to keep running 1.07.
What do you think?

echo "n" | nvmevolume.sh

[image]

[image]

[image]

PeterSuh-Q3, 2023-08-27:
echo "n" | Processing doesn't work properly.
I ended up deleting the update check routine.

https://github.com/PeterSuh-Q3/tcrp-addons/commit/71e41832c69a66ba1b718902f4970ef286625aef

PeterSuh-Q3, 2023-08-27:
There are additional confirmations.
DS3622xs+ does not activate the NVMe volume even though the script is executed normally and the libhwcontrol.so.1 file is patched well.
Was it improved in v1.0.8?

Once in v1.0.7
In the case of DS1621xs+, which is the most similar model,
All functions are fine.

wjz304, 2023-08-27:
> echo "n" | Processing doesn't work properly. I ended up deleting the update check routine.
> 
> [PeterSuh-Q3/tcrp-addons@71e4183](https://github.com/PeterSuh-Q3/tcrp-addons/commit/71e41832c69a66ba1b718902f4970ef286625aef)

Yes, we'd better delete or comment out the update

closed 2023-09-01 by PeterSuh-Q3