RROrg/rr #190: BUG - Stuck when boot

doandat943, 2023-08-27:
Install ENV:

motherboard: ASUS X541U
CPU: Intel Core i3 Skylake 6100U
NIC: AX88179
Arpl version: 23.8.11

DSM model: DS918+
DSM version: 7.2 Update 1

Issue: Bootloader stuck like this (Configure Loader)

[image]

logs:  

...

doandat943, 2023-08-27:
im just try [23.8.9](https://github.com/wjz304/arpl-i18n/releases/tag/23.8.9), it working normal

wjz304, 2023-08-27:
https://github.com/wjz304/arpl-i18n/issues/175

wjz304, 2023-09-15:
Please test the latest version and provide as much information as possible above the 'kernel panic'

wjz304, 2023-09-16:
No further feedback, if there are still issues, please create a new issue and provide detailed information

closed 2023-09-16 by wjz304

doandat943, 2023-09-18:
> No further feedback, if there are still issues, please create a new issue and provide detailed information

Sorry for late feedback, im just flash lastest version and it working like a charm. Thank you so much!!!