RROrg/rr #195: Question: addons storage

marboroman, 2023-09-01:
[wjz304](https://github.com/wjz304), please tell where the addons chosen in loader are placed in DSM file system. How to run them in ssh when DSM is up?

wjz304, 2023-09-01:
It is not fixed, some can be found in DSM, while others cannot be seen
I can only say that most of the scripts will be placed in `/bin` or `/sbin`
Generally, scripts name is same as the addon name

marboroman, 2023-09-01:
ok, found them in /bin  -  will they be updated when update them in loader menu and rebuild the loader?

marboroman, 2023-09-01:
I actually tried to run hdddb.sh from /bin and it suggested to update itself but it doesn't work. Probably because it's name doesn't match original script's filename.

wjz304, 2023-09-01:
Yes, this problem has some headaches,

The current plan is to block its update processing, and I will update them uniformly.

closed 2023-09-04 by wjz304