RROrg/rr #197: How to use intel_pstate scaling driver?

sergejostir, 2023-09-03:
Hello.

I'm using the DS918+ image on DSM 7.2 with Intel N3150 CPU. I would like to use the intel_pstate scaling driver instead of the acpi_cpufreq. Is it possible to do that?

Best regards

wjz304, 2023-09-03:
Adding cmdline should do the trick

sergejostir, 2023-09-03:
Hello.

Thank you for your answer. Added the line using the menu
[image]

But the scaling_driver is still acpi-cpufreq
[image]

I can see in /var/log/kern.log that intel_pstate=enable was added to the "Kernel command line".

closed 2023-09-03 by sergejostir