RROrg/rr #198: BUG : console in PROXMOX

nonelocker, 2023-09-04:
Install ENV:  
 * motherboard:  A520I AC
 * CPU:  AMD 5700G
 * NIC:  Realtek 

Arpl version:  9.0  in PROXMOX

DSM model:  3622xs+
DSM version:  7.2

Issue:  console does not work in Proxmox console window

logs:  I was in arpl 6.8. and upgrade to 7.x , 8.x, 9.0
         But 7.x~9.0 version does not working console in proxmix console.
         There is no problem in arpl 6.8
        What is wrong?

...

wjz304, 2023-09-04:
not working ?
Do you have any pictures?

nonelocker, 2023-09-04:
[image]
Above screen, console screen does not appear.

wjz304, 2023-09-04:
[image]

Are you referring to the one on the right?

wjz304, 2023-09-04:
My English is not good, I don't know which console you are referring to,

The console on the right side in my above screenshot is a serial port,
If you want noVNC on the left to display the console, you need a model with a nuclear display and install the console addons.

closed 2023-09-05 by wjz304