RROrg/rr #199: 版本 23.9.0 14盘位，每次重启后会丢失后3个盘位，然后需要进磁盘管理修复

jiang123574, 2023-09-05:
初步怀疑是由于引导过快，导致磁盘还没加载好就引导好了，导致磁盘在引导好之后才加载

lzr325, 2023-09-05:
HddEnableDynamicPower=1改为HddEnableDynamicPower=0试下。DS920+也出现了这个问题，改为HddEnableDynamicPower=0后问题暂时未复现 @wjz304

jiang123574, 2023-09-05:
> HddEnableDynamicPower=1改为HddEnableDynamicPower=0试下。DS920+也出现了这个问题，改为HddEnableDynamicPower=0以后问题暂时未复现@wjz304

谢谢，明天我试一下，被困扰很久

jiang123574, 2023-09-06:
> HddEnableDynamicPower=1改为HddEnableDynamicPower=0试下。DS920+也出现了这个问题，改为HddEnableDynamicPower=0后问题暂时未复现 @wjz304

问下这个参数是加在哪里，引导里没有找到这个参数

jiang123574, 2023-09-06:
> HddEnableDynamicPower=1改为HddEnableDynamicPower=0试下。DS920+也出现了这个问题，改为HddEnableDynamicPower=0后问题暂时未复现 @wjz304

尝试在cmdline加上了这个参数  还是没用

snailium, 2023-09-06:
基本上确定是引入了“显示连接在SAS HBA上的硬盘序列号”这个功能导致的。所有连接到SAS HBA的硬盘都会受影响

jiang123574, 2023-09-06:
> 基本上确定是引入了“显示连接在SAS HBA上的硬盘序列号”这个功能导致的。所有连接到SAS HBA的硬盘都会受影响

但是更早的版本也可以显示序列号，然后并不会有这样的bug

snailium, 2023-09-07:
> > 基本上确定是引入了“显示连接在SAS HBA上的硬盘序列号”这个功能导致的。所有连接到SAS HBA的硬盘都会受影响
> 
> 但是更早的版本也可以显示序列号，然后并不会有这样的bug

早前版本只有在DS3622xs+这些原生支持SAS的机型上才能显示序列号，在像DS918+这种需要loader做SAS兼容的机型上无法显示序列号。本次Peter Suh的更新是在loader层面上的sas shim里面加上了传递序列号的功能

jiang123574, 2023-09-07:
> > > 基本上确定是引入了“显示连接在SAS HBA上的硬盘序列号”这个功能导致的。所有连接到SAS HBA的硬盘都会受影响
> > 
> > 
> > 但是更早的版本也可以显示序列号，然后并不会有这样的bug
> 
> 早前版本只有在DS3622xs+这些原生支持SAS的机型上才能显示序列号，在像DS918+这种需要loader做SAS兼容的机型上无法显示序列号。本次Peter Suh的更新是在loader层面上的sas shim里面加上了传递序列号的功能

原来是这样，不过我的本身引导的就是DS3622XS+

aarpro, 2023-09-09:
I can't find the setting **HddEnableDynamicPower=**
Where to look it to change?
I have a similar problem to tow DS918 server and DS3622  - 2 SSD cache on motherboard SATA, 4 HDD on LSI SAS 9211-4i controller. When rebooting system section error in the last HDD disk. 

recovery when pressing the corresponding button occurs instantly

DS918
[image]
[image]

DS3622
[image]
[image]

aarpro, 2023-09-09:
> HddEnableDynamicPower=1改为HddEnableDynamicPower=0试下。DS920+也出现了这个问题，改为HddEnableDynamicPower=0后问题暂时未复现 @wjz304

I can't find the setting **HddEnableDynamicPower=**
Where to look it to change?
I have a similar problem https://github.com/wjz304/arpl-i18n/issues/206

wjz304, 2023-09-09:
> HddEnableDynamicPower

add it to ”cmdline“

wjz304, 2023-09-09:
https://github.com/wjz304/arpl-addons/releases/tag/23.9.2
@jiang123574 优化了一下 *portcfg 和disks相关的参数获取方式，请测试。

aarpro, 2023-09-09:
> https://github.com/wjz304/arpl-addons/releases/tag/23.9.2 @jiang123574Оптимизирован метод получения параметров, связанных с *portcfg и дисками, протестируйте.
addon 23.9.2
does not see the last disk at all
Accordingly, the system has passed into a critical state

[image]

aarpro, 2023-09-09:
> https://github.com/wjz304/arpl-addons/releases/tag/23.9.2 @jiang123574 优化了一下 *portcfg 和disks相关的参数获取方式，请测试。

addon 23.9.2
does not see the last disk at all
Accordingly, the system has passed into a critical state
[image]

wjz304, 2023-09-09:
取以下 日志
```
ls /dev/
lspci -d ::106
lspci -d ::107
ls -l /sys/class/scsi_host
cat /etc/synoinfo.conf | grep portcfg
cat /etc/synoinfo.conf | grep disks
```

aarpro, 2023-09-09:
> 取以下 日志 (在 dsm系统下)
I'm not very strong in Linux :(
I do it on the command line, when is DSM load? Right ?

wjz304, 2023-09-09:
Yes, after the dsm system boot, connect SSH and operate it

If you have returned to the old version, then give up temporarily

aarpro, 2023-09-09:
> Yes, after the dsm system boot, connect SSH and operate it
> 
> If you have returned to the old version, then give up temporarily

ok, but to return to the experimental version of additions 23.9.2 I need to wait for recovery
[image]

snailium, 2023-09-09:
> HddEnableDynamicPower=1改为HddEnableDynamicPower=0试下。DS920+也出现了这个问题，改为HddEnableDynamicPower=0后问题暂时未复现 @wjz304

Tried adding HddEnableDynamicPower=0 to cmdline, no improvement.

Tried 23.9.1, no improvement.

DS3622XS+ on PVE, with LSI9207-8i (SAS2308)

snailium, 2023-09-09:
`$ ls /dev/
block               ptya7  ptyde  ptyr5  ptyuc  ptyy3   sdh2       tty30  ttyaf  ttye6  ttyrd  ttyv0  ttyy7
bnxt_lfc            ptya8  ptydf  ptyr6  ptyud  ptyy4   sdh5       tty31  ttyb0  ttye7  ttyre  ttyv1  ttyy8
bsg                 ptya9  ptye0  ptyr7  ptyue  ptyy5   sdi        tty32  ttyb1  ttye8  ttyrf  ttyv2  ttyy9
btrfs-control       ptyaa  ptye1  ptyr8  ptyuf  ptyy6   sdi1       tty33  ttyb2  ttye9  ttys0  ttyv3  ttyya
bus                 ptyab  ptye2  ptyr9  ptyv0  ptyy7   sdi2       tty34  ttyb3  ttyea  ttyS0  ttyv4  ttyyb
char                ptyac  ptye3  ptyra  ptyv1  ptyy8   sdi5       tty35  ttyb4  ttyeb  ttys1  ttyv5  ttyyc
console             ptyad  ptye4  ptyrb  ptyv2  ptyy9   sdj        tty36  ttyb5  ttyec  ttyS1  ttyv6  ttyyd
core                ptyae  ptye5  ptyrc  ptyv3  ptyya   sdj1       tty37  ttyb6  ttyed  ttys2  ttyv7  ttyye
cpu                 ptyaf  ptye6  ptyrd  ptyv4  ptyyb   sdj2       tty38  ttyb7  ttyee  ttyS2  ttyv8  ttyyf
cpu_dma_latency     ptyb0  ptye7  ptyre  ptyv5  ptyyc   sdj5       tty39  ttyb8  ttyef  ttys3  ttyv9  ttyz0
dm-0                ptyb1  ptye8  ptyrf  ptyv6  ptyyd   sdk        tty4   ttyb9  ttyp0  ttyS3  ttyva  ttyz1
dm-1                ptyb2  ptye9  ptys0  ptyv7  ptyye   sdk1       tty40  ttyba  ttyp1  ttys4  ttyvb  ttyz2
dm-2                ptyb3  ptyea  ptys1  ptyv8  ptyyf   sdk2       tty41  ttybb  ttyp2  ttys5  ttyvc  ttyz3
dm-3                ptyb4  ptyeb  ptys2  ptyv9  ptyz0   sdk5       tty42  ttybc  ttyp3  ttys6  ttyvd  ttyz4
dm-4                ptyb5  ptyec  ptys3  ptyva  ptyz1   sg0        tty43  ttybd  ttyp4  ttys7  ttyve  ttyz5
dm-5                ptyb6  ptyed  ptys4  ptyvb  ptyz2   sg1        tty44  ttybe  ttyp5  ttys8  ttyvf  ttyz6
fd                  ptyb7  ptyee  ptys5  ptyvc  ptyz3   sg2        tty45  ttybf  ttyp6  ttys9  ttyw0  ttyz7
full                ptyb8  ptyef  ptys6  ptyvd  ptyz4   sg3        tty46  ttyc0  ttyp7  ttysa  ttyw1  ttyz8
fuse                ptyb9  ptyp0  ptys7  ptyve  ptyz5   sg4        tty47  ttyc1  ttyp8  ttysb  ttyw2  ttyz9
i2c-0               ptyba  ptyp1  ptys8  ptyvf  ptyz6   sg5        tty48  ttyc2  ttyp9  ttysc  ttyw3  ttyza
kmem                ptybb  ptyp2  ptys9  ptyw0  ptyz7   shm        tty49  ttyc3  ttypa  ttysd  ttyw4  ttyzb
kmsg                ptybc  ptyp3  ptysa  ptyw1  ptyz8   snapshot   tty5   ttyc4  ttypb  ttyse  ttyw5  ttyzc
kvm                 ptybd  ptyp4  ptysb  ptyw2  ptyz9   stderr     tty50  ttyc5  ttypc  ttysf  ttyw6  ttyzd
log                 ptybe  ptyp5  ptysc  ptyw3  ptyza   stdin      tty51  ttyc6  ttypd  ttyt0  ttyw7  ttyze
loop0               ptybf  ptyp6  ptysd  ptyw4  ptyzb   stdout     tty52  ttyc7  ttype  ttyt1  ttyw8  ttyzf
loop1               ptyc0  ptyp7  ptyse  ptyw5  ptyzc   synobios   tty53  ttyc8  ttypf  ttyt2  ttyw9  urandom
loop2               ptyc1  ptyp8  ptysf  ptyw6  ptyzd   synoboot   tty54  ttyc9  ttyq0  ttyt3  ttywa  vcs
loop3               ptyc2  ptyp9  ptyt0  ptyw7  ptyze   synoboot1  tty55  ttyca  ttyq1  ttyt4  ttywb  vcs1
loop4               ptyc3  ptypa  ptyt1  ptyw8  ptyzf   synoboot2  tty56  ttycb  ttyq2  ttyt5  ttywc  vcs2
loop5               ptyc4  ptypb  ptyt2  ptyw9  ram0    synoboot3  tty57  ttycc  ttyq3  ttyt6  ttywd  vcs3
loop6               ptyc5  ptypc  ptyt3  ptywa  ram1    tty        tty58  ttycd  ttyq4  ttyt7  ttywe  vcs4
loop7               ptyc6  ptypd  ptyt4  ptywb  ram10   tty0       tty59  ttyce  ttyq5  ttyt8  ttywf  vcs5
loop-control        ptyc7  ptype  ptyt5  ptywc  ram11   tty1       tty6   ttycf  ttyq6  ttyt9  ttyx0  vcs6
mapper              ptyc8  ptypf  ptyt6  ptywd  ram12   tty10      tty60  ttyd0  ttyq7  ttyta  ttyx1  vcsa
md0                 ptyc9  ptyq0  ptyt7  ptywe  ram13   tty11      tty61  ttyd1  ttyq8  ttytb  ttyx2  vcsa1
md1                 ptyca  ptyq1  ptyt8  ptywf  ram14   tty12      tty62  ttyd2  ttyq9  ttytc  ttyx3  vcsa2
md2                 ptycb  ptyq2  ptyt9  ptyx0  ram15   tty13      tty63  ttyd3  ttyqa  ttytd  ttyx4  vcsa3
md3                 ptycc  ptyq3  ptyta  ptyx1  ram2    tty14      tty7   ttyd4  ttyqb  ttyte  ttyx5  vcsa4
mem                 ptycd  ptyq4  ptytb  ptyx2  ram3    tty15      tty8   ttyd5  ttyqc  ttytf  ttyx6  vcsa5
memory_bandwidth    ptyce  ptyq5  ptytc  ptyx3  ram4    tty16      tty9   ttyd6  ttyqd  ttyu0  ttyx7  vcsa6
mpt2ctl             ptycf  ptyq6  ptytd  ptyx4  ram5    tty17      ttya0  ttyd7  ttyqe  ttyu1  ttyx8  vg1
mpt3ctl             ptyd0  ptyq7  ptyte  ptyx5  ram6    tty18      ttya1  ttyd8  ttyqf  ttyu2  ttyx9  vg2
network_latency     ptyd1  ptyq8  ptytf  ptyx6  ram7    tty19      ttya2  ttyd9  ttyr0  ttyu3  ttyxa  vga_arbiter
network_throughput  ptyd2  ptyq9  ptyu0  ptyx7  ram8    tty2       ttya3  ttyda  ttyr1  ttyu4  ttyxb  vhost-scsi
null                ptyd3  ptyqa  ptyu1  ptyx8  ram9    tty20      ttya4  ttydb  ttyr2  ttyu5  ttyxc  zero
port                ptyd4  ptyqb  ptyu2  ptyx9  random  tty21      ttya5  ttydc  ttyr3  ttyu6  ttyxd  zram0
ptmx                ptyd5  ptyqc  ptyu3  ptyxa  rtc     tty22      ttya6  ttydd  ttyr4  ttyu7  ttyxe
pts                 ptyd6  ptyqd  ptyu4  ptyxb  rtc0    tty23      ttya7  ttyde  ttyr5  ttyu8  ttyxf
ptya0               ptyd7  ptyqe  ptyu5  ptyxc  sdg     tty24      ttya8  ttydf  ttyr6  ttyu9  ttyy0
ptya1               ptyd8  ptyqf  ptyu6  ptyxd  sdg1    tty25      ttya9  ttye0  ttyr7  ttyua  ttyy1
ptya2               ptyd9  ptyr0  ptyu7  ptyxe  sdg2    tty26      ttyaa  ttye1  ttyr8  ttyub  ttyy2
[... cut ...]
lrwxrwxrwx 1 root root 0 Sep  9 00:12 host5 -> ../../devices/pci0000:00/0000:00:1f.2/ata6/host5/scsi_host/host5
lrwxrwxrwx 1 root root 0 Sep  9 00:12 host6 -> ../../devices/pci0000:00/0000:00:1c.0/0000:01:00.0/host6/scsi_host/host6
lrwxrwxrwx 1 root root 0 Sep  9 00:12 host7 -> ../../devices/pci0000:00/0000:00:1a.7/usb1/1-1/1-1:1.0/host7/scsi_host/host7`

`$ cat /etc/synoinfo.conf | grep portcfg
esataportcfg="0x00"
internalportcfg="0x3ffff"`

`$ cat /etc/synoinfo.conf | grep disks
maxdisks="18"`

snailium, 2023-09-09:
DVA3221 on baremetal, LSI9200-4i4e (SAS2308)

Even SATA disks on SAS HBA have problem.

> $ ls /dev/
> block               ptp0   ptydb  ptyr8  ptyv5  ptyz2   sdk2      stderr     tty56  ttyd1  ttyqe  ttyu7  ttyy4
> bnxt_lfc            pts    ptydc  ptyr9  ptyv6  ptyz3   sdk3      stdin      tty57  ttyd2  ttyqf  ttyu8  ttyy5
> bsg                 ptya0  ptydd  ptyra  ptyv7  ptyz4   sdl       stdout     tty58  ttyd3  ttyr0  ttyu9  ttyy6
> btrfs-control       ptya1  ptyde  ptyrb  ptyv8  ptyz5   sdl1      synobios   tty59  ttyd4  ttyr1  ttyua  ttyy7
> bus                 ptya2  ptydf  ptyrc  ptyv9  ptyz6   sdl2      synoboot   tty6   ttyd5  ttyr2  ttyub  ttyy8
> char                ptya3  ptye0  ptyrd  ptyva  ptyz7   sdl3      synoboot1  tty60  ttyd6  ttyr3  ttyuc  ttyy9
> console             ptya4  ptye1  ptyre  ptyvb  ptyz8   sdm       synoboot2  tty61  ttyd7  ttyr4  ttyud  ttyya
> core                ptya5  ptye2  ptyrf  ptyvc  ptyz9   sdm1      synoboot3  tty62  ttyd8  ttyr5  ttyue  ttyyb
> cpu                 ptya6  ptye3  ptys0  ptyvd  ptyza   sdm2      tty        tty63  ttyd9  ttyr6  ttyuf  ttyyc
> cpu_dma_latency     ptya7  ptye4  ptys1  ptyve  ptyzb   sdm3      tty0       tty7   ttyda  ttyr7  ttyv0  ttyyd
> dm-0                ptya8  ptye5  ptys2  ptyvf  ptyzc   sdn       tty1       tty8   ttydb  ttyr8  ttyv1  ttyye
> dm-1                ptya9  ptye6  ptys3  ptyw0  ptyzd   sdn1      tty10      tty9   ttydc  ttyr9  ttyv2  ttyyf
> dm-2                ptyaa  ptye7  ptys4  ptyw1  ptyze   sdn2      tty11      ttya0  ttydd  ttyra  ttyv3  ttyz0
> dm-3                ptyab  ptye8  ptys5  ptyw2  ptyzf   sdn3      tty12      ttya1  ttyde  ttyrb  ttyv4  ttyz1
> dm-4                ptyac  ptye9  ptys6  ptyw3  ram0    sdo       tty13      ttya2  ttydf  ttyrc  ttyv5  ttyz2
> dm-5                ptyad  ptyea  ptys7  ptyw4  ram1    sdo1      tty14      ttya3  ttye0  ttyrd  ttyv6  ttyz3
> dm-6                ptyae  ptyeb  ptys8  ptyw5  ram10   sdo2      tty15      ttya4  ttye1  ttyre  ttyv7  ttyz4
> dm-7                ptyaf  ptyec  ptys9  ptyw6  ram11   sdo3      tty16      ttya5  ttye2  ttyrf  ttyv8  ttyz5
> dm-8                ptyb0  ptyed  ptysa  ptyw7  ram12   sdp       tty17      ttya6  ttye3  ttys0  ttyv9  ttyz6
> fd                  ptyb1  ptyee  ptysb  ptyw8  ram13   sdp1      tty18      ttya7  ttye4  ttyS0  ttyva  ttyz7
> full                ptyb2  ptyef  ptysc  ptyw9  ram14   sdp2      tty19      ttya8  ttye5  ttys1  ttyvb  ttyz8
> fuse                ptyb3  ptyp0  ptysd  ptywa  ram15   sdp3      tty2       ttya9  ttye6  ttyS1  ttyvc  ttyz9
> i2c-0               ptyb4  ptyp1  ptyse  ptywb  ram2    sdq       tty20      ttyaa  ttye7  ttys2  ttyvd  ttyza
> i2c-1               ptyb5  ptyp2  ptysf  ptywc  ram3    sdq1      tty21      ttyab  ttye8  ttyS2  ttyve  ttyzb
> i2c-2               ptyb6  ptyp3  ptyt0  ptywd  ram4    sdq2      tty22      ttyac  ttye9  ttys3  ttyvf  ttyzc
> i2c-3               ptyb7  ptyp4  ptyt1  ptywe  ram5    sdq3      tty23      ttyad  ttyea  ttyS3  ttyw0  ttyzd
> isda                ptyb8  ptyp5  ptyt2  ptywf  ram6    sdr       tty24      ttyae  ttyeb  ttys4  ttyw1  ttyze
> kmem                ptyb9  ptyp6  ptyt3  ptyx0  ram7    sdr1      tty25      ttyaf  ttyec  ttys5  ttyw2  ttyzf
> kmsg                ptyba  ptyp7  ptyt4  ptyx1  ram8    sdr2      tty26      ttyb0  ttyed  ttys6  ttyw3  urandom
> kvm                 ptybb  ptyp8  ptyt5  ptyx2  ram9    sdr3      tty27      ttyb1  ttyee  ttys7  ttyw4  usb
> log                 ptybc  ptyp9  ptyt6  ptyx3  random  sds       tty28      ttyb2  ttyef  ttys8  ttyw5  vcs
> loop0               ptybd  ptypa  ptyt7  ptyx4  rtc     sds1      tty29      ttyb3  ttyp0  ttys9  ttyw6  vcs1
> loop1               ptybe  ptypb  ptyt8  ptyx5  rtc0    sds2      tty3       ttyb4  ttyp1  ttysa  ttyw7  vcs2
> loop2               ptybf  ptypc  ptyt9  ptyx6  sdb     sds3      tty30      ttyb5  ttyp2  ttysb  ttyw8  vcs3
> loop3               ptyc0  ptypd  ptyta  ptyx7  sdd     sdt       tty31      ttyb6  ttyp3  ttysc  ttyw9  vcs4
> loop4               ptyc1  ptype  ptytb  ptyx8  sde     sdt1      tty32      ttyb7  ttyp4  ttysd  ttywa  vcs5
> loop5               ptyc2  ptypf  ptytc  ptyx9  sde1    sdt2      tty33      ttyb8  ttyp5  ttyse  ttywb  vcs6
> loop6               ptyc3  ptyq0  ptytd  ptyxa  sde2    sdt3      tty34      ttyb9  ttyp6  ttysf  ttywc  vcsa
> loop7               ptyc4  ptyq1  ptyte  ptyxb  sde3    sg0       tty35      ttyba  ttyp7  ttyt0  ttywd  vcsa1
> loop-control        ptyc5  ptyq2  ptytf  ptyxc  sdf     sg1       tty36      ttybb  ttyp8  ttyt1  ttywe  vcsa2
> mapper              ptyc6  ptyq3  ptyu0  ptyxd  sdf1    sg10      tty37      ttybc  ttyp9  ttyt2  ttywf  vcsa3
> mcelog              ptyc7  ptyq4  ptyu1  ptyxe  sdf2    sg11      tty38      ttybd  ttypa  ttyt3  ttyx0  vcsa4
> md0                 ptyc8  ptyq5  ptyu2  ptyxf  sdf3    sg12      tty39      ttybe  ttypb  ttyt4  ttyx1  vcsa5
> md1                 ptyc9  ptyq6  ptyu3  ptyy0  sdg     sg13      tty4       ttybf  ttypc  ttyt5  ttyx2  vcsa6
> md2                 ptyca  ptyq7  ptyu4  ptyy1  sdg1    sg14      tty40      ttyc0  ttypd  ttyt6  ttyx3  vfio
> md3                 ptycb  ptyq8  ptyu5  ptyy2  sdg2    sg15      tty41      ttyc1  ttype  ttyt7  ttyx4  vg1
> md5                 ptycc  ptyq9  ptyu6  ptyy3  sdg3    sg16      tty42      ttyc2  ttypf  ttyt8  ttyx5  vg3
[... cut ...]
> lrwxrwxrwx 1 root root 0 Sep  9 06:45 host4 -> ../../devices/pci0000:00/0000:00:01.1/0000:02:00.0/host4/scsi_host/host4
> lrwxrwxrwx 1 root root 0 Sep  9 06:45 host5 -> ../../devices/pci0000:00/0000:00:14.0/usb1/1-13/1-13:1.0/host5/scsi_host/host5
> lrwxrwxrwx 1 root root 0 Sep  9 06:45 host6 -> ../../devices/tcm_loop_0/tcm_loop_adapter_0/host6/scsi_host/host6

> $ cat /etc/synoinfo.conf | grep portcfg
> esataportcfg="0x00"
> internalportcfg="0x3fffffffffff"
> usbportcfg="0x3c00000000000"

> $ cat /etc/synoinfo.conf | grep disks
> maxdisks="26"

aarpro, 2023-09-09:
> 取以下 日志 (在 dsm系统下)
> 
> ```
> ls /dev/
> lspci -d ::106
> lspci -d ::107
> ls -l /sys/class/scsi_host
> cat /etc/synoinfo.conf | grep portcfg
> cat /etc/synoinfo.conf | grep disks
> ```

```
XXXXXX@DS918_2:/$ ls /dev/
block            md3                 ptybf  ptyea  ptyr5  ptyu0  ptywb  ptyz6   sdg2              tty25  tty7   ttyc8  ttyp3  ttyre  ttyu5  ttyx0  ttyzb
bsg              mem                 ptyc0  ptyeb  ptyr6  ptyu1  ptywc  ptyz7   sdg5              tty26  tty8   ttyc9  ttyp4  ttyrf  ttyu6  ttyx1  ttyzc
btrfs-control    memory_bandwidth    ptyc1  ptyec  ptyr7  ptyu2  ptywd  ptyz8   sdh               tty27  tty9   ttyca  ttyp5  ttys0  ttyu7  ttyx2  ttyzd
bus              mpt2ctl             ptyc2  ptyed  ptyr8  ptyu3  ptywe  ptyz9   sdh1              tty28  ttya0  ttycb  ttyp6  ttyS0  ttyu8  ttyx3  ttyze
char             mpt3ctl             ptyc3  ptyee  ptyr9  ptyu4  ptywf  ptyza   sdh2              tty29  ttya1  ttycc  ttyp7  ttys1  ttyu9  ttyx4  ttyzf
console          network_latency     ptyc4  ptyef  ptyra  ptyu5  ptyx0  ptyzb   sdh5              tty3   ttya2  ttycd  ttyp8  ttyS1  ttyua  ttyx5  urandom
core             network_throughput  ptyc5  ptyp0  ptyrb  ptyu6  ptyx1  ptyzc   sg0               tty30  ttya3  ttyce  ttyp9  ttys2  ttyub  ttyx6  vcs
cpu              null                ptyc6  ptyp1  ptyrc  ptyu7  ptyx2  ptyzd   sg1               tty31  ttya4  ttycf  ttypa  ttyS2  ttyuc  ttyx7  vcs1
cpu_dma_latency  port                ptyc7  ptyp2  ptyrd  ptyu8  ptyx3  ptyze   sg2               tty32  ttya5  ttyd0  ttypb  ttys3  ttyud  ttyx8  vcs2
dm-0             ptmx                ptyc8  ptyp3  ptyre  ptyu9  ptyx4  ptyzf   sg3               tty33  ttya6  ttyd1  ttypc  ttyS3  ttyue  ttyx9  vcs3
dm-1             ptp0                ptyc9  ptyp4  ptyrf  ptyua  ptyx5  ram0    sg4               tty34  ttya7  ttyd2  ttypd  ttys4  ttyuf  ttyxa  vcs4
dm-2             pts                 ptyca  ptyp5  ptys0  ptyub  ptyx6  ram1    sg5               tty35  ttya8  ttyd3  ttype  ttys5  ttyv0  ttyxb  vcs5
dm-3             ptya0               ptycb  ptyp6  ptys1  ptyuc  ptyx7  ram10   sg6               tty36  ttya9  ttyd4  ttypf  ttys6  ttyv1  ttyxc  vcs6
dm-4             ptya1               ptycc  ptyp7  ptys2  ptyud  ptyx8  ram11   shared_cache_vg1  tty37  ttyaa  ttyd5  ttyq0  ttys7  ttyv2  ttyxd  vcsa
dri              ptya2               ptycd  ptyp8  ptys3  ptyue  ptyx9  ram12   shm               tty38  ttyab  ttyd6  ttyq1  ttys8  ttyv3  ttyxe  vcsa1
fb0              ptya3               ptyce  ptyp9  ptys4  ptyuf  ptyxa  ram13   snapshot          tty39  ttyac  ttyd7  ttyq2  ttys9  ttyv4  ttyxf  vcsa2
fd               ptya4               ptycf  ptypa  ptys5  ptyv0  ptyxb  ram14   stderr            tty4   ttyad  ttyd8  ttyq3  ttysa  ttyv5  ttyy0  vcsa3
full             ptya5               ptyd0  ptypb  ptys6  ptyv1  ptyxc  ram15   stdin             tty40  ttyae  ttyd9  ttyq4  ttysb  ttyv6  ttyy1  vcsa4
fuse             ptya6               ptyd1  ptypc  ptys7  ptyv2  ptyxd  ram2    stdout            tty41  ttyaf  ttyda  ttyq5  ttysc  ttyv7  ttyy2  vcsa5
i2c-0            ptya7               ptyd2  ptypd  ptys8  ptyv3  ptyxe  ram3    synobios          tty42  ttyb0  ttydb  ttyq6  ttysd  ttyv8  ttyy3  vcsa6
i2c-1            ptya8               ptyd3  ptype  ptys9  ptyv4  ptyxf  ram4    synoboot          tty43  ttyb1  ttydc  ttyq7  ttyse  ttyv9  ttyy4  vg1
i2c-2            ptya9               ptyd4  ptypf  ptysa  ptyv5  ptyy0  ram5    synoboot1         tty44  ttyb2  ttydd  ttyq8  ttysf  ttyva  ttyy5  vga_arbiter
i2c-3            ptyaa               ptyd5  ptyq0  ptysb  ptyv6  ptyy1  ram6    synoboot2         tty45  ttyb3  ttyde  ttyq9  ttyt0  ttyvb  ttyy6  vhost-scsi
i2c-4            ptyab               ptyd6  ptyq1  ptysc  ptyv7  ptyy2  ram7    synoboot3         tty46  ttyb4  ttydf  ttyqa  ttyt1  ttyvc  ttyy7  zero
i2c-5            ptyac               ptyd7  ptyq2  ptysd  ptyv8  ptyy3  ram8    tty               tty47  ttyb5  ttye0  ttyqb  ttyt2  ttyvd  ttyy8  zram0
kmem             ptyad               ptyd8  ptyq3  ptyse  ptyv9  ptyy4  ram9    tty0              tty48  ttyb6  ttye1  ttyqc  ttyt3  ttyve  ttyy9  zram1
kmsg             ptyae               ptyd9  ptyq4  ptysf  ptyva  ptyy5  random  tty1              tty49  ttyb7  ttye2  ttyqd  ttyt4  ttyvf  ttyya  zram2
kvm              ptyaf               ptyda  ptyq5  ptyt0  ptyvb  ptyy6  rtc     tty10             tty5   ttyb8  ttye3  ttyqe  ttyt5  ttyw0  ttyyb  zram3
log              ptyb0               ptydb  ptyq6  ptyt1  ptyvc  ptyy7  rtc0    tty11             tty50  ttyb9  ttye4  ttyqf  ttyt6  ttyw1  ttyyc
loop0            ptyb1               ptydc  ptyq7  ptyt2  ptyvd  ptyy8  sdc     tty12             tty51  ttyba  ttye5  ttyr0  ttyt7  ttyw2  ttyyd
loop1            ptyb2               ptydd  ptyq8  ptyt3  ptyve  ptyy9  sdc1    tty13             tty52  ttybb  ttye6  ttyr1  ttyt8  ttyw3  ttyye
loop2            ptyb3               ptyde  ptyq9  ptyt4  ptyvf  ptyya  sdd     tty14             tty53  ttybc  ttye7  ttyr2  ttyt9  ttyw4  ttyyf
loop3            ptyb4               ptydf  ptyqa  ptyt5  ptyw0  ptyyb  sdd1    tty15             tty54  ttybd  ttye8  ttyr3  ttyta  ttyw5  ttyz0
loop4            ptyb5               ptye0  ptyqb  ptyt6  ptyw1  ptyyc  sde     tty16             tty55  ttybe  ttye9  ttyr4  ttytb  ttyw6  ttyz1
[... cut ...]
XXXXXX@DS918_2:/$ lrwxrwxrwx 1 root root 0 Sep  9 22:45 host5 -> ../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/host5/scsi_host/host5
-sh: ../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/host5/scsi_host/host5: No such file or directory
XXXXXX@DS918_2:/$ XXXXXX@DS918_2:/$ cat /etc/synoinfo.conf | grep portcfg
-sh: XXXXXX@DS918_2:/$: No such file or directory
XXXXXX@DS918_2:/$ esataportcfg="0x00"
XXXXXX@DS918_2:/$ internalportcfg="0x7f"
XXXXXX@DS918_2:/$ usbportcfg="0x00"
XXXXXX@DS918_2:/$ XXXXXX@DS918_2:/$ cat /etc/synoinfo.conf | grep disks
```

wjz304, 2023-09-12:
> ../../devices/tcm_loop_0/tcm_loop_adapter_0/host6/scsi_host/host6

 ../../devices/tcm_loop_0/tcm_loop_adapter_0/host6/scsi_host/host6
能确定这个是什么设备吗？

wjz304, 2023-09-12:
> ```
> XXXXXX@DS918_2:/$ ls /dev/
> block            md3                 ptybf  ptyea  ptyr5  ptyu0  ptywb  ptyz6   sdg2              tty25  tty7   ttyc8  ttyp3  ttyre  ttyu5  ttyx0  ttyzb [...]

5个sata + 1个 usb， 看不到的哪个应该在usb里面了

wjz304, 2023-09-12:
> > 5个sata + 1个 usb， 看不到的哪个应该在usb里面了
> 
> Do I need to change something in the settings? edit config file ? which one ?
> 
> UPD Unfortunately, I have already dismantled the NAS from which the logs send to you.
> 
> Now I have 3 working servers:
> 
> ASRock J3355B-ITX / Celeron J3355 / 2 SSD cache on motherboard SATA and 4 HDD on LSI SAS 9211-4i controller / Arpl and addons version: 23.9.1 - after rebooting: system section error in the last HDD disk on LSI SAS
> 
> Advantech ASMB-260I-21A1 / Atom C3558 / 2 SSD cache on motherboard SATA and 4 HDD on LSI SAS 9211-4i controller / Arpl and addons version: 23.9.1 - after rebooting: system section error in the last HDD disk on LSI SAS
> 
> problem HDD system section error is solved by instant recovery after pressing the corresponding button in DSM interface
> 
> China ITX-B360V3,0 motherboard (6 SATA / 2 NVME) / i5-9500T / 2 NVME cache on motherboard slots and 5 HDD on motherboard SATA / Arpl and addons version: 23.9.1 - no any problems.

According to @snailium, I currently do not have the issue of environmental reproduction
https://github.com/wjz304/arpl-i18n/issues/201#issue-1883109624

aarpro, 2023-09-12:
> According to @snailium, I currently do not have the issue of environmental reproduction

If I understood you correctly, then no logs are required from me anymore?
also not required testing addons version: 23.9.2
Only expectation...
Right ?

wjz304, 2023-09-12:
> > According to @snailium, I currently do not have the issue of environmental reproduction
> 
> If I understood you correctly, then no logs are required from me anymore? also not required testing addons version: 23.9.2 Only expectation... Right ?

Yes

wjz304, 2023-09-12:
I will do some tests these days

aarpro, 2023-09-13:
> I will do some tests these days

I updated to [ARPL 23.9.2](https://github.com/wjz304/arpl-i18n/releases/tag/23.9.2), addon 23.9.3
After loading is no installed DSM on server :(
[2023-09-14_003818](https://github.com/wjz304/arpl-i18n/assets/14928083/4c99f5de-a858-4c16-a7df-347758e05a0c)

wjz304, 2023-09-13:
64570u3 和 64570u2 的 rd(bootloader) 是一样的

wjz304, 2023-09-13:
可以去一份日志吗？参考 https://github.com/wjz304/arpl-i18n/issues/173

aarpro, 2023-09-13:
> 可以去一份日志吗？参考 #173
I have already installed DSM DSM 7.2-64570 Update 3 - does not see HDD on LSI SAS 9211-4i controller - only HDD on board SATA

Now I will collect the log

aarpro, 2023-09-13:
> 可以去一份日志吗？参考 #173

I didn't have a directory /mnt/p1/logs

[image]

have a directory /mnt/logs/jr  

[image]

aarpro, 2023-09-13:
I will try over Loader .. ARC example

snailium, 2023-09-13:
23.9.2 even worse.

`dmesg | grep mpt` only gives the following

> [    2.178522] mpt3sas version 09.102.00.00 loaded
> [    2.180882] mpt2sas_cm0: 64 BIT PCI BUS DMA ADDRESSING SUPPORTED, total mem (8097280 kB)
> [    2.236077] mpt2sas_cm0: MSI-X vectors supported: 16, no of cores: 4, max_msix_vectors: -1
> [    2.256695] mpt2sas0-msix0: PCI-MSI-X enabled: IRQ 34
> [    2.256894] mpt2sas0-msix1: PCI-MSI-X enabled: IRQ 35
> [    2.257087] mpt2sas0-msix2: PCI-MSI-X enabled: IRQ 36
> [    2.257262] mpt2sas0-msix3: PCI-MSI-X enabled: IRQ 37
> [    2.257476] mpt2sas_cm0: iomem(0x00000000c2040000), mapped(0xffffc90000360000), size(65536)
> [    2.257753] mpt2sas_cm0: ioport(0x000000000000d000), size(256)
> [    2.351125] mpt2sas_cm0: Allocated physical memory: size(15846 kB)
> [    2.351335] mpt2sas_cm0: Current Controller Queue Depth(7931),Max Controller Queue Depth(8192)
> [    2.351616] mpt2sas_cm0: Scatter Gather Elements per IO(128)
> [    2.398515] mpt2sas_cm0: LSISAS2308: FWVersion(20.00.07.00), ChipRevision(0x05), BiosVersion(07.39.02.00)
> [    2.398928] mpt2sas_cm0: Protocol=(
> [    2.400159] mpt2sas_cm0: sending port enable !!
> [    2.404169] mpt2sas_cm0: host_add: handle(0x0001), sas_addr(0x500304801b3d9400), phys(8)
> [    2.411039] mpt2sas_cm0: port enable: SUCCESS
> 

It only detects one SATA HDD, none of the SAS HDD reported.

But, somehow, in Storage Manager I can see 1xSATA and 2xSAS. Another 2xSAS totally missing.

[image]

aarpro, 2023-09-13:
> I will try over Loader .. ARC example

ARC loaders same problems
see only three discs of four HDD and on third HDD system section error
4 HDD on LSI SAS 9211-4i

wjz304, 2023-09-14:
> 23.9.2 更糟糕。
> 
> `dmesg | grep mpt`仅提供以下内容
> 
> > [ 2.178522] mpt3sas 版本 09.102.00.00 已加载 [ 2.180882] mpt2sas_cm0：支持 64 位 PCI 总线 DMA 寻址，内存总量 （8097280 kB） [ 2.236077] mpt2sas_cm0：支持的 MSI-X 矢量：16，内核数：4，max_msix_vectors：-1 [ 2.256695] mpt2sas0-msix0：启用 PCI-MSI-X：IRQ 34 [ 2.256894] mpt2sas0-msix1：启用 PCI-MSI-X：IRQ 35 [ 2.257087] mpt2sas0-msix2：启用 PCI-MSI-X：IRQ 36[ 2.257262] mpt2sas0-msix3： 启用 PCI-MSI-X： IRQ 37 [ 2.257476] mpt2sas_cm0： IOMem（0x00000000c2040000）， 映射（0xffffc90000360000）， 大小 （65536） [ 2.257753] mpt2sas_cm0： 物联网（0x000000000000d000）， 大小（256） [ 2.351125] mpt2sas_cm0： 分配的物理内存： 大小（15846 kB） [ 2.351335] mpt2sas_cm0：当前控制器队列深度（7931），最大控制器队列深度（8192） [ 2.351616] mpt2sas_cm0：每 IO 分散收集元素（128） [ 2.398515] mpt2sas_cm0： LSISAS2308： FWVersion（20.00.07.00）， 芯片修订版（0x05）， 生物版本（07.39.02.00） [ 2.398928] mpt2sas_cm0： 协议=（ [ 2.400159] mpt2sas_cm0： 发送端口启用！！ [ 2.404169] mpt2sas_cm0： host_add： 手柄（0x0001）， sas_addr（0x500304801b3d9400）， 物理（8） [ 2.411039] mpt2sas_cm0：端口启用：成功
> 
> 它只检测到一个 SATA 硬盘，没有报告任何 SAS 硬盘。
> 
> 但是，不知何故，在存储管理器中，我可以看到 1xSATA 和 2xSAS。另一个 2xSAS 完全丢失。
> 
> [image]

ls -d /dev/sd*
lspci -d ::106
lspci -d ::107
ls -l /sys/class/scsi_host
cat /etc/synoinfo.conf | grep portcfg
cat /etc/synoinfo.conf | grep disks

snailium, 2023-09-14:
> $ ls -d /dev/sd*
> /dev/sdg   /dev/sdg2  /dev/sdg5  /dev/sdh1  /dev/sdh5  /dev/sdi1  /dev/sdi5  /dev/sdj1  /dev/sdj5  /dev/sdk1  /dev/sdk5
> /dev/sdg1  /dev/sdg3  /dev/sdh   /dev/sdh2  /dev/sdi   /dev/sdi2  /dev/sdj   /dev/sdj2  /dev/sdk   /dev/sdk2

> $ lspci -d ::106
> 0000:00:1f.2 Class 0106: Device 8086:2922 (rev 02)
> 0001:09:00.0 Class 0106: Device 1b4b:9235 (rev 11)
> 0001:0c:00.0 Class 0106: Device 1b4b:9235 (rev 11)

> $ lspci -d ::107
> 0000:01:00.0 Class 0107: Device 1000:0086 (rev 05)

> $ ls -l /sys/class/scsi_host
> total 0
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host0 -> ../../devices/pci0000:00/0000:00:1f.2/ata1/host0/scsi_host/host0
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host1 -> ../../devices/pci0000:00/0000:00:1f.2/ata2/host1/scsi_host/host1
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host2 -> ../../devices/pci0000:00/0000:00:1f.2/ata3/host2/scsi_host/host2
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host3 -> ../../devices/pci0000:00/0000:00:1f.2/ata4/host3/scsi_host/host3
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host4 -> ../../devices/pci0000:00/0000:00:1f.2/ata5/host4/scsi_host/host4
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host5 -> ../../devices/pci0000:00/0000:00:1f.2/ata6/host5/scsi_host/host5
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host6 -> ../../devices/pci0000:00/0000:00:1c.0/0000:01:00.0/host6/scsi_host/host6
> lrwxrwxrwx 1 root root 0 Sep 13 19:33 host7 -> ../../devices/pci0000:00/0000:00:1a.7/usb1/1-1/1-1:1.0/host7/scsi_host/host7

> $ cat /etc/synoinfo.conf | grep portcfg
> esataportcfg="0x00"
> internalportcfg="0x1ff"
> usbportcfg="0x00"

> $ cat /etc/synoinfo.conf | grep disks
> maxdisks="9"

Looks like maxdisk issue.

wjz304, 2023-09-14:
Let me give a brief explanation:
The hard drive displayed on WEBUI is affected by maxdisks, internalportcfg, esatportcfg, and usbportcfg, resulting in inaccurate display.
`ls -d /dev/sd* `  returns the actual HDD that has been driven.

wjz304, 2023-09-14:
But Addons 23.9.3, I made adjustments and theoretically should be able to provide the correct value

wjz304, 2023-09-14:
[internalportcfg,esataportcfg,usbportcfg计算.xlsx](https://github.com/wjz304/arpl-i18n/files/12602828/internalportcfg.esataportcfg.usbportcfg.xlsx)

aarpro, 2023-09-14:
> 64570u3 和 64570u2 的 rd(bootloader) 是一样的

format all disk -->burn new flash drive [arpl-i18n 23.9.2)](https://github.com/wjz304/arpl-i18n/releases/tag/23.9.2) --> All default settings (exclusion serial number and MAK)

[image]

wjz304, 2023-09-14:
> > Let me give a brief explanation: The hard drive displayed on WEBUI is affected by maxdisks, internalportcfg, esatportcfg, and usbportcfg, resulting in inaccurate display. `ls -d /dev/sd* ` returns the actual HDD that has been driven.
> 
> Thanks for pointing that out.
> 
> Couple questions.
> 
> 1. If I understand correct, maxdisks should be set automatically by ARPL script. How come it sets to 9 in 23.9.2? Given in 23.9.0, it was 18.（之前版本maxdisks被设置成18，为什么这次设成了9？）
> 2. How to get rid of some SAS/SATA controllers?（如何让ARPL屏蔽一些SAS/SATA控制器？例如Proxmox VM会自动添加一个SATA控制器，但是上面不会出现任何硬盘，控制器也无法删除，有没有办法让系统忽略掉这个控制器？就是下图里面的ATA控制器）
> 
> [image]

1. 之前的 maxdisks 并不十分准确，即使现在也不保证100%准确。收到很多因素的干扰，只能根据一些条件计算，当然适当的加大的结果总是好的。
2.  目前没办法，另外 arpl 和dsm 识别到的控制前也不完全一致， arpl 和dsm是两个系统，驱动不一样最终识别出来的控制器也不保证100%一致。

wjz304, 2023-09-14:
> format all disk -->burn new flash drive [arpl-i18n 23.9.2)](https://github.com/wjz304/arpl-i18n/releases/tag/23.9.2) --> All default settings (exclusion serial number and MAK)
> 
> does not see the last HDD disk on LSI SAS 9211-4i controller (install 2 SDD on MB SATA and 4 HDD on controller)

```
ls -d /dev/sd*
lspci -d ::106
lspci -d ::107
ls -l /sys/class/scsi_host
cat /etc/synoinfo.conf | grep portcfg
cat /etc/synoinfo.conf | grep disks
```
The hard drive displayed on WEBUI is affected by maxdisks, internalportcfg, esatportcfg, and usbportcfg, resulting in inaccurate display.
ls -d /dev/sd*  returns the actual HDD that has been driven.

wjz304, 2023-09-14:
@snailium  不过你倒是提醒了我，或许可以通过屏蔽驱动达到预期的效果

snailium, 2023-09-14:
> 
>     1. 之前的 maxdisks 并不十分准确，即使现在也不保证100%准确。收到很多因素的干扰，只能根据一些条件计算，当然适当的加大的结果总是好的。
> 
>     2. 目前没办法，另外 arpl 和dsm 识别到的控制前也不完全一致， arpl 和dsm是两个系统，驱动不一样最终识别出来的控制器也不保证100%一致。

你说的没错，ARPL和DSM看到的不一样，下面是ARPL看到的，结合你的表格，计算是没错的，但是明显DSM看到了更多

顺便问一下，

> esataportcfg="0x00"
> internalportcfg="0x1ff"
> usbportcfg="0x00"
> 

这里面缺少了 sataportcfg（不带e），是不是把sataportcfg设置上就能好了？

[image]

snailium, 2023-09-14:
> @snailium 不过你倒是提醒了我，或许可以通过屏蔽驱动达到预期的效果

但是 ahci 驱动无法屏蔽

wjz304, 2023-09-14:
internalportcfg ≈ sataportcfg

wjz304, 2023-09-14:
> > @snailium 不过你倒是提醒了我，或许可以通过屏蔽驱动达到预期的效果
> 
> 但是 ahci 驱动无法屏蔽

临时冒出的一个想法而已

snailium, 2023-09-14:
好像并不是DSM看到更多，而是ARPL和DSM里面看到的控制器顺序不一样，DSM明显是先SATA控制器再SAS HBA，ARPL可能是先挂栽SAS然后才是SATA？

snailium, 2023-09-14:
要不，计算maxdisks的时候，按照最大数量算？6xSATA + 8xSAS = 14，再怎么样也错不了

wjz304, 2023-09-14:
不用纠结这些，所以我没有在arpl下做 *portcfg 的设置。

wjz304, 2023-09-14:
> 要不，计算maxdisks的时候，按照最大数量算？6xSATA + 8xSAS = 14，再怎么样也错不了

maxdisks 应该等于 sd* 最大的角标，即 如果存在 sdz，maxdisks应该为 26

snailium, 2023-09-14:
> > 要不，计算maxdisks的时候，按照最大数量算？6xSATA + 8xSAS = 14，再怎么样也错不了
> 
> maxdisks 应该等于 sd* 最大的角标，即 如果存在 sdz，maxdisks应该为 26

这么算是不对的，目前我的SAS HBA上只挂载了5块硬盘，如果按最大脚标计算，maxdisks=11，但是我可以在系统启动之后再插入3块硬盘，后插入的这三块就已经多于maxdisks了，系统不会识别。如果只考虑到sdz的情况，是否可以尝试将maxdisks设置成常量26，不用计算了

wjz304, 2023-09-14:
是的，可以， 

我上面也说了 Addons 23.9.3 会自动设置这些，每次开机都会根据当前硬盘去自动设置这些，而且当存在 HBA时，是会自动设置为26的

wjz304, 2023-09-14:
https://github.com/wjz304/arpl-i18n/issues/199#issuecomment-1718497290 这不应该是 Addons 23.9.3 的结果

jiang123574, 2023-09-14:
为什么旧版本的引导不会有这样的问题呢，是不是考虑对3622这种原生支持hba卡的设备做出一些区别

wjz304, 2023-09-14:
> 为什么旧版本的引导不会有这样的问题呢，是不是考虑对3622这种原生支持hba卡的设备做出一些区别

你指的旧版是哪个版本？ 
那不是还有 usb识别为sata，sata识别为usb的问题，还有前些天 Q3 说的 sn的问题。 
然而其实我哪个版本也没问题啊 [image]

jiang123574, 2023-09-14:
> > 为什么旧版本的引导不会有这样的问题呢，是不是考虑对3622这种原生支持hba卡的设备做出一些区别
> 
> 你指的旧版是哪个版本？ 那不是还有 usb识别为sata，sata识别为usb的问题，还有前些天 Q3 说的 sn的问题。 然而其实我哪个版本也没问题啊 [image]

哦 好像确实，我忘记我当时用的哪个版本了，直通卡没有问题，插上usb移动固态硬盘被识别为内部硬盘

snailium, 2023-09-14:
> 为什么旧版本的引导不会有这样的问题呢，是不是考虑对3622这种原生支持hba卡的设备做出一些区别

23.9.0出问题是因为上游合并了Broadcom的mpt3sas驱动（41.00.00.00，详见 #209 ）
23.9.2又回滚了驱动，改回了Synology的mpt3sas驱动（09.102.00.00）

但是，我发现23.9.2的ARPL部分用了更新的mpt3sas驱动（41.03.00.00）

[74 more replies]

closed 2023-09-19 by wjz304