RROrg/rr #201: 23.9.0 Missing storage pool on SAS HBA, bad performance when boot up

snailium, 2023-09-06:
Install ENV:  
 * motherboard:  Asus Prime Z390-A
 * CPU:  Intel Core i7 8700
 * NIC:  Onboard + X520

Arpl version:  23.9.0

DSM model:  DVA3221
DSM version:  DSM 7.2-64570 Update 3

Issue:

Version 23.9.0 introduces a new LKMS feature to show real HDD serial number for those HDD connected to SAS HBA. This is a good move. We've been waiting for this feature for a long time.

However, after upgraded to 23.9.0, due to HDD serial number change, all storage pool on SAS HBA will show as missing.

Also, I noticed, when system is starting up, the HDD pops up one by one. It seems the loader doesn't pass the HDD information until DSM boots up. This results unstable storage pool recognization.

23.9.0版本引入了新的LKMS功能，可以显示连在SAS HBA上的硬盘序列号了。

但是，因为序列号改变了，之前系统里的存储池全部都会变成“丢失”状态。

另外，我还注意到，在DSM启动之后（可以从网页登陆之后），连接在SAS HBA上的硬盘一个接一个被识别（而不是像以前那样，进入系统之前就已经全部识别）。貌似loader在启动DSM的时候并没有将SAS HBA上的硬盘信息传送给DSM，而是在DSM完全启动之后才将硬盘信息传入。这导致存储池识别问题。同时因为系统分区不同步，还经常出现“系统分区损坏”的提示（截图见 #199）。

目前问题疑似仅限于连接在SAS HBA上的硬盘，直连主板SATA口的硬盘不受影响。

logs:  

...

snailium, 2023-09-06:
To solve "missing" storage pool.

1. Remove all HDD related to that pool.
2. Remove pool from Storage Manager
3. Put HDD back in (hot plug also works)
4. Wait until all HDD are identified by DSM
5. Do a "Repair" or "Reconstruction" on storage pool.

wjz304, 2023-09-09:
https://github.com/wjz304/arpl-i18n/issues/199

wjz304, 2023-09-09:
https://github.com/wjz304/arpl-i18n/issues/206

wjz304, 2023-09-13:
https://github.com/wjz304/arpl-i18n/releases/tag/23.9.2

wjz304, 2023-09-15:
Follow up on #199

closed 2023-09-15 by wjz304