RROrg/rr #202: BUG

unecc, 2023-09-06:
Install ENV:  
 * motherboard:  PROXMOX
 * CPU:  emulated
 * NIC:  

Arpl version:  23.9

DSM model:  ds920
DSM version:  7.2

Issue:  
ETH0 1000 TIMEOUT

[image]

wjz304, 2023-09-06:
[image]

unecc, 2023-09-06:
sorry, what means?

wjz304, 2023-09-06:
don't mind it if you can connect to the device normally

wjz304, 2023-09-06:
It takes a certain amount of time to obtain the ip. In order to start it as soon as possible, if you do not obtain it within the timeout period, you will not wait and go directly to the next step.
so, it is not necessarily true that there is no ip

wjz304, 2023-09-06:
If you can't connect to the device, look here first
https://github.com/wjz304/arpl-i18n/issues/175

wjz304, 2023-09-16:
No further feedback, if there are still issues, please create a new issue and provide detailed information

wjz304, 2023-09-16:
https://github.com/wjz304/arpl-i18n/issues/218

closed 2023-09-16 by wjz304