RROrg/rr #203: Stuck on boot on SA6400 Intel Celeron N5105

Juliyo, 2023-09-06:
Install ENV:  
 * motherboard:  Shenzhen Wangang Technology Co., Ltd AERO 3 Lite
 * CPU:  Intel Celeron N5105
 * NIC:  What is this?

Arpl version:  23.9.0

DSM model:  SA6400
DSM version:  7.1 and 7.2

Issue:  
kernel crashing on boot

logs:  can't get logs as kernel is crashing

I'm trying to install model SA6400, but after build/booting I can't find the device in network to install DSM. It seems that kernel is crashing because the cursor stops blinking and my keyboard loses power. I tried recommendations in your post "About not find DSM after "boot the loader"" but I still get same problem.
I'm a bit lost on the matter, but from what I've read in forums, it seems that for Intel processors, you should use the jim3ma modified version. However, the result I'm getting with it is the same.

wjz304, 2023-09-18:
https://github.com/wjz304/arpl-i18n/issues/173

Juliyo, 2023-09-19:
> #173

[image]

wjz304, 2023-09-19:
try get the boot logs:
enable dbgutils addon in arpl,
then build, boot and wait 10 minutes,
finally check /mnt/p1/logs in arpl.

Juliyo, 2023-09-19:
I tried but when booting I can't use terminal anymore. I have to physical reboot PC and then after reboot If I check /mnt/p1/logs folder, it does not exists.

Juliyo, 2023-09-19:
😟 ok I understand. Thank you

wjz304, 2023-09-19:
Is your bootloader USB?

Juliyo, 2023-09-19:
Yes

wjz304, 2023-09-19:
> Yes. Why?

Nothing, sa6400 does not support sata, if sata, it will cause kernel panic

Juliyo, 2023-09-19:
Ok. Then I just can wait for future updates. Normally, this kind of problem is related with some unsupported driver? Right?

closed 2023-09-21 by Juliyo