RROrg/rr #206: Hard disk system section error on the last HDD disk

aarpro, 2023-09-09:
Install ENV:  
 * motherboard:  GIGABYTE GA-H110TN / ASRock J3355B-ITX / Adwantech ASMB-260I-21A1
 * CPU:  Core i5-6600T / Celeron J3355  / Atom C3558
 
Arpl version: 23.9.1

DSM model:  DS918+ / DS918+ / DS3622
DSM version: DSM 7.2-64570 Update 3 on all

two DS918 server and DS3622 - 2 SSD cache on motherboard SATA, 4 HDD on LSI SAS 9211-4i controller. 
When rebooting system section error in the last HDD disk.

Problem after installation addon modification storage panel.
[image]

DS918+ error
[image]
[image]

DS3622 error
[image]
[image]

wjz304, 2023-09-15:
Follow up on #199

closed 2023-09-15 by wjz304

aarpro, 2023-09-15:
noe its OK - Thk