RROrg/rr #208: change quantities HDD (storage panel addon)

aarpro, 2023-09-09:
what parameters I have to add to change addons set default value RACK_24_BAY 1X8

Is it possible to setings quantity and size disk in storage panel add to menu?

[image]
[image]

wjz304, 2023-09-09:
[image]
[image]

wjz304, 2023-09-09:
Some models do not display m2,(such as DS3622)

closed 2023-09-10 by wjz304