RROrg/rr #209: DS918+ (ApolloLake) and Dell Perc H310 cannot be used.

PeterSuh-Q3, 2023-09-10:
It is processed normally in DELL PERC H200 or 9211-4i that use the same mpt3sas module.

It seems that joint confirmation of the cause of the module error is necessary.
If we can determine the cause, we will report further.

`[   20.848770] mpt3sas version 41.00.00.00 loaded
[   20.856223] mpt3sas 0000:01:00.0: can't disable ASPM; OS doesn't have ASPM control
[   20.866706] mpt2sas_cm0: 64 BIT PCI BUS DMA ADDRESSING SUPPORTED, total mem (16276444 kB)
[   20.874994] mpt2sas_cm0: doorbell is in use  (line=4858)
[   20.880380] mpt2sas_cm0: _base_get_ioc_facts: handshake failed (r=-14)
[   20.886988] mpt2sas_cm0: _base_unmap_resources
[   20.891585] mpt2sas_cm0: failure at /tmp/input/drivers/scsi/mpt3sas/mpt3sas_scsih.c:16229/_scsih_probe()!`

PeterSuh-Q3, 2023-09-10:
I'm sorry.
It seems to me like an intermittent error.
Afterwards, I repeated the test again.
Migrated from DS3622xs+ to DS918+,
After doing a clean disk, I tried reinstalling DS918+.
In this repeated test, H310 operates normally again.
We will watch the instability issue of H310 a little more closely.

wjz304, 2023-09-10:
mpt3sas 驱动有待优化

wjz304, 2023-09-10:
考虑 使用syno 官方的 mpt3sas 驱动，你觉得呢？

snailium, 2023-09-11:
Looks like queue depth issue.

Reference: https://bugzilla.kernel.org/show_bug.cgi?id=209177

Solution for normal linux is to let mpt3sas initialized with large queue depth.

`GRUB_CMDLINE_LINUX_DEFAULT="quiet splash mpt3sas.max_queue_depth=10000"`

Not sure how to add the option to ARPL.

PeterSuh-Q3, 2023-09-11:
> 考虑 使用syno 官方的 mpt3sas 驱动，你觉得呢？

另一个目的是让 mpt3sas 在 ds920+ 上工作。
我尝试了Synology的官方驱动程序版本09.102.00.00。
目前，pocopico 使用的是从 Broadcom 导入的最新版本 41.00.00.00。
这次回头还有什么好处吗？

https://github.com/PeterSuh-Q3/arpl-modules/tree/e06992f583ef918122fa721d75e50d58e9e3e28d/src/4.x/drivers/scsi/mpt3sas

PeterSuh-Q3, 2023-09-11:
> Looks like queue depth issue.
> 
> Reference: https://bugzilla.kernel.org/show_bug.cgi?id=209177
> 
> Solution for normal linux is to let mpt3sas initialized with large queue depth.
> 
> `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash mpt3sas.max_queue_depth=10000"`
> 
> Not sure how to add the option to ARPL.

In ChatGPT’s guide,
It says to add it to Synology /etc.default/syslinux.cfg, but I'm not sure if it's correct.
I'll take a closer look when I get home.
Thank you for answer.

wjz304, 2023-09-11:
> 看起来像队列深度问题。
> 
> 参考： https://bugzilla.kernel.org/show_bug.cgi?id=209177
> 
> 普通 Linux 的解决方案是让 mpt3sas 使用大队列深度初始化。
> 
> `GRUB_CMDLINE_LINUX_DEFAULT="quiet splash mpt3sas.max_queue_depth=10000"`
> 
> 不确定如何将选项添加到 ARPL。

添加到 cmdline 即可

snailium, 2023-09-12:
I'm thinking if the mpt3sas driver caused #199 #201 #206.

I've tried to apply max_queue_depth in cmdline, but this is incoorect. It only get applied when booting DSM. I'm looking for a method to apply max_queue_depth for ARPL (not DSM). There is no `/etc/default/grub`.

FYI,
I can see HDDs are correct identified in ARPL (not DSM), by running `dmesg | grep mpt` and `ls -l /sys/class/scsi_host`

snailium, 2023-09-13:
Not sure if this is mpt3sas driver issue.

Here I tried "direct boot" instead of entering ARPL.

`dmesg | grep mpt` shows the following

`[    2.398623] mpt3sas version 41.00.00.00 loaded
[    2.406834] mpt2sas_cm0: 64 BIT PCI BUS DMA ADDRESSING SUPPORTED, total mem (8097280 kB)
[    2.482053] mpt2sas_cm0: IOC Number : 0
[    2.482180] mpt2sas_cm0: CurrentHostPageSize is 0: Setting default host page size to 4k
[    2.502510] mpt2sas0-msix0: PCI-MSI-X enabled: IRQ 34
[    2.502510] mpt2sas0-msix1: PCI-MSI-X enabled: IRQ 35
[    2.502511] mpt2sas0-msix2: PCI-MSI-X enabled: IRQ 36
[    2.502511] mpt2sas0-msix3: PCI-MSI-X enabled: IRQ 37
[    2.502512] mpt2sas_cm0: iomem(0x00000000c2040000), mapped(0xffffc90000460000), size(65536)
[    2.502513] mpt2sas_cm0: ioport(0x000000000000d000), size(256)
[    2.575050] mpt2sas_cm0: IOC Number : 0
[    2.575206] mpt2sas_cm0: CurrentHostPageSize is 0: Setting default host page size to 4k
[    2.612078] mpt2sas_cm0: scatter gather: sge_in_main_msg(1), sge_per_chain(9), sge_per_io(128), chains_per_io(15)
[    2.612713] mpt2sas_cm0: request pool(0xffff88026b400000) - dma(0x26b400000): depth(8320), frame_size(128), pool_size(1040 kB)
[    2.628912] mpt2sas_cm0: sense pool(0xffff880032200000) - dma(0x32200000): depth(8059), element_size(96), pool_size (755 kB)
[    2.629454] mpt2sas_cm0: reply pool(0xffff880264a00000) - dma(0x264a00000): depth(8384)frame_size(128), pool_size(1048 kB)
[    2.629922] mpt2sas_cm0: config page(0xffff8800342c0000) - dma(0x342c0000): size(512)
[    2.630247] mpt2sas_cm0: Allocated physical memory: size(18510 kB)
[    2.630506] mpt2sas_cm0: Current Controller Queue Depth(8056), Max Controller Queue Depth(8192)
[    2.691038] mpt2sas_cm0: LSISAS2308: FWVersion(20.00.07.00), ChipRevision(0x05)
[    2.691357] mpt2sas_cm0: Protocol=(Initiator,Target), Capabilities=(TLR,EEDP,Snapshot Buffer,Diag Trace Buffer,Task Set Full,NCQ)
[    2.692088] mpt3sas 0000:01:00.0: Enabled Extended Tags as Controller Supports
[    2.693103] mpt2sas_cm0: sending port enable !!
[    2.695175] mpt2sas_cm0: hba_port entry: ffff880270781340, port: 255 is added to hba_port list
[    2.698293] mpt2sas_cm0: host_add: handle(0x0001), sas_addr(0x500304801b3d9400), phys(8)
[    2.700188] mpt2sas_cm0: handle(0xa) sas_address(0x4433221103000000) port_type(0x1)
[    2.701701] mpt2sas_cm0: handle(0xb) sas_address(0x5000c50056891ce1) port_type(0x1)
[    2.703266] mpt2sas_cm0: handle(0x9) sas_address(0x5000c50056896795) port_type(0x1)
[    2.704043] mpt2sas_cm0: port enable: SUCCESS
[    2.704440] mpt2sas_cm0: detecting: handle(0x000d), sas_address(0x5000c50056896855), phy(6)
[    2.704786] mpt2sas_cm0: REPORT_LUNS: handle(0x000d), retries(0)
[    2.712923] mpt2sas_cm0: TEST_UNIT_READY: handle(0x000d), lun(0)
[    2.713435] mpt2sas_cm0: handle(0xd) sas_address(0x5000c50056896855) port_type(0x1)
[    2.728611]  end_device-6:0: mpt3sas_transport_port_add: added: handle(0x000a), sas_addr(0x4433221103000000)
[   17.793545]  end_device-6:1: mpt3sas_transport_port_add: added: handle(0x000d), sas_addr(0x5000c50056896855)
[   17.795183] mpt2sas_cm0: detecting: handle(0x000c), sas_address(0x5000c500568970b9), phy(4)
[   17.795465] mpt2sas_cm0: REPORT_LUNS: handle(0x000c), retries(0)
[   17.810351] mpt2sas_cm0: TEST_UNIT_READY: handle(0x000c), lun(0)
[   17.810818] mpt2sas_cm0: handle(0xc) sas_address(0x5000c500568970b9) port_type(0x1)
[   32.896036]  end_device-6:2: mpt3sas_transport_port_add: added: handle(0x000b), sas_addr(0x5000c50056891ce1)
[   47.975743]  end_device-6:3: mpt3sas_transport_port_add: added: handle(0x000c), sas_addr(0x5000c500568970b9)
[   48.784972] systemd[1]: Reached target Login Prompts.
[   48.785181] systemd[1]: Starting Login Prompts.
[   63.072647]  end_device-6:4: mpt3sas_transport_port_add: added: handle(0x0009), sas_addr(0x5000c50056896795)
`

Given I have the following attached on the same SAS2308 HBA.
> Port 0: SATA SSD (0x4433221103000000)
> Port 1: SAS HDD (0x5000c50056891ce1)
> Port 3: SAS HDD (0x5000c50056896795)
> Port 5: SAS HDD (0x5000c50056896855)
> Port 7: SAS HDD (0x5000c500568970b9)

System detected only the first three (1xSATA + 2xSAS) at the beginning, and detected the other 2xSAS (Port 5 and 7) way behind.

After boot into DSM, Storage Manager shows 2xSAS corrupted, which are Port 3 and 7. It is not always the last detected shown as corrupted. In my case Port 3 (0x5000c50056896795) was detected from the beginning, but still shown as corrupted.

BTW, on this DS3622xs+, Storage Manger doesn't show SMART information for SAS drives. Which I think it did before. That may be another evidence that the MPT driver introduces some problems.

PeterSuh-Q3, 2023-09-13:
This is a review of what I experienced in the past on Dell Perc H310 with DSM 6.2.3.
At that time, there was a problem in which some disks were not recognized by the HBA, and there was a difficult time because the RAID in DSM kept failing.

According to the advice of RAID controller experts, most of the IT Mode firmware used in SAS controllers are legacy firmware.
It is recommended to change the Storage Boot Option to Legacy instead of EFI during BIOS setup in MOBO.
Of course, EFI firmware also exists separately.

[image]

By doing this, the SAS controller's BIOS boot screen will additionally appear as shown below during BIOS booting, and sufficient timing to accurately detect disks can be obtained during this process.

[image]

As advised by experts, I always use the Storage Boot Option when using an HBA. Fast Boot is also not used.
After maintaining these settings, the phenomenon of losing disks from DSM has decreased significantly.
I think this theory can be applied to DSM 7 as well.

So far, there has been one user complaining of this phenomenon, one from ARPL and one from TCRP.
We are asking two users about their BIOS Storage Boot Option settings.

wjz304, 2023-09-15:
Follow up on #199

closed 2023-09-15 by wjz304