RROrg/rr #214: full RAM size is not showing on DSM

aarpro, 2023-09-12:
* Advantech ASMB-260I-21A1
 * Atom C3558  
 * RAM 4x16
 
Arpl version: 23.9.1 
DSM model: DS3622
DSM version: DSM 7.2-64570 Update 3

DSM not showing full RAM size (64GB)

When viewing from the terminal - 64GB
[image]

When viewing from DSM - only 32GB
[image]

and incomprehensible size of reserved memory
[image]

I tried changing in the settings file /etc/synoinfo.conf & /etc.defaults/synoinfo.conf
my old values
```
mem_max_mb="49152"
mem_min_mb="16384"
```
my new values
```
mem_max_mb="65536"
mem_min_mb="16384"
```
The result is negative

I didn't think it might be a loaders problem, but to me gave advice to change loader to ARC. [xpenology.com/forum](https://xpenology.com/forum/topic/69644-not-showing-full-ram-size/?do=findComment&comment=449769)
I have no physical access to my server now, and I can't try another loader

I also found advice to add to cmdline menu, parameter disable_mtrr_trim with null value.
but I not sure that this will not break loading of the NAS

aarpro, 2023-09-12:
i add in cmdline menu, **disable_mtrr_trim** with null value.
and **direct boot** set to true

result is negative - nothing has changed :(

wjz304, 2023-09-13:
[image]
[image]

我没有做任何修改，但似乎不能复现你的问题.

wjz304, 2023-09-13:
物理机？

aarpro, 2023-09-13:
> 物理机？

yes
But I no have physical access yet.
A few days later I can check RAM and check BIOS settings

aarpro, 2023-09-13:
> 物理机？

When loading, I see 64GB RAM, I think this is no problem of memory modules, its soft problems

[image]

wjz304, 2023-09-15:
Have you made any progress?

aarpro, 2023-09-15:
> Have you made any progress?

no - I no have physical access to the server yet
but i see BIOS 64GB RAM, loaders 64GB RAM - DCM detected only 32GB, and reserved -32GB RAM
so i thunk its is soft problems

In what direction I need to look for?

wjz304, 2023-09-15:
I don't know, just a little skeptical. Is it a swap?

aarpro, 2023-09-15:
> swap

What settings can I change to change swap size or over swap parameters ?

zak-forsyth, 2023-09-19:
What does dmesg say about your memory?

`dmesg | grep -i memory`

Does it look something similar to this?...

`dmesg | grep -i memory
[    0.000000] WARNING: BIOS bug: CPU MTRRs don't cover all of memory, losing 29440MB of RAM.
[    0.000000] WARNING: CPU: 0 PID: 0 at arch/x86/kernel/cpu/mtrr/cleanup.c:972 mtrr_trim_uncached_memory+0x458/0x47f()
[    0.000000]  [<ffffffff8190368c>] mtrr_trim_uncached_memory+0x458/0x47f`

If so `disable_mtrr_trim` will fix it.

aarpro, 2023-09-22:
> 
> `dmesg | grep -i memory`
> 

```
root@DS3622:~# dmesg | grep -i memory
[    0.000000] Base memory trampoline at [ffff880000096000] 96000 size 28672
[    0.000000] Early memory node ranges
[    0.000000] PM: Registered nosave memory: [mem 0x00000000-0x00000fff]
[    0.000000] PM: Registered nosave memory: [mem 0x000a0000-0x000fffff]
[    0.000000] PM: Registered nosave memory: [mem 0x00100000-0x00100fff]
[    0.000000] PM: Registered nosave memory: [mem 0x00102000-0x00102fff]
[    0.000000] PM: Registered nosave memory: [mem 0x76d35000-0x76d44fff]
[    0.000000] PM: Registered nosave memory: [mem 0x7d63c000-0x7d6dbfff]
[    0.000000] PM: Registered nosave memory: [mem 0x7d7b1000-0x7db06fff]
[    0.000000] PM: Registered nosave memory: [mem 0x7db07000-0x7f442fff]
[    0.000000] PM: Registered nosave memory: [mem 0x7f800000-0x7fffffff]
[    0.000000] PM: Registered nosave memory: [mem 0x80000000-0xdfffffff]
[    0.000000] PM: Registered nosave memory: [mem 0xe0000000-0xefffffff]
[    0.000000] PM: Registered nosave memory: [mem 0xf0000000-0xfcffffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfd000000-0xfe7fffff]
[    0.000000] PM: Registered nosave memory: [mem 0xfe800000-0xfeffffff]
[    0.000000] PM: Registered nosave memory: [mem 0xff000000-0xffffffff]
[    0.000000] Memory: 65752288K/67070324K available (5633K kernel code, 882K rw                                                                                data, 1776K rodata, 936K init, 1568K bss, 1318036K reserved, 0K cma-reserved)
[    0.008644] Initializing cgroup subsys memory
[    0.020242] Freeing SMP alternatives memory: 24K
[    3.938591] Freeing initrd memory: 59804K
[    5.856347] Freeing unused kernel memory: 936K
[    5.857378] Freeing unused kernel memory: 500K
[    5.858645] Freeing unused kernel memory: 272K
[    7.024694] mpt2sas_cm0: Allocated physical memory: size(7579 kB)
```

This is what I see

wjz304, 2023-09-23:
加上试试，

aarpro, 2023-09-23:
> If so `disable_mtrr_trim` will fix it.

> 加上试试，

I can't understand how to do it  :(

wjz304, 2023-09-23:
[image]

aarpro, 2023-09-23:
> 加上试试，

[image]

[image]

unchanged - does not work as it should

wjz304, 2023-09-23:
'value' does not need to be filled in

zak-forsyth, 2023-09-23:
What they said. No value for mtrr trim.

aarpro, 2023-09-23:
> 'value' does not need to be filled in

[image]

all the same

[image]

wjz304, 2023-09-23:
```
free -m
cat /proc/meminfo
```

aarpro, 2023-09-23:
> in dsm shell:
> 
> ```
> free -m
> cat /proc/meminfo
> ```

```
root@DS3622:~# free -m
              total        used        free      shared  buff/cache   available
Mem:          64335        1838       60532         112        1964       61868
Swap:         40651           0       40651
root@DS3622:~# cat /proc/meminfo
MemTotal:       65879776 kB
MemFree:        61985320 kB
MemAvailable:   63353380 kB
Buffers:           23000 kB
Cached:          1495004 kB
SwapCached:            0 kB
Active:          1877448 kB
Inactive:         887124 kB
Active(anon):    1264320 kB
Inactive(anon):    98948 kB
Active(file):     613128 kB
Inactive(file):   788176 kB
Unevictable:        1676 kB
Mlocked:            1676 kB
SwapTotal:      41627564 kB
SwapFree:       41627564 kB
Dirty:               144 kB
Writeback:             0 kB
AnonPages:       1248292 kB
Mapped:           296724 kB
Shmem:            115044 kB
Slab:             572576 kB
SReclaimable:     493764 kB
SUnreclaim:        78812 kB
KernelStack:       10080 kB
PageTables:        15480 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:    74567452 kB
Committed_AS:    3154760 kB
VmallocTotal:   34359738367 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
DirectMap4k:       17144 kB
DirectMap2M:     3090432 kB
DirectMap1G:    66060288 kB
root@DS3622:~#
```

wjz304, 2023-09-23:
There doesn't seem to be anything unusual, it's just 32G displayed

aarpro, 2023-09-24:
> There doesn't seem to be anything unusual, it's just 32G displayed

I understand correctly that displayed 
`Reserved -32325344KB (minus!)`
This is only a display in DSM

I just don't pay attention to it?

wjz304, 2023-09-24:
try to add a fake mac2 to enable netif_num is 2

aarpro, 2023-09-24:
> try to add a fake mac2 to enable netif_num is 2

I have a valid sn/mac and this is added, I see my server on account.synology.com and QuickConnect is work...
on my mb is one GbE Lan

or I have to add the second MAC in any case ? Does this have any relation with the quantity RAM?

wjz304, 2023-09-24:
> > try to add a fake mac2 to enable netif_num is 2
> 
> I have a valid sn/mac and this is added, I see my server on account.synology.com and QuickConnect is work... on my mb is one GbE Lan
> 
> or I have to add the second MAC in any case ? Does this have any relation with the quantity RAM?

I'm not sure,

It's just that I just found someone on a Chinese forum with a similar situation, and after adding a PCIe network card, this situation disappeared

aarpro, 2023-09-24:
> I'm not sure...

[image]

no, It did not help, all the same

[image]

wjz304, 2023-09-25:
Okay, if it doesn't affect usage,  don't attention to it first

AuxXxilium, 2023-09-25:
try to add "crashkernel" "192M" to cmdline in addition to "disable_mtrr_trim" "0" and report again please

aarpro, 2023-09-26:
> ry to add "crashkernel" "192M" to cmdline in addition to "disable_mtrr_trim" "0" and report again please

[image]

unchanged :(

[image]

[image]

```
root@DS3622:~# free -m
              total        used        free      shared  buff/cache   available
Mem:          64143        1828       60398         112        1916       61686
Swap:         40535           0       40535
root@DS3622:~# cat /proc/meminfo
MemTotal:       65683168 kB
MemFree:        61831956 kB
MemAvailable:   63168568 kB
Buffers:           23184 kB
Cached:          1462380 kB
SwapCached:            0 kB
Active:          1840156 kB
Inactive:         879244 kB
Active(anon):    1251832 kB
Inactive(anon):    98556 kB
Active(file):     588324 kB
Inactive(file):   780688 kB
Unevictable:        1676 kB
Mlocked:            1676 kB
SwapTotal:      41508780 kB
SwapFree:       41508780 kB
Dirty:              1416 kB
Writeback:             0 kB
AnonPages:       1235520 kB
Mapped:           296596 kB
Shmem:            114896 kB
Slab:             573616 kB
SReclaimable:     494608 kB
SUnreclaim:        79008 kB
KernelStack:       10384 kB
PageTables:        15828 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:    74350364 kB
Committed_AS:    3236336 kB
VmallocTotal:   34359738367 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
DirectMap4k:       17144 kB
DirectMap2M:     3090432 kB
DirectMap1G:    66060288 kB
root@DS3622:~#
```

AuxXxilium, 2023-09-26:
do you have tried to force reinstall dsm?

aarpro, 2023-09-26:
> do you have tried to force reinstall dsm?

I remember correctly: To save the data, I need to delete the first partition on all HDD, and reinstall DSM ?

AuxXxilium, 2023-09-27:
no. you have an option to "force reinstall dsm" in bootmenu. only your settings can be reverted.

aarpro, 2023-10-07:
> do you have tried to force reinstall dsm?

sorry for the long answer

so

i reinstall DSM - now install [arpl-i18n](https://github.com/wjz304/arpl-i18n) and full all (addons/modules) update
no changes :(
I have 6 modules x 16 Gb - Installing them in any sequence does not give a positive result in the same way
[image]
If I install any 2 modules - all is ok - DCM detected 32GB
If I install any 3 modules - same problem - DCM not detected 48GB - only 32GB
[image]

Just in case, below the link to datasheet and user manual my motherboard - maybe I didn't turn on something in the BIOS settings ? 

[ASMB-260I-21A1](https://drive.google.com/drive/folders/1Yt1k7sRh4Uz4C1LGkxwa1Zl9FrK1BQyc?usp=sharing)

wjz304, 2023-10-07:
You can use windows pe  system to troubleshoot dsm problems

wjz304, 2023-10-07:
Oh, no need. The previous log has been identified as 64gb.

aarpro, 2023-10-08:
> Oh, no need. The previous log has been identified as 64gb.

i reinstall DSM - new usb flash, DS923+ 

[image]

no changes :(

[image]

full RAM size is not showing on DSM

aarpro, 2023-10-08:
I'm back to DSM DS3622
Installed other memory modules
3 x 16  Apcher 2666

[image]

no changes :(

[image]

aarpro, 2023-10-08:
> no. you have an option to "force reinstall dsm" in bootmenu. only your settings can be reverted.

i install arc-23.10.2c DS3622

Memory installed 3x16 2666 RDIMM
Arc Loader info showing only 32GB

[image]

[image]

DSM showing 32GB total, **46GB** free, and reserved ~16GB

[image]

free -m showing
```
              total        used        free      shared  buff/cache   available
Mem:          48207         424       47386          15         397       47379
Swap:         30975           0       30975
root@DS3622test:~# cat /proc/meminfo
MemTotal:       49364724 kB
MemFree:        48523940 kB
MemAvailable:   48517348 kB
Buffers:           18928 kB
Cached:           356068 kB
SwapCached:            0 kB
Active:           374768 kB
Inactive:         170652 kB
Active(anon):     175820 kB
Inactive(anon):    11512 kB
Active(file):     198948 kB
Inactive(file):   159140 kB
Unevictable:        1496 kB
Mlocked:            1496 kB
SwapTotal:      31719340 kB
SwapFree:       31719340 kB
Dirty:                88 kB
Writeback:             0 kB
AnonPages:        171956 kB
Mapped:           134648 kB
Shmem:             15424 kB
Slab:              74080 kB
SReclaimable:      31776 kB
SUnreclaim:        42304 kB
KernelStack:        4960 kB
PageTables:         7688 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:    56401700 kB
Committed_AS:     941004 kB
VmallocTotal:   34359738367 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
DirectMap4k:       15096 kB
DirectMap2M:     3092480 kB
DirectMap1G:    49283072 kB
```

wjz304, 2023-10-09:
try lowering the memory of the integrated graphics card in the BIOS settings.

aarpro, 2023-10-09:
> try lowering the memory of the integrated graphics card in the BIOS settings.

sorry - There are no such settings in this BIOS :(

wjz304, 2023-10-09:
Sorry, I don't know

I can't reproduce this problem. Although I found some similar cases, there seems to be no solution. Some of the above attempts are just guesswork.

AuxXxilium, 2023-10-09:
> i install arc-23.10.2c DS3622
>
>
>
> Memory installed 3x16 2666 RDIMM
>
> Arc Loader info showing only 32GB
>
>
>
> [image]
>
>
>
> [image]
>
>
>
> DSM showing 32GB total, **46GB** free, and reserved ~16GB
>
>
>
> [image] [...]
try ramfix option in arc.

aarpro, 2023-10-09:
> try ramfix option in arc.

- Is this option somewhere on the Arc menu?
- i install 4 modules too - it is no wrong slot install.

AuxXxilium, 2023-10-09:
yes it is in arc options. i think in cmdline. 

do you have enabled memory remap in bios or other options like mirroring?

aarpro, 2023-10-09:
> do you have enabled memory remap in bios or other options like mirroring?

There are no any memory settings in this BIOS :(

AuxXxilium, 2023-10-09:
> > do you have enabled memory remap in bios or other options like mirroring?
> 
> 
> 
> There are no any memory settings in this BIOS :(  
> 
> BIOS and usermanual https://drive.google.com/file/d/1_vBlO7E0zF8sGdbBG_I-zgD2RGbAEQHU/view?usp=sharing

what's wrong with this vendor? no memory option, no way to set memory speed, scrub, mirroring and so on...

aarpro, 2023-10-10:
> try ramfix option in arc.

- Install 4x16GB memory modules on 4 slots
- set BIOS by default
- BIOS dtected 64GB
- ARC loaders at the first load detected only 32GB
- Ramfix option added

[image]

aarpro, 2023-10-10:
> what's wrong with this vendor? no memory option, no way to set memory speed, scrub, mirroring and so on...

advantech is very peculiar :(
I often deal with them - I do not recommend them to anyone

aarpro, 2023-10-10:
> Sorry, I don't know

- ARC loaders at the first load detected only 32GB
- arpl-i18n loaders detected only 64GB

[image]

and It does not matter which bootloader - DSM not detected full (64GB) RAM

AuxXxilium, 2023-10-10:
arc is reading your memory sticks installed. arpl is showing the free memory that the system has calculated. dsm is mixing both.

aarpro, 2023-10-10:
> arc is reading your memory sticks installed

ok 
but installed 64gb (4x16) - Why Arc not detected it ?

wjz304, 2023-10-10:
@aarpro 
try add `disable_mtrr_cleanup` to cmdline 

and get the log:
```
dmesg | grep mtrr
```

aarpro, 2023-10-10:
> because your bios didn't output the correct value.

ok 
Apparently i will have to use only 32GB

aarpro, 2023-10-10:
> try add `disable_mtrr_cleanup` to cmdline

[image]

> and get the log:

```
root@DS3622:~# dmesg | grep mtrr
[    0.000000] Command line: withefi disable_mtrr_cleanup earlyprintk syno_hw_version=DS3622xs+ console=ttyS0,115200n8 netif_num=1 pid=0x5571 earlycon=uart8250,io,0x3f8,115200n8 mac1=001132FA6CD5 sn=2150SQRW1ZAHH vid=0x0781 elevator=elevator loglevel=15 HddHotplug=0 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 syno_hdd_detect=0 panic=0 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0
[    0.000000] Kernel command line: withefi disable_mtrr_cleanup earlyprintk syno_hw_version=DS3622xs+ console=ttyS0,115200n8 netif_num=1 pid=0x5571 earlycon=uart8250,io,0x3f8,115200n8 mac1=001132FA6CD5 sn=2150SQRW1ZAHH vid=0x0781 elevator=elevator loglevel=15 HddHotplug=0 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 syno_hdd_detect=0 panic=0 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0
root@DS3622:~# dmidecode -t 17
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0038, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0036
        Error Information Handle: Not Provided
        Total Width: 8 bits
        Data Width: 8 bits
        Size: 16384 MB
        Form Factor: DIMM
        Set: None
        Locator: DIMM0
        Bank Locator: BANK 0
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 2666 MT/s
        Manufacturer: Crucial
        Serial Number: 06C2B641
        Asset Tag: BANK 0 DIMM0 AssetTag
        Part Number: 36ASF2G72PZ-2G3A3
        Rank: 2
        Configured Memory Speed: 2133 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V

Handle 0x003A, DMI type 17, 40 bytes
Memory Device
        Array Handle: 0x0036
        Error Information Handle: Not Provided
        Total Width: 8 bits
        Data Width: 8 bits
        Size: 16384 MB
        Form Factor: DIMM
        Set: None
        Locator: DIMM1
        Bank Locator: BANK 0
        Type: DDR4
        Type Detail: Synchronous Unbuffered (Unregistered)
        Speed: 2666 MT/s
        Manufacturer: Crucial
        Serial Number: 07C2B641
        Asset Tag: BANK 0 DIMM1 AssetTag
        Part Number: 36ASF2G72PZ-2G3A3
        Rank: 2
        Configured Memory Speed: 2133 MT/s
        Minimum Voltage: 1.2 V
        Maximum Voltage: 1.2 V
        Configured Voltage: 1.2 V
```

wjz304, 2023-10-11:
> > try add `disable_mtrr_cleanup` to cmdline
> 
> [image]
> 
> > and get the log:
> 
> ```
> root@DS3622:~# dmesg | grep mtrr
> [    0.000000] Command line: withefi disable_mtrr_cleanup earlyprintk syno_hw_version=DS3622xs+ console=ttyS0,115200n8 netif_num=1 pid=0x5571 earlycon=uart8250,io,0x3f8,115200n8 mac1=0011XXXXXXXX sn=2150SXXXXXX vid=0x0781 elevator=elevator loglevel=15 HddHotplug=0 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 syno_hdd_detect=0 panic=0 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0
> [    0.000000] Kernel command line: withefi disable_mtrr_cleanup earlyprintk syno_hw_version=DS3622xs+ console=ttyS0,115200n8 netif_num=1 pid=0x5571 earlycon=uart8250,io,0x3f8,115200n8 mac1=0011XXXXXXXXX sn=2150XXXXXXXXX vid=0x0781 elevator=elevator loglevel=15 HddHotplug=0 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 syno_hdd_detect=0 panic=0 vender_format_version=2 syno_hdd_powerup_seq=0 log_buf_len=32M root=/dev/md0
> root@DS3622:~# dmidecode -t 17
> # dmidecode 3.2
> Getting SMBIOS data from sysfs.
> SMBIOS 3.0.0 present.
> 
> Handle 0x0038, DMI type 17, 40 bytes
> Memory Device
>         Array Handle: 0x0036
>         Error Information Handle: Not Provided
>         Total Width: 8 bits
>         Data Width: 8 bits
>         Size: 16384 MB
>         Form Factor: DIMM
>         Set: None
>         Locator: DIMM0
>         Bank Locator: BANK 0
>         Type: DDR4
>         Type Detail: Synchronous Unbuffered (Unregistered)
>         Speed: 2666 MT/s
>         Manufacturer: Crucial
>         Serial Number: 06C2B641
>         Asset Tag: BANK 0 DIMM0 AssetTag
>         Part Number: 36ASF2G72PZ-2G3A3
>         Rank: 2
>         Configured Memory Speed: 2133 MT/s
>         Minimum Voltage: 1.2 V
>         Maximum Voltage: 1.2 V
>         Configured Voltage: 1.2 V
> 
> Handle 0x003A, DMI type 17, 40 bytes
> Memory Device
>         Array Handle: 0x0036
>         Error Information Handle: Not Provided
>         Total Width: 8 bits
>         Data Width: 8 bits
>         Size: 16384 MB
>         Form Factor: DIMM
>         Set: None
>         Locator: DIMM1
>         Bank Locator: BANK 0
>         Type: DDR4
>         Type Detail: Synchronous Unbuffered (Unregistered)
>         Speed: 2666 MT/s
>         Manufacturer: Crucial
>         Serial Number: 07C2B641
>         Asset Tag: BANK 0 DIMM1 AssetTag
>         Part Number: 36ASF2G72PZ-2G3A3
>         Rank: 2
>         Configured Memory Speed: 2133 MT/s
>         Minimum Voltage: 1.2 V
>         Maximum Voltage: 1.2 V
>         Configured Voltage: 1.2 V
> ```

Is this with 2 memory inserted?

aarpro, 2023-10-11:
> Is this with 2 memory inserted?

no 
4x16 in 4 slots

wjz304, 2023-10-11:
in arpl shell
```
dmidecode -t memory
```

wjz304, 2023-10-11:
[image]
Is this option turned on?

wjz304, 2023-10-11:
This is my machine with two slots, when one slot is inserted.
[image]

aarpro, 2023-10-11:
> Is this option turned on?

yes

[22 more replies]

closed 2024-11-14 by aarpro