RROrg/rr #216: SAS Drive Support

HPMSGEN8, 2023-09-15:
Good Day

I would like to know if SAS Drives are supported on DSM 7.2 via an HBA ?

Can you do an SATA/SAS Mix on an Storage Pool or does it have to be SATA or SAS for the Array.

Thanks

wjz304, 2023-09-15:
非DT型号是可以的， SATA or SAS 可以混搭

PeterSuh-Q3, 2023-09-16:
> Good Day
> 
> I would like to know if SAS Drives are supported on DSM 7.2 via an HBA ?
> 
> Can you do an SATA/SAS Mix on an Storage Pool or does it have to be SATA or SAS for the Array.
> 
> Thanks

SATA and SAS can be mixed, but it is not recommended to use them together in a RAID group.
Recently issues were identified with the mpt3sas module and SAS disk, so I plan to conduct additional tests after purchasing the disk.

closed 2023-09-16 by HPMSGEN8