RROrg/rr #217: 23.9.2 编译引导失败

penv13, 2023-09-15:
Arpl version:  23.9.2

DSM model:  DS918+
DSM version:  7.1-42661 Update 1

Issue:  
ESXI中的虚拟机，之前已使用的是旧的RedPill的引导并正常使用7.1，希望变更使用大佬的引导。
挂载新的引导进入web设置界面，点击恢复dsm系统成功识别为7.1.1并下载后编译引导，出现以下错误。
希望大佬能够指导，非常感谢。

logs:  
Patching zImage....
Patching Ramdisk....
UNRECOVERY ERROR: Patching with ramdisk-42951-post-init-script patch patching file us/sbin/init.post
Hunk 1 FAILED 31/31.
OptPrjQuotaw"$(GetPQMountOpt "${RootMountPath}*")"
# shellcheck disable SC2046
Mount "${RootMountPath}" /tmpRoot -o barrier 1, noatime ${OptPrjQuota}
十#######################################################################################
+SED_PATH='/tmpRoot/us/bin/sed'
+@@@CONFIG-MANIPULATORS-TOOLS@@@
+@@@CONFIG-GENERATED@@@
+UPSTART="/tmpRoot/usr/share/init"
+
+if ! echo; then
+ _replace_in_file '^start on' '#start on' $UPSTART/tty. conf
+ _replace_in_file "console output" "console none" $UPSTART/syno_poweroff_task. conf
+ _replace_in_file "console output" "console none" $UPSTART/burnin_loader. conf
+ _replace_in_file "console output" "console none" $UPSTART/udevtrigger.conf
+ _replace_in_file "console output" "console none" $UPSTART/bs-poweroff. conf
+ _replace_in_file "console output" "console none" $UPSTART/udevd. conf
+else 
+ _replace in file "#start on' 'start on' SUPSTART/tty.conf
+fi
+/addons/addons. sh late
+#######################################################################################
Mkdir -p /tmpRoot/initrd
Umount /proc >/dev/null 2>&1

Randisk打补丁失败：
Patching with ramdisk-42951-post-init-script.patch patching file us/sbin/init.post Hunk 1 FAILED 31/31.
OptPrjQuota="$(GetPQMountOpt "${RootMountPath}")" #shellcheck disable-SC2046 Mount "${RootMountPath}' /tmpRoot -o barrier=1, noatime ${OptPrjQuota)
＋#######################################################################################
+SED_PATH-'/tmpRoot/usr/bin/sed'
+@@@CONFIG-MANIPULATORS-TOOLS@@@ + +@@@CONFIG-GENERATED@@@ +
+UPSTART-'/tmpRoot/us/share/init" + +if ! echo; then + _replace_in_file '^start on' '#start on' $UPSTART/tty.conf +
replace in_file "console output" "console none" SUPSTART/syno_poweroff_task.conf +
"console none" SUPSTART/burnin_loader.conf +
-replace_in_file "console output"
_replace in_file "console output" "console none" SUPSTART/udevtrigger.conf+
_replace_in_file "console output" "console none" $UPSTART/bs-poweroff.conf + _replace_in_file "console output" "console none" $UPSTART/udevd. conf +else + _replace_in_file '^#start on' 'start on' SUPSTART/tty.conf +fi + +/addons/addons. sh
late +#######################################################################################
Mkdir -p
/tmpRoot/initrd
Umount /proc >/dev/null 2>&1
...

wjz304, 2023-09-15:
引导的问题，现在不兼容 7.1.0  
1. 直接重编7.1 （会升级到7.1.1）
2. 等我研究研究如何解决

PeterSuh-Q3, 2023-09-16:
> Arpl version: 23.9.2
> 
> DSM model: DS918+ DSM version: 7.1-42661 Update 1
> 
> Issue: ESXI中的虚拟机，之前已使用的是旧的RedPill的引导并正常使用7.1，希望变更使用大佬的引导。 挂载新的引导进入web设置界面，点击恢复dsm系统成功识别为7.1.1并下载后编译引导，出现以下错误。 希望大佬能够指导，非常感谢。
> 
> logs: Patching zImage.... Patching Ramdisk.... UNRECOVERY ERROR: Patching with ramdisk-42951-post-init-script patch patching file us/sbin/init.post Hunk 1 FAILED 31/31. OptPrjQuotaw"$(GetPQMountOpt "${RootMountPath}*")" #shellcheck disable SC2046 Mount "${RootMountPath}" /tmpRoot -o barrier 1, noatime ${OptPrjQuota} +####################################################################################### +SED_PATH='/tmpRoot/us/bin/sed' +@@@CONFIG-MANIPULATORS-TOOLS@@@ +@@@CONFIG-GENERATED@@@ +UPSTART="/tmpRoot/usr/share/init" + +if ! echo; then
> 
> * _replace_in_file '^start on' '#start on' $UPSTART/tty. conf
> * _replace_in_file "console output" "console none" $UPSTART/syno_poweroff_task. conf
> * _replace_in_file "console output" "console none" $UPSTART/burnin_loader. conf
> * _replace_in_file "console output" "console none" $UPSTART/udevtrigger.conf
> * _replace_in_file "console output" "console none" $UPSTART/bs-poweroff. conf
> * _replace_in_file "console output" "console none" $UPSTART/udevd. conf
>   +else
> * _replace in file "#start on' 'start on' SUPSTART/tty.conf
>   +fi
>   +/addons/addons. sh late
>   +#######################################################################################
>   Mkdir -p /tmpRoot/initrd
>   Umount /proc >/dev/null 2>&1
> 
> Randisk打补丁失败： Patching with ramdisk-42951-post-init-script.patch patching file us/sbin/init.post Hunk 1 FAILED 31/31. OptPrjQuota="$(GetPQMountOpt "${RootMountPath}")" #shellcheck disable-SC2046 Mount "${RootMountPath}' /tmpRoot -o barrier=1, noatime ${OptPrjQuota) +####################################################################################### +SED_PATH-'/tmpRoot/usr/bin/sed' +@@@CONFIG-MANIPULATORS-TOOLS@@@ + +@@@CONFIG-GENERATED@@@ + +UPSTART-'/tmpRoot/us/share/init" + +if ! echo; then + _replace_in_file '^start on' '#start on' $UPSTART/tty.conf + replace in_file "console output" "console none" SUPSTART/syno_poweroff_task.conf + "console none" SUPSTART/burnin_loader.conf + -replace_in_file "console output" _replace in_file "console output" "console none" SUPSTART/udevtrigger.conf+ _replace_in_file "console output" "console none" $UPSTART/bs-poweroff.conf + _replace_in_file "console output" "console none" $UPSTART/udevd. conf +else + _replace_in_file '^#start on' 'start on' SUPSTART/tty.conf +fi + +/addons/addons. sh late +####################################################################################### Mkdir -p /tmpRoot/initrd Umount /proc >/dev/null 2>&1 ...

Recently at TCRP FRIEND, I also witnessed the same phenomenon as above.

The github.com domain is being blocked by web filtering in the company's firewall.
To bypass the raw.githubusercontent.com domain, I added a script as shown below.

https://github.com/PeterSuh-Q3/tcrpfriend/commit/026fcf7f2ef12509365d43dd7d1965a0559b5b1e

There may be a firewall web filtering or SSL authentication issue that affects the RAMDISK patch process.
Please check it out.

wjz304, 2023-09-17:
https://github.com/wjz304/arpl-i18n/releases/tag/23.9.3

penv13, 2023-09-17:
works fine with 23.9.3. thanks.

closed 2023-09-17 by penv13